espressif/esp-idf #11134: esp32-s3 build fail. fatal error: driver/temperature_sensor.h: No such file or directory

AirWind001, 2023-04-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDFv4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

build successfully

### What is the actual behavior?

fatal error: driver/temperature_sensor.h: No such file or directory

### Steps to reproduce.

1. Create a new project using idf.py create-project temp
2. copy main file from examples/peripherals/temperature_sensor)/temp_sensor/ of current git repository
3. idf.py set-target esp32-s3
4. idf.py build
...

### Build or installation Logs.

```plain
C:\Users\arvind\temp_sensor_test\temp>idf.py set-target esp32-s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'c:\users\arvind\temp_sensor_test\temp\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory c:\users\arvind\temp_sensor_test\temp\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 c:\users\arvind\temp_sensor_test\temp"...
-- Found Git: C:/Users/arvind/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1")
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/arvind/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe

-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Users/arvind/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Users/arvind/tools/xtensa-esp32s3-elf/esp-2021r2-patch3-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file C:/Users/arvind/temp_sensor_test/temp/sdkconfig
-- Found Python3: C:/Users/arvind/python_env/idf4.4_py3.8_env/Scripts/python.exe (found version "3.8.7") found components: Interpreter
[... 163 lines cut ...]
esptool.py v3.3-dev
Creating esp32s3 image...
Merged 1 ELF section
Successfully created esp32s3 image.
Generated C:/Users/arvind/temp_sensor_test/temp/build/bootloader/bootloader.bin
[101/101] cmd.exe /C "cd /D C:\Users\arvind\temp_sensor_test\temp\build\bootloader\esp-idf\esptool_py && C:\Users\arvind\python_env\idf4.4_py3.8_env\Scripts\python.exe C:/Users/arvind/frameworks/esp-idf-v4.4.1/components/partition_table/check_sizes.py --offset 0x8000 bootloader 0x0 C:/Users/arvind/temp_sensor_test/temp/build/bootloader/bootloader.bin"
Bootloader binary size 0x51a0 bytes. 0x2e60 bytes (36%) free.
[1048/1057] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/temp_sensor_main.c.obj
FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/temp_sensor_main.c.obj
ccache C:\Users\arvind\tools\xtensa-esp32s3-elf\esp-2021r2-patch3-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DHAVE_CONFIG_H -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -DWITH_POSIX -Iconfig -I../main -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/newlib/platform_include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/freertos/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/freertos/include/esp_additions/freertos -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/freertos/port/xtensa/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/freertos/include/esp_additions -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include/soc -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_hw_support/include/soc/esp32s3 -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_hw_support/port/esp32s3/. -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_hw_support/port/esp32s3/private_include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/heap/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/log/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/lwip/include/apps -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/lwip/include/apps/sntp -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/lwip/lwip/src/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/lwip/port/esp32/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/lwip/port/esp32/include/arch -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/soc/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/soc/esp32s3/. -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/soc/esp32s3/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/hal/esp32s3/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/hal/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/hal/platform_port/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_rom/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_rom/include/esp32s3 -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_rom/esp32s3 -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_common/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_system/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_system/port/soc -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_system/port/public_compat -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/xtensa/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/xtensa/esp32s3/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/driver/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/driver/esp32s3/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_pm/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_ringbuf/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/efuse/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/efuse/esp32s3/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/vfs/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_wifi/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_event/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_netif/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_eth/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/tcpip_adapter/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_phy/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_phy/esp32s3/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_ipc/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/app_trace/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_timer/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/mbedtls/port/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/mbedtls/mbedtls/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/mbedtls/esp_crt_bundle/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/app_update/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/spi_flash/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/bootloader_support/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/nvs_flash/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/pthread/include -IC:/Users/arvind/frameworks/esp-idf-v4.4.1/components/esp_gdbs [... cut ...]
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[1050/1057] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/manager.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1

C:\Users\arvind\temp_sensor_test\temp>
```

### More Information.

I'm trying to run an example code that's present on idf v5.0 on my toolchain which is v4.4 because the example file from v4.4 doesn't support esp32s3.
same error when setting the target to esp32-s2.

gmdriscoll, 2023-04-05:
Not sure if this is a bug or a general pathing issue, but have the same problem with any files that used to be in the framework-espidf/components/driver/include directory.   The header files all seem to be in the framework-espidf/components/driver/include/driver directory.  I can't figure out how to get the compiler to look in the driver/include/driver directory.

AirWind001, 2023-04-05:
Uninstall and clean reinstall fixes the issue

closed 2023-04-05 by AirWind001

Aecnoril, 2023-04-28:
I tried reinstalling esp-idf and it did nothing.
Some of my includes work, but some don't. Currently I'm stuck on the esp_tls.h even though the folder is right there next to other components that I cán include..

I can manually add the path to my cmakelists, but I'd have to do that manually for about a third of all the components.. It seems completely random which includes it will accept and which it will simply spit out

igrr, 2023-04-28:
@Aecnoril have you added esp_tls component to REQUIRES or PRIV_REQUIRES list in your CMakeLists.txt?