espressif/esp-idf #11135: I2S channel randomly reboots incorrectly

Pandre4, 2023-04-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

**IDF VERSION :**
idf-5.0.1

**Operating System used.**
Ubuntu 22.04.2 LTS

**How did you build your project?**
VS Code IDE

**Development Kit.**
Custom card with SOC ESP32-C3-WROOM

**What is the expected behavior?**
No difference in microphone values after a reboot 

**What is the actual behavior?**
When i restart my ESP (reset_button, via a command to set "esp_restart" or by switching off the power supply), the i2s channel randomly reboots incorrectly generating values with an unexplained gain.

**Steps to reproduce.**
On my card, there are two MEMS microphones ICS 43434 connected to the ESP32C3 via I2S.

The microphones work on 24 bits according to the Philips standard. 

```
/**
 * @brief I2S driver parameters
 * 
 */
#define I2S_NUM_PORT                     0
#define I2S_NUM_CHANNELS            2               // Stereo (2) vs Mono (1) - here Stereo
#define I2S_SAMPLE_RATE                 16000           // Hz - no more than 50kHz for the onboard microphones

#define I2S_SAMPLE_SIZE                 4               // bytes per sample - Philips format: 24 bits extended to 32 bits
#define I2S_BUFFER_COUNT                2               // number of buffers (>= 2)
#define I2S_BUFFER_LEN                  256             // samples per buffer (size is limited to 511 samples)

                                                        // next parameter is strictly bounded to 4096

#define I2S_DMA_BUFFER_SIZE             I2S_BUFFER_LEN*I2S_NUM_CHANNELS*I2S_SAMPLE_SIZE
#define I2S_RECV_BUFFER_SIZE            I2S_DMA_BUFFER_SIZE * I2S_BUFFER_COUNT/I2S_SAMPLE_SIZE + 64 //should be able to take all the data in all dma buffers,

/**
 * @brief I2S hardware interface
 * 
 */
#define I2S_SCK_GPIO                    GPIO_NUM_0
#define I2S_SD_GPIO                     GPIO_NUM_1
#define I2S_WS_GPIO                     GPIO_NUM_2

/**
 * @brief description of the i2s microphones and the i2s GPIO
 * 
 */

static void
i2s_driver_config(){

    i2s_chan_config_t rx_chan_cfg = {
        .id = I2S_NUM_0,                         /*!< I2S port id */
        .role = I2S_ROLE_MASTER,                 /*!< I2S role, I2S_ROLE_MASTER or I2S_ROLE_SLAVE */
        .dma_desc_num = I2S_BUFFER_COUNT,        /*!< I2S DMA buffer number, it is also the number of DMA descriptor */
        .dma_frame_num = I2S_BUFFER_LEN,         /*!< I2S frame number in one DMA buffer. One frame means one-time sample data in all slots,*/
        .auto_clear = true
    };

    ESP_ERROR_CHECK(i2s_new_channel(&rx_chan_cfg, NULL, &rx_chan));
  
    i2s_std_config_t i2s_std_config = {
        .clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(I2S_SAMPLE_RATE),
        .slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_32BIT,I2S_SLOT_MODE_STEREO),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,
            .bclk = I2S_SCK_GPIO,
            .ws = I2S_WS_GPIO,
            .dout = I2S_GPIO_UNUSED,
            .din = I2S_SD_GPIO,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };
    /* Initialize the channel */
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(rx_chan, &i2s_std_config));
	ESP_ERROR_CHECK(i2s_channel_enable(rx_chan));
}
```

I have many task in my projects, including an i2s read task to save sound signal in a buffer.
The listening task receives samples encoded on 32 bit from microphone, but representing 24-bit signals left-shifted, the value of the sample is then shifted to right by 8 bits.

_From ICS 43434 datasheet_
[image]

```
/* listening task */
void
listening_task(void *arg){
	ESP_LOGI("Listening task", "time since boot: %lld us", esp_timer_get_time());

	uint32_t sub_sampling_count = 0;
	uint32_t im;	

	forever: {
		tstpt_1_toggle();
		if (i2s_channel_read(rx_chan, i2s_read_buffer, I2S_DMA_BUFFER_SIZE , &listening.i2s_bytes_read, portMAX_DELAY)==ESP_OK){

			im = listening.i2s_bytes_read/I2S_SAMPLE_SIZE;

			for(int i=0; i < im; i+=2){
				/*** mean of the left and right microphones sound ***/
				listening.sound_prl_f32 = mic_raw2mPa((mic_decode(i2s_read_buffer[i]) + mic_decode(i2s_read_buffer[i+1]))/2);
				}
			}
		} else {
    		vTaskDelay(listening.task_delay);
    	}
    } goto forever;
}

static int32_t
mic_decode(int32_t raw){
    return (int)(raw >> 8);
}

static float 
mic_raw2mPa(int32_t raw){
    return microphone.mpa_full_scale*(float)raw/microphone.binary_full_scale;
}

float 
mic_mPa2dB(float mpa){
    return microphone.ref_point_db + 20.0*log10(mpa/(float)microphone.ref_point_mpa);
}
```

After a start-up, the i2s sometimes initializes incorrectly and it's totally random. If the initialization of the i2s went wrong the values received will always be with a gain as long as the ESP has not restarted. Depending to the versions of my project, this bug is more or less rare without any logic. 

In calm conditions I should have values between 4 and 6 mPa ( above 47 dB) but sometimes with bad i2s init, I have values between 12 and 14 mPa  (above 57dB).

I tried to add delays during the initialization phase, i tried to reduce and comment out a big part of my project but this bug persists.
The problem also appears on other ESP32-C3.

I don't know if the problem is with the i2s initialization or if I misread my values. 
This bug stay very strange for my.

I would like to know if there are any known problems with the initialization of i2s.

Thank you for you help.

L-KAYA, 2023-04-06:
Hi @Pandre4 , could you provide more details about the log when the soc crashed and how the read buffer allocated and used? This issue looks like some buffer might be overflowed.

And another questions is that, how you deal with the 24-bit width data in 32-bit aligned buffer?

Pandre4, 2023-04-06:
Hi @L-KAYA the soc doesn't crash, the i2s is simply in a bad state and i need to restart the soc and hope that the i2s initializes correctly.
I use a UART task to display the average sound level (mPa and dB) on the UART device every 2 seconds or more. But even without making the average my read sample listening.sound_prl_f32 sometimes has a gain.

When I use the i2s_channel_read function, the receive buffer is saved in `i2s_read_buffer`.
`static uint32_t i2s_read_buffer[I2S_RECV_BUFFER_SIZE];`

Then I read the raw value into the` i2s_read_buffer`, the microphone datasheet indicates that the 24-bit coded sample is sent on the 32-bit MSB side (left side). So I shift i2s_read_buffer[i] by 8 to the right via `mic_decode(int32_t raw)` to read a 24-bit sample from a 32-bit value. 

In fact, when the wrong initialization of the i2s appears, I have the impression that the right shift is no longer on 8 bit but on 6 bit, generating then a value 4 times too big.
I played a frequency: with the bug I had 800 mPa and without the bug I had only 200 mPa. So well a factor of 4 may be related to shift of 2 less

L-KAYA, 2023-04-06:
The `2 bits` shift is really weird, not likely to be the alignment issue. 

If you have a logic analyzer or an oscilloscope, you can monitor the data on line to see if the shift exists already. If you find the shift, I guess it might be the clock jitter whiling initializing I2S that make the Mic gets two bit clock slow. You can take a try to set the clock pins to a certain level first

Pandre4, 2023-04-06:
I read on an oscilloscope the SD line and the data is always good with or whitout the i2s bug. I read the SD line on an oscilloscope and the data is still good with or without the i2s bug. The 24-bit sample is always on the left with the 8 bits at 0 on the right.

L-KAYA, 2023-04-06:
> I read on an oscilloscope the SD line and the data is always good with or whitout the i2s bug. I read the SD line on an oscilloscope and the data is still good with or without the i2s bug. The 24-bit sample is always on the left with the 8 bits at 0 on the right.

Nice found! Then it can only be the issue inner the soc. Have you tried to adjust the `I2S_BUFFER_COUNT` and `I2S_BUFFER_LEN` ? These two parameters determine the DMA buffer size, which might the cause of the misalignment. You can increase the `I2S_BUFFER_COUNT` while decrease the `I2S_BUFFER_LEN` like: `I2S_BUFFER_COUNT=6` `I2S_BUFFER_COUNT=120`.

And you can try different `slot_bit` and `data_bit` as well:
```c
    i2s_std_config_t i2s_std_config = {
        .clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(I2S_SAMPLE_RATE),
        .slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_32BIT,I2S_SLOT_MODE_STEREO),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,
            .bclk = I2S_SCK_GPIO,
            .ws = I2S_WS_GPIO,
            .dout = I2S_GPIO_UNUSED,
            .din = I2S_SD_GPIO,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };
    i2s_std_config.slot_cfg.data_bit_width = I2S_DATA_BIT_WIDTH_24BIT;
    i2s_std_config.slot_cfg.slot_bit_width = I2S_SLOT_BIT_WIDTH_32BIT;
    /* Initialize the channel */
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(rx_chan, &i2s_std_config));
```

Pandre4, 2023-04-06:
I'm not sure I understand the difference beetwen I2S_DATA_BIT_WIDTH_24BIT and I2S_SLOT_BIT_WIDTH_32BIT. Because my microphones ICS 43434 use a 32 bit ws signal and not a 24 bit one even though the sample is a 24 bit one. 

[image]

I tried anyway, with `I2S_BUFFER_LEN` with a buffer size multiple of 3 and i get this error:
```
D (311) i2s_common: rx channel is registered on I2S0 successfully
D (311) i2s_common: DMA malloc info: dma_desc_num = 2, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 264
E (321) i2s_std: i2s_std_set_clock(65): The 'mclk_multiple' should be the multiple of 3 while using 24-bit data width
E (331) i2s_std: i2s_channel_init_std_mode(223): initialize channel failed while setting clock
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42007ed0
```
Followed by a core dump. The use of 24 bits may be incompatible with theses microphones.

Otherwise, I'm testing with different DMA buffer sizes, this can be time consuming to test as a bad i2s initialization can appear only 1 time 10.

L-KAYA, 2023-04-07:
Sorry that the code didn't work, I missed one more config:
```c
    i2s_std_config_t i2s_std_config = {
        .clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(I2S_SAMPLE_RATE),
        .slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_32BIT,I2S_SLOT_MODE_STEREO),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,
            .bclk = I2S_SCK_GPIO,
            .ws = I2S_WS_GPIO,
            .dout = I2S_GPIO_UNUSED,
            .din = I2S_SD_GPIO,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };
    /* Set data bit-width to 24 means your sample is 24 bits, I2S will help to shift the data by hardware */
    i2s_std_config.slot_cfg.data_bit_width = I2S_DATA_BIT_WIDTH_24BIT;
    /* Set slot bit-width to 32 means there still will be 32 bit in one slot */
    i2s_std_config.slot_cfg.slot_bit_width = I2S_SLOT_BIT_WIDTH_32BIT;
    /* Set the mclk_multiple to a multiple of 3, so that it can calculate an integer division to a 24-bit data */
    i2s_std_config.clk_cfg.mclk_multiple = 384;
    /* Initialize the channel */
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(rx_chan, &i2s_std_config));
```
By setting the configurations above, you don't need to shift the data manually, the hardware will do it for you while receiving. The data received in the buffer will aligned with 3 bytes (i.e. 24 bits). And take care that you may need a buffer that size to a multiple of 3 in this case. Hope it can work.

And in your log, the `dma_desc_buf_size` seems not to be a correct value:
> D (311) i2s_common: DMA malloc info: dma_desc_num = 2, dma_desc_buf_size = dma_frame_num * slot_num * data_bit_width = 264
Could you try to set the default dma setting:
```c
i2s_chan_config_t rx_chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_0, I2S_ROLE_MASTER);
rx_chan_cfg.auto_clear = true;
ESP_ERROR_CHECK(i2s_new_channel(&rx_chan_cfg, NULL, &rx_chan));
```

Pandre4, 2023-04-07:
In this case for the reading, how big should my reception buffer (i2s_read_buffer) be?
`i2s_channel_read(rx_chan, i2s_read_buffer, I2S_DMA_BUFFER_SIZE, &listening.i2s_bytes_read,portMAX_DELAY)`

Because currently, I define this buffer as follows: `uint32_t i2s_read_buffer[I2S_RECV_BUFFER_SIZE];`

If the data is 3-bytes aligned, I need to define a uint8_t for i2s_read_buffer and concatenate 3 uint8_t to form a uint32_t to process my sample, right?

Pandre4, 2023-04-07:
I retested my complete program with this configuration :

- always with 32 bits sample with 8 bits right shift
- dma_desc_num =  120
- dma_frame_num  = 6

And after many restart after several dozen restarts, the i2s initialization start well, the bug never came back.
But if I add a void function which do nothing in my main task (the main task is just a setup task) or if I just add a simple vTaskDelay the wrong i2s init may re-appear.

I tried to increase the size of the main task in the menuconfig, but it didn't change anything. 

I really don't understand, it gives me the impression that i2s start well if the program checksum is only even or something like that. Why does my program work or not work with one less or one more line of C?

L-KAYA, 2023-04-11:
Sounds really weird. Could you register a `on_recv_q_ovf` callback by `i2s_channel_register_event_callback` to see if there are any data dropped?

And could you confirm one more thing, as you find the data is correct on line, can we monitor the output clocks like BCLK and WS as well? I want to make sure that there is no clock pulse missing and the clocks (especially WS) is aligned with the data in philips format.

Pandre4, 2023-04-11:
I create an `on_recv_q_ovf` callback by `i2s_channel_register_event_callback`. During the initialization, this event is called only 125 or 126 times depending on the start-up and this event is never called back later. When the i2s bug appears, it has no influence on the on the callback, it is always called 125 or 126 times.

**_With bad i2s init:_**
_WS with DATA:_
[image]
_WS with BCLK:_
[image]

**_With good i2s init_:**
_WS with DATA:_
[image]
_WS with BCLK:_
[image]

There are absolutely no difference beetwen these signals. I sample my sound signal at 16 kHz on 2 channels. So the WS signal is 31.25 ms at the high and low states and BCLK is 1/32 ms. I see nothing strange in these signals.

L-KAYA, 2023-04-12:
I found the `bclk` seem to be slightly delayed at the bad stage (comparing with the edge of `ws` signal), the figures remind me that the sample edge of bclk can affect the final result as well, there are some tips might work:

1. Try to adjust the bclk

Invert the `bclk` by setting `bclk_invert` flag, it may produce a fixed data shift, but as we do shifting for each slot, it should not matter. 

2. Try to reduce the `bclk` jitter

Sometimes the jitter might occur while setting to a low sample rate, you can take a try to set the sample rate to a relative high requency like `44100 Hz` then switch back to the target sample rate `16000 Hz`, (i.e. set the clock config continuously, no need to enable the channel before setting back to the target sample rate).

Pandre4, 2023-04-13:
I tried to fix` bclk_invert` to` true` with changing data shift from 8 to 9, it's also work, but the bug continues to randomly appear. 

And i tried to do something like that to change the sample rate, it's doesn't help :

```
static void
i2s_driver_config(){

    i2s_chan_config_t rx_chan_cfg = {
        .id = I2S_NUM_0,                         /*!< I2S port id */
        .role = I2S_ROLE_MASTER,                 /*!< I2S role, I2S_ROLE_MASTER or I2S_ROLE_SLAVE */
        .dma_desc_num = I2S_BUFFER_COUNT,        /*!< I2S DMA buffer number, it is also the number of DMA descriptor */
        .dma_frame_num = I2S_BUFFER_LEN,         /*!< I2S frame number in one DMA buffer. One frame means one-time sample data in all slots,*/
        .auto_clear = true,
    };

    ESP_ERROR_CHECK(i2s_new_channel(&rx_chan_cfg, NULL, &rx_chan));
    
    i2s_std_config_t i2s_std_config = {
        .clk_cfg = I2S_STD_CLK_DEFAULT_CONFIG(44100),
        .slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_32BIT,I2S_SLOT_MODE_STEREO),
        .gpio_cfg = {
            .mclk = I2S_GPIO_UNUSED,
            .bclk = I2S_SCK_GPIO,
            .ws = I2S_WS_GPIO,
            .dout = I2S_GPIO_UNUSED,
            .din = I2S_SD_GPIO,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };
   
    i2s_std_config.clk_cfg.sample_rate_hz = 16000;

    /* Initialize the channel */
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(rx_chan, &i2s_std_config));
    ESP_ERROR_CHECK(i2s_channel_enable(rx_chan));
}
```
Furthermore, working on a different sample rate does not seem to correct the problem. Although there may be an ideal frequency to use.

L-KAYA, 2023-04-13:
One more configure is worth to take a try:
enable the left alignment by 
`i2s_std_config .slot_cfg.left_align = true;`

Still, I don't think it would be the DMA issue, but to get rid of the possibility totally, you can also register a `on_recv` callback and use `esp_rom_printf` to print the DMA buffer directly in event data, like:
```c
#include <inttypes.h>
static boo i2s_on_receive_callback(i2s_chan_handle_t handle, i2s_event_data_t *event, void *user_ctx)
{
    uint32_t *recv_buf = event->data;
    for (int i = 0; i < event->size / sizeof(uint32_t); i++) {
        esp_rom_printf("%"PRIu32" ", recv_buf[i]);
    }
    return false;
}
i2s_event_callbacks_t cbs = {
        .on_recv = i2s_on_receive_callback,
        .on_recv_q_ovf = NULL,
        .on_sent = NULL,
        .on_send_q_ovf = NULL,
    };
    TEST_ESP_OK(i2s_channel_register_event_callback(tx_chan, &cbs, NULL));
```

Pandre4, 2023-04-14:
I can read only the first DMA buffers before i get an` Interrupt wdt timeout on CPU0`, but it's pretty normal.

`1070277872 1070279796 1920 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 4293034496 261632 4292982784 211968 4292963840 189440 4292919808 91648 4292789760 4294868992 4292525056 4294683136 4292332032 4294597632 4292252160 4294392320 4292053504 4294195712 4291649024 4293960192 4291429888 4293602304 4290989568 4293391360 4290904576 4293103616 4290543104 4292755968 4290139136 4292726272 4290121728 4292640256 4290034176 4292436480 4289768448 4292162560 4289496064 4291840512 4289067520 4291672576 4289033728 4291917312 4289091584 4291797504 4289035264 4291935232 4289112064 4292107776 4289262592 4292170752 4289323008 4292454912 4289590272 4292531712 4289643520 4292809728 4289907200 4292897280 4289985536 4292976640 4290018304 4292873728 4289989632 4292975104 4290017792 4293021184 4290049024 4293584896 4288276480 4293667840 4288233984 4293427200 4288015872 4293310464 4287941632 4293326848 4287994880 4293174784 4287828480 4293317632 4287968256 4293224448 4287830528 4293042176 4287761408 4293050368 4287622656 4292898816 4287538688 4292722688 4287453184 4292939776 4287608320 4292935168 4287615488 4293051904 4287697920 4293039104 4287676416 4293059584 4287730688 4292978176 4287780864 4292976640 4287778304 4293021696 4287678976 4293203968 4287825408 4293148160 4287890432 4293237760 4287870976 4293142528 4287801856 4293076992 4287782400 4292995072 4287823360 4293158912 4287848448 4293091840 4287810048 4293115392 4287787008 4292915200 4287728128 4292843008 4287575552 4292789248 4287495680 4292724736 4292734464 4293757952 4292729344 4294002176 4292820480 4294038016 4292901376 4294158848 4293090816 4294235648 4293198336 4294064128 4292965376 4294225920 4293071360 4294336512 4293195264 4294198784 4293126144 4294067712 4293041664 4294142976 4293219328 4294408704 4293393408 4294453760 4293530624 4294541312 4293573120 4294414336 4293560320 4294628352 4293626368 4294762496 4293887488 4294661120 4293817856 4294579712 4293652480 4294647808 4293608960 4294480896 4293601280 4294494208 1070281752 1070283676 1920 4293822464 4292985856 4293806080 4293009408 4293834240 4293050368 4293804544 4292978688 4293816320 4293028352 4293979136 4293186048 4293872640 4293091840 4293664768 4292924928 4293849600 4293124096 4293887488 4293199872 4293788160 4292958208 4293878272 4293138944 4294038016 4293217792 4294118912 4293207040 4294236160 4293445120 4294246400 4293386752 4294178304 4293349888 4293565440 4293201408 4293489152 4292996096 4293156864 4292659712 4292880896 4292336128 4292341248 4291795968 4291873280 4291456512 4291814400 4291239936 4291671040 4291262464 4291626496 4291151360 4291635200 4291024896 4291771392 4291270656 4291607040 4291226624 4291627008 4291214848 4291625472 4291134976 4291935232 4291344384 4292296704 4291794432 4292521984 4292013056 4292769280 4292283392 4293128704 4292572672 4293466624 4293079040 4294014464 4293459456 4294390784 4293884928 4294696448 4294154240 4294912000 4294401024 430080 4294920192 771072 235008 1097216 663552 1534976 1058304 1737216 1338880 2093568 1660416 2002944 1732608 2019328 1644544 1854464 1514496 1594368 1215488 1354752 926208 1295872 811008 1397760 1400832 1055232 1095680 708608 771072 330752 395776 4294957056 4294906880 4294587392 4294640128 4294290432 4294305280 4293911552 4293797376 4293565440 4293430272 4293291008 4293210112 4293027328 4292919296 4292570112 4292516864 4292086784 4292063232 4291790336 4291620352 4291350528 4291302400 4291171840 4291129344 4290874368 4290704384 4290469376 4290494976 4290267648 4290230272 4290300928 4290297344 4290258944 4290285056 4290190848 4290171904 4290227712 4290114048 4290441216 4290379264 4290756096 4290605568 4291067392 4291001344 4291486208 4291372544 4291872768 4291765760 4292277760 4292097024 4292773376 4292674560 4293245440 4293155328 4293717504 4293547008 4294255104 4294031872 6656 4294651904 4294414336 4294374400 4294538752 4294466048 4294662656 4294664192 4294884864 4294691328 25088 19456 228864 139264 561664 493056 857600 766464 1018880 1048576 1302016 1220608 1445376 1411072 1585152 1535488 1712640 1738752 1808384 1905664 1966592 1939456 2072064 2128384 2037248 2196480 2255360 2265600 2469376 2472448 2475520 2523648 2544640 2582016 2511872 2383360 2212864 2205696 2012160 2017792 1831936 1746432 1701888 1644544 1454080 1522176 1296384 1346560 1167360 1163264 931328 1061376 905216 876032 540160 567296 490496 486400 455168 459776 456192 357376 372736 471552 362496 323584 349184 324096 390144 273920 315904 354304 471552 416768 367616 365056 535040 428032 607232 476672 636416 585216 675328 603648 543232 5120 4294800896 208384 49152 329728 132096 458240 27136 391168 263168 603648 224768 664576 207360 623104 200704 583680 347648 795648 332800 732160 186880 650240 392704 724992 379392 754176 328192 657408 294400 705536 129024 444928 4294782976 143872 4294721536 117248 4294695936 42949 [... cut ...]
Here i used these DMA buffer values :
`#define I2S_BUFFER_COUNT                4               
#define I2S_BUFFER_LEN                  240    `

I don't know if there is a problem in these values, but if I print 1024 samples (ok, there are less samples displayed) with a sine wave sound played. There is no cut-off. In the signal processing after the acquisition I have no problem of data loss just a gain when the i2s bug appears.

[image]

And this line : `i2s_std_config .slot_cfg.left_align = true;`, does not change anything.

Pandre4, 2023-04-14:
I retried to print only the first three buffer in playing a sine wave sound and I get these values :

_With I2S Bug:_
1070277720,1070279644,1920,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,0,93843456,77432832,113917952,96499712,129863680,112336896,142778368,125247488,151724032,135041024,152260608,136638464,146706432,133066752,133722112,120612864,113481728,101605376,89673728,79077376,62810112,52787200,33968128,24270848,4194307,4290961411,4271142915,4263133187,4243138563,4235767811,4218087427,4211488771,4195557379,4189261827,4173176835,4167385091,4156176387,4150304771,4141541379,4135776259,4129572867,4123834371,4122273795,4116400131,4117334019,4110340099,4118751235,4110258179,4125325315,4116508675,4136712195,4127612931,4154576899,4143826947,4175970307,4164382723,4199901187,4187025411,4224804867,4211003395,4248303619,4233791491,4270903299,4255365123,4291964931,4275886080,19251200,2093056,40804352,33808384,24723456,8837123,4293830659,4278841347,4268058627,4253978627,4242839555,4229904387,4218689539,4206553091,4195620867,4184170499,4174929923,4164415491,4157067267,4146972675,4141701123,4131158019,4130752515,4120434691,4124252163,4113131523,4122488835,4110379011,4126803971,4113905667,4135413763,4122169347,4149983235,4135237635,4169547779,4153624579,4192249859,4175456259,4217249795,4199286787,4242921475,4223870979,4267751427,4248057859,4292337667,1070281600,1070283524,1920,4149690371,4144023555,4135393283,4129458179,4126337027,4120137731,4119611395,4112635907,4118261763,4110981123,4124069891,4115976195,4133474307,4124602371,4147890179,4138141699,4167907331,4156295171,4191295491,4178307075,4216143875,4202133507,4242284547,4226705411,4267638787,4251473923,4292087811,4274976768,21989376,4020224,45631488,27043840,68263936,48816128,88064000,8460291,4276834307,4276178947,4250802179,4250892291,4228192259,4229351427,4207355907,4209410051,4187727875,4190392323,4172339203,4174557187,4158058499,4160454659,4147689475,4150372355,4141283331,4143472643,4138762243,4139968515,4139978755,4140703747,4146937859,4146767875,4158337027,4156821507,4175220739,4173000707,4197210115,4194086915,4221315075,4216250371,4246085635,4240267267,4271529987,4264820739,4294365187,4286865408,22675456,13881344,44050432,34822144,64872448,54644736,85305344,74735616,103708672,92354560,122673152,110948352,139137024,126175232,151109632,138622976,160206848,148312064,162342912,151250944,158160896,148617216,146968576,139048960,127606784,121155584,104951808,99348480,78895104,74434560,50276352,4144478211,4147828739,4153249795,4164257795,4168259587,4183891971,4186908675,4207392771,4209268739,4232443907,4232921091,4258177027,4258004995,4283848707,4282607616,12683264,10551296,36782080,32894976,58707968,54339584,80271360,74754048,99112960,93335552,118228992,112214016,136665088,129751040,150919168,144136192,162877440,156192768,168798208,162496512,165789696,161038336,156473344,153440256,139147264,137261056,116066304,115820544,90290176,90925056,62160896,64421888,34174976,37115904,6383616,10579971,4274331651,4280174595,4250015747,4256299011,4227385347,4234473475,4205627395,4213514243,4185165827,4193871875,4169097219,4177862659,4153643011,4162467843,4143448067,4151885827,4137334787,4145928195,4134277123,82466816,110489600,101748736,131239936,122601472,149407744,139780096,163121152,153907200,172738560,163799040,173557760,165834752,166656000,161089536,151330816,147202048,129126400,126369792,104165376,102303744,76748800,75933696,47620096,47706112,18153472,20015107,4284368899,4287346691,4256284675,4260767747,4230873091,4236238851,4208248835,4213985283,4185964547,4192616451,4168630275,4175321091,4153370627,4160503811,4141967363,4148873219,4135608323,4142370819,4132186115,4137988099,4134246403,4139386883,4142432259,4146683907,4153698307,4157374467,4170809347,4172965891,4191811587,4192563203,4214351875,4213895171,4238268419,4236865539,4263393283,4260605955,4287092739,4283541504,15646720,10790912,38318080,91545600,61229056,62396416,31299584,33978368,3407872,7055363,4271257603,4275771395,4245995523,4251924483,4222107651,4228620291,4198318083,4205903875,4177840131,4186216451,4159498243,4167579651,4144082947,4152072195,4134170627,4143042563,4129204227,4136591363,4130396163,4137369603,4138371075,4144435203,4150091779,4155379715,4165992451,4169400323,4186664963,4188706819,4209375235,4210483203,4232491011,4232923139,4256829443,4255768579,4280330243,4278114304,9469952,6117376,34660352,29730816,58791936,52965376,82710528,76044288,103362560,95660032,122081280,114274304,141322240,132855808,156049408,147265536,168988672,160014336,177158144,168744960,177901568,170299392,170149888,164579328,153704448,149354496,130588672,4147345411,4136949763,4143556611,4139829251,4145655811,4146331651,4151678979,4157765635,4162033667,4175728643,4178638851,4197324803,4198973443,4222373891,4222941187,4248932355,4248363011,427499 [... cut ...] 4288924160,4289996288,4282601984,4283920384,4277083136,4278595584,4271529472,4273222656,4266805760,4268667904,4263158784,4264927232,4259631616,4261484032,4257610240,4259367936,4256407552,4258175488,4256274432,4257844736,4257984000,4259291136,33094656
`
And I made this plot from python :
[image]

The red signal is with the i2s bug and the blue signal is without the i2s bug. The sound played is the same with the same amplitude.
This "magical gain" (if I may say so) is clearly visible on the graph.

Pandre4, 2023-04-17:
Ok thanks, but your .zip file is empty.

L-KAYA, 2023-04-17:
Sorry, I re-uploaded

[i2s_IDFGH-9799.zip](https://github.com/espressif/esp-idf/files/11247776/i2s_IDFGH-9799.zip)

Pandre4, 2023-04-17:
Can you repload it please ? The main folder is empty :\

L-KAYA, 2023-04-17:
> Can you repload it please ? The main folder is empty :\

Done

Pandre4, 2023-04-18:
I flashed your code with twos ESP32C3 (slave and master) and it's work well. I find the right values.

However, I tried to reproduce the I2S bug. So in the menuconfig I have disabled this MPI option, it's probably not recommended, but it seems pretty good to reproduce the bad i2s initialization:
[image]

And randomly after reboot I get these values :
[image]

Pandre4, 2023-04-20:
I may have found something. I was working with ESP-IDF v5.0.1, now I'm trying to work with the master branch and the I2S init bug doesn't seem to reappear. I'll keep testing in the next few days, but it's pretty positive.

L-KAYA, 2023-04-23:
Hi @Pandre4 , how is it going? Does it work stably on master?

Pandre4, 2023-04-24:
Hi @L-KAYA, I don't kow how and why, but yes, if I build and flash my project with ESP-IDF master, it's work very well without the I2S problem.

L-KAYA, 2023-04-24:
Glad to hear that, I'm going to close this issue, if you have any other problems, welcome to reopen this issue again or create another issue, thanks!

closed 2023-04-24 by L-KAYA