espressif/esp-idf #11139: sdkconfig.defaults can't overwirte

myfreax, 2023-04-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-96-gdab3f38f0f

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

sdkconfig.defaults can't overwirte default option

### What is the actual behavior?

sdkconfig.defaults can overwirte default option

### Steps to reproduce.

1. Checkout idf 4.4 and navigate to `esp-idf/examples/system/unit_test/test`
2. Change `CONFIG_ESP_TASK_WDT` eq `y`
3. Run command `idf.py build`
4. Run comand `cat sdkconfig | grep CONFIG_ESP_TASK_WDT`. It show `CONFIG_ESP_TASK_WDT is not set`

### Build or installation Logs.

```plain
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/myfreax/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/myfreax/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /home/myfreax/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/myfreax/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /home/myfreax/.espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: /var/sdk/esp32/idf/esp-idf/components/freemodbus.
-- Project sdkconfig file /var/sdk/esp32/idf/esp-idf/examples/system/unit_test/test/sdkconfig
Loading defaults file /var/sdk/esp32/idf/esp-idf/examples/system/unit_test/test/sdkconfig.defaults...
-- Found Python3: /home/myfreax/.espressif/python_env/idf4.4_py3.8_env/bin/python3.8 (found version "3.8.10") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "unit_test_test" version: v4.4.4-96-gdab3f38f0f-dirty
-- Adding linker script /var/sdk/esp32/idf/esp-idf/examples/system/unit_test/test/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /var/sdk/esp32/idf/esp-idf/components/esp_system/ld/esp32s3/sections.ld.in
-- Adding linker script /var/sdk/esp32/idf/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
[... 9 lines cut ...]
-- Test components: testable::test
-- Test component paths: /var/sdk/esp32/idf/esp-idf/examples/system/unit_test/components/testable/test
-- Configuring done
-- Generating done
-- Build files have been written to: /var/sdk/esp32/idf/esp-idf/examples/system/unit_test/test/build
[1/1079] Generating memory.ld linker script...
[2/1079] Generating project_elf_src_esp32s3.c
[3/1079] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/ecjpake.c.obj
[4/1079] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedtls.dir/debug.c.obj
[5/1079] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1M,
*******************************************************************************
[6/1079] Building C object esp-
.............
Project build complete. To flash, run this command:
/home/myfreax/.espressif/python_env/idf4.4_py3.8_env/bin/python ../../../../components/esptool_py/esptool/esptool.py -p (PORT) -b 460800 --before default_reset --after hard_reset --chip esp32s3  write_flash --flash_mode dio --flash_size detect --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/unit_test_test.bin
or run 'idf.py -p (PORT) flash'
```
```

dobairoland, 2023-04-06:
> 1. Checkout idf 4.4 and navigate to esp-idf/examples/system/unit_test/test
> 2. Change CONFIG_ESP_TASK_WDT eq y

Does `sdkconfig` exist at this moment in `esp-idf/examples/system/unit_test/test`? The content of `sdkconfig.defaults` is considered only if there is no `sdkconfig` at the moment you run `idf.py build`.

This is also explained here: https://docs.espressif.com/projects/esp-idf/en/v4.4.4/esp32/api-guides/build-system.html#custom-sdkconfig-defaults

> This file will be used when creating a new config from scratch, or when any new config value hasn’t yet been set in the sdkconfig file.

myfreax, 2023-04-07:
@dobairoland Thank your help，It work well.

closed 2023-04-07 by myfreax