espressif/esp-idf #11140: SPI Slave send and receive example only works after I ground both boards together

danielyang2000, 2023-04-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

VScode ESP-IDF extension V5.0.1
receiver and sender board: ESP32-S3-WROOM2

Hello,

I have a question about the SPI slave send and receive example. When the boards are not grounded together, the receiver and the sender ESP32 boards print out garbage. It was only after I grounded them together that both boards started to receive and print out real messages. What is the reason for this?

Should this be posted in the SPI Slave example README so that other people know about both boards need to be grounded?

Thank you,
Daniel

wanckl, 2023-04-10:
@danielyang2000 Hello, this is understandable IMO
As SPI protocol without ack signal, they don't know if others is ready, no matter master and slave, if not grounded together, it is theoretically right they receive garbage as receive pin is floating

Alvin1Zhang, 2023-08-07:
@danielyang2000 Thanks for reporting, would you please help share if any updates for the issue? Thanks.

wanckl, 2023-08-07:
@danielyang2000 In `examples/peripherals/spi_slave` folder, there is a `sender` and a `receiver` sub folder:   
- sender: Actually run as SPI master, send data to slave
- receiver: Actually run as SPI slave, receive data from master   

They are just a whole communication example using SPI drivers. other examples like `peripherals/spi_master`, `peripherals/spi_slave_hd/append_mode`, `peripherals/spi_slave_hd/seg_mod` are all similarly a whole communication, they are used together / respectively.    

For slave example/receive, as it work as a slave, they are input and waiting for select from master, if you disconnect it each other, the noise on floating pin may delusion slave as selected, NOISE, NOISE !!

wanckl, 2025-01-06:
@danielyang2000 hello, any question else ?

closed 2025-03-06 by Alvin1Zhang