espressif/esp-idf #11141: USB HID example howto enter suspend mode

mmar22, 2023-04-06:
### Is your feature request related to a problem?

Example now work with mouse and keyboards, but dont show way , howto send device into suspend mode.
Perfect too show deep sleep enter and exit.

### Describe the solution you'd like.

In design i have mouse IC, that lower power consumtion in suspend mode. I require enter it before deep sleep.
I try search in headers , but no info.
Good is add suspend resume into example.

Dazza0, 2023-04-11:
@mmar22 Just wanted to clarify a few things and add some comments

> Example now work with mouse and keyboards, but dont show way , howto send device into suspend mode.

We have a suspend/resume command expose at the [HCD level](https://github.com/espressif/esp-idf/blob/v5.0.1/components/usb/private_include/hcd.h#L98), but we haven't exposed this to the higher layers yet due to the lack of testing.

Do you also require remote wakeup funcitonality?

> In design i have mouse IC, that lower power consumtion in suspend mode. I require enter it before deep sleep.

Deep and light sleep probably won't be compatible with USB Host operation, as putting the ESP into deep/light sleep will power-off/clock-gate the USB peripheral.

mmar22, 2023-04-11:
Hi Darian,
ok not exposed yet, but have you an example test command, for test enter suspend from host to device.?
And yes deep sleep is only next step after putting device into suspend. (Wake up reinit all)
My idea wait to usb claim mouse IC and send suspend.
MM

From: Darian 
Sent: Tuesday, April 11, 2023 10:25 AM
To: espressif/esp-idf 
Cc: mmar22 ; Mention 
Subject: Re: [espressif/esp-idf] USB HID example howto enter suspend mode (IDFGH-9806) (Issue #11141)

@mmar22 Just wanted to clarify a few things and add some comments

  Example now work with mouse and keyboards, but dont show way , howto send device into suspend mode.

We have a suspend/resume command expose at the HCD level, but we haven't exposed this to the higher layers yet due to the lack of testing.

Do you also require remote wakeup funcitonality?

  In design i have mouse IC, that lower power consumtion in suspend mode. I require enter it before deep sleep.

Deep and light sleep probably won't be compatible with USB Host operation, as putting the ESP into deep/light sleep will power-off/clock-gate the USB peripheral.

closed 2026-02-16 by tore-espressif