espressif/esp-idf #11142: in IDF 4.4.3,How to deinit blufi in Nimble mode?

casitelwithdean, 2023-04-06:
### Is your feature request related to a problem?

Hi, I am using esp32s3wroom1  ,and using blufi in nimble mode,but I couldn't find a API to deinit blufi.

By the way ,i see IDF5.0 add the deinit API of nimble mode,but I can't  find those APIS  in IDF4.4.3
However ,bluedroid does have ,but for some reason ,i need to use nimble mode in IDF4.4.3.
So i wanna ask if there is a easy way to deinit blufi in nimble mode?
Thanks a lot!

rahult-github, 2023-04-07:
Hi @casitelwithdean  , 

Please check if attached patch helps. This should be applied at $IDF_PATH . 

[0001-Nimble-blufi-Added-an-API-to-stop-Bluetooth-Blufi.txt](https://github.com/espressif/esp-idf/files/11176018/0001-Nimble-blufi-Added-an-API-to-stop-Bluetooth-Blufi.txt)

rahult-github, 2023-08-07:
Closing issue . Please feel free to reopen if the patch is not helping.

closed 2023-08-07 by rahult-github