espressif/esp-idf #11143: SPI Slave on ESP32-S3-WROOM-1 receive wrong data on GPIO13

mc-ff, 2023-04-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF 5.0.1
SPI Slave example on ESP32S3 module  connected with a custom SPI Master.
SPI Slave bus on SPI2_HOST with signals on IOMUX pins (as in the example). Here is my buscfg :

```
spi_bus_config_t buscfg={
        .mosi_io_num=GPIO_MOSI,
        .miso_io_num=GPIO_MISO,
        .sclk_io_num=GPIO_SCLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };
```

and my slvcfg :

```
    spi_slave_interface_config_t slvcfg={
        .mode=0,
        .spics_io_num=GPIO_CS,
        .queue_size=3,
        .flags=0,
        //.post_setup_cb=my_post_setup_cb,
       // .post_trans_cb=my_post_trans_cb
        .post_setup_cb=NULL,
        .post_trans_cb=NULL
    };
```

At every transaction, my code prints out the bytes received. If MISO pin is connected to ESP32 i cannot receive the correct packet :

_Received: 37 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5_

but if i remove the MISO pin connection the RX packet is correct :

_Received: A6 FA 55 E0 5F E0 60 00 00 08 04 00 02 01 00 00 00 00 00 00 6E 30 80 00 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5 A5?_

Same strange behavior if use SPI3_HOST on the same GPIO pins.

wanckl, 2023-04-21:
@mc-ff Hello
Firstly, MISO pin is output for slave, if it affect slave to receive data, did you try to use other different pins, or this issue only appears on GPIO13 ?

Secondly, could you check at your master side like pin config(pull up or down, IO current capability), spi clock speed, hardware connection. if passable, could you use logic analyzer or scope to see if data is correct on spi bus, with and without MISO connect.

Because I can't reproduce your issue on two ESP32S3 board (master and slave) due to your description:   
based on spi slave example:
```
SPI2_HOST
iomux_pins
use_dma
clock_freq=1000000
with and without MISO pin
```

mc-ff, 2023-04-21:
@wanckl thanks for your support.
Before writing i tried moving the MISO to a different pin with no success. 
What i have noticed is that the SPI works fine if :

- MISO pin is disabled by setting .miso_io_num=-1
- MISO pin is enabled to GPIO13 and disconnected from the master board. Checking the MISO signal with the oscilloscope shows that the ESP32S3 sends the correct packet on MISO .

Connecting the MISO pin to the other board leads to two problems:

- the data received from the SPI Slave is corrupted, even though  the MOSI,SCK and CS signals are correct.
- MISO signals are not drived well by the SPI SLAVE (wrong packet payload)

Inserting a 100 ohm series resistance on the MISO output resolve the problem. This is weird because :

- each SPI signal is translated to the correct voiltage level by an SN74LV1T34DBVR (the SPI Master uses 5V levels)
- On the Master side, the SPI signals is directly connected to the master chip.
- The circuit is the same as the old IOT board which was equipped with an ESP32-WROOM-1
- the clock rate is very slow (19200 baud rate). A transaction occurs every 50 ms and is 24 bytes long.

Finally i noticed that the ESP32S3 MISO signal level is set on the first rising edge of the clock, but in mode 0 i think the signal needs to be changed on the falling edge of the clock.

i

wanckl, 2023-04-23:
@mc-ff  Sorry, I'm confused:
> ESP32S3 MISO signal level is set on the first rising edge of the clock   

So is you mean `ESP32S3 MISO signal incorrect on first rising edge` and connect to master, then lead to your slave can't receive correct data ? or master can't receive correct data ?

Ans is only first bit level set on the first rising edge? or every bit is set on rising edge? you are right that *in mode 0 signal needs to be changed on the falling edge*.

As you can see correct data on MISO when it disconnected, so if it just first bit is incorrect, your master should receive remain correct data.

For the first bit, please check when your CS line is active, you should remain at least 1 clock time between CS active and first clock rising edge for slave to setup data on MISO pin.

And as your data wrong when MISO is connected, would you compare what different before and after MISO connected the signal is by your scope, if possible, you can past graph here.

Thanks, waiting for your reply.

mc-ff, 2023-04-27:
@wanckl , i apologize for my unclear explanation. I add some images from my oscilloscope to be more clear :

First case : 
- MISO disconnected, MOSI + CS + SCK connected.
- SPI slave mode 0, SPI2_HOST with all signal configured : MISO->IO13, MOSI->IO11, CS->IO10, SCK->IO12 (using IOMUX or Matrix does not have effects on this issue)
- 100 ohm series resistance on MISO.
- SPI Slave sends 0xA5 continuosly, must receive a5 f4 4b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 13 80 00 00

Everything is ok as you can see from this screenshot of my oscilloscope (CS = Yellow, SCK = Green, MOSI = Purple, MISO = blue) :
[image]

Second case : 
- MISO + MOSI + CS + SCK connected.
- SPI slave mode 0, SPI2_HOST with all signal configured : MISO->IO13, MOSI->IO11, CS->IO10, SCK->IO12 (using IOMUX or Matrix does not have effects on this issue)
- 100 ohm series resistance on MISO.
- SPI Slave sends 0xA5 continuosly, must receive a5 f4 4b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 13 80 00 00

SPI Slave receives wrong data (b8 ef fe e2 e3 be 00 02 00 02 00 02 00 02 00 02 00 02 00 02 00 02 00 00 00 00) and MISO signal are different from A5 (CS = Yellow, SCK = Green, MOSI = Purple, MISO = blue) :
[image]

Third case : 
- MISO + MOSI + CS + SCK connected.
- SPI slave mode 0, SPI2_HOST with all signal configured : MISO->IO13, MOSI->IO11, CS->IO10, SCK->IO12 (using IOMUX or Matrix does not have effects on this issue). Force hw->slave.rsck_data_out = 1
- 100 ohm series resistance on MISO.
- SPI Slave sends 0xA5 continuosly, must receive a5 f4 4b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 13 80 00 00

Again, everything is ok, SPI Slave receives correct data and sends out the correct output.

Fourth case : 
- MISO + MOSI + CS + SCK connected.
- SPI slave mode 0, SPI2_HOST with all signal configured : MISO->IO13, MOSI->IO11, CS->IO10, SCK->IO12 (using IOMUX or Matrix does not have effects on this issue). Force hw->slave.rsck_data_out = 1
- NO100 ohm series resistance on MISO.
- SPI Slave sends 0xA5 continuosly, must receive a5 f4 4b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 79 13 80 00 00

SPI Slave receives wrong data (d3 7f d1 5f 03 1d 7f ff c0 00 00 00 00 00 00 00 00 00 00 00 00 0f 34 00 00) and MISO signal are different from A5 (CS = Yellow, SCK = Green, MOSI = Purple, MISO = blue) :
[image]

> For the first bit, please check when your CS line is active, you should remain at least 1 clock time between CS active and first clock rising edge for slave to setup data on MISO pin.

Checking with the oscilloscope, distance between CS and first SCK rising edge is about one SCK clock.

Thank you for your time

wanckl, 2023-04-28:
@mc-ff  Hi, thank you for your hard and detailed report.
IMO, I think maybe a hardware noise issue, especially between your MISO and MOSI pin.

As you can see correct data when MISO disconnected or not configured, or even a serial resister connected, these all means there are no drive, or only week drive on MISO line. and peripheral is OK.

From your picture 3th, we can find MISO maybe high level follow MOSI line.

maybe you can have a try that: MOSI disconnect, MISO + SCK + CS connect. and check your hardware connection, and every `SN74LV1T34DBVR` chip. if possible, try on another board.

Sorry for not helping you, waiting for your reply, good luck!

S0urceror, 2024-04-07:
Update, got it working.

I noticed that adding a 220 Ohm resistor to the MISO line influenced the MOSI line in between bytes. Then I added a similar 220 Ohm resistor to the MOSI line. Et voila! It works.

I have now stable communication from master to slave and from slave to master at 2,25 Mhz clock. So you have to load them symmetrically.

Don't know what it is. According to the datasheet the ESP32 can sink 28mA and source 40mA. The master in this case was not doing more then 31uA.

[image]

wanckl, 2025-03-11:
@mc-ff @S0urceror `gpio_set_drive_capability()` after spi initialised?

closed 2025-12-04 by Alvin1Zhang