espressif/esp-idf #11144: [mesh] v4.4 - LTE mesh poor performances

KonssnoK, 2023-04-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello there,
now that we have partial fixes for 
https://github.com/espressif/esp-idf/issues/9955
https://github.com/espressif/esp-idf/issues/11006

we enabled modem handling in our project.

We already have one beta installation that works exclusively on LTE connection -> Fixed root mesh with LTE and no WIFI

Apparently, there are some issues:
- it's impossible to perform normal OTA to devices on the network, it always fails. I think this is because the update rate is super slow 3KB/s.
- devices are going offline persistently until reset
- devices are reporting messages multiple times.

Question:
- do you have possibility to create a setup with an esp32s3 connected to a modem? (we have UART connection)
- if not, should we organize some hardware for you?

Meanwhile we will setup in our offices some LTE installations to monitor and in order to give you logs.

KonssnoK, 2023-04-09:
further investigation is leading the disconnections events to multiple MQTT errors.
We are now starting to investigate the MQTT library, forking it.

Here an example:
```
[1] MQTT_EVENT_ERROR 32792 78 0 1 0 0
[694] MQTT_EVENT_ERROR 32774 0 0 1 0 0
[19] MQTT_EVENT_ERROR 32769 0 0 1 0 119
[33] MQTT_EVENT_ERROR 32769 0 0 1 0 11
[12] MQTT_EVENT_ERROR 32794 80 0 1 0 0
[4] MQTT_EVENT_ERROR 32794 29312 0 1 0 0
[1] MQTT_EVENT_ERROR 32794 76 0 1 0 0
```
We'll provide more info further on... not sure if we'll do PRs thought.

The other issues are still relevant to mesh:
- it's impossible to perform normal OTA to devices on the network, it always fails. I think this is because the update rate is super slow 3KB/s.
- devices are reporting messages multiple times.

zhangyanjiaoesp, 2023-04-10:
@KonssnoK 

> Question:
> - do you have possibility to create a setup with an esp32s3 connected to a modem? (we have UART connection)
> - if not, should we organize some hardware for you?

It will be better if you can organized some hardware for us, this can ensure the consistency between our tests and yours.

KonssnoK, 2023-04-13:
@zhangyanjiaoesp how would you see us making a dedicated VPN access to hardware in Zurich? (or Paris/Milan)

We have the strong belief that we will have problems in China the SIM/modem.

We would give access to a laptop with a device with the modem attached (and maybe other devices in the same network)

KonssnoK, 2023-04-17:
@zhangyanjiaoesp other question: does the mesh protocol internally implements retries? 
To understand if those multiple packets that we see could be caused by the mesh layer of if it's the MQTT layer

zhangyanjiaoesp, 2023-04-17:
@KonssnoK 

>  how would you see us making a dedicated VPN access to hardware in Zurich? (or Paris/Milan)

We can have  a try.
> does the mesh protocol internally implements retries?

When call `esp_mesh_send()` to send P2P mesh data, there have mesh retries to ensure the reliability of transmission.

zhangyanjiaoesp, 2023-04-17:
> > When call `esp_mesh_send()` to send P2P mesh data, there have mesh retries to ensure the reliability of transmission.
> 
> Sorry i didn't understand, if i call esp_mesh_send with type P2P -> there are internal retries?

yes

KonssnoK, 2023-04-25:
Hello @zhangyanjiaoesp ,
i have a question to solve another issue we are currently facing with the mesh layer.

Currently we have the devices sometimes getting stuck in the following condition:
[image]
which, after quite some work, came out being because the tcpip task is currently doing
[image]

coming from the code (which should be as in the default example):

```
static esp_err_t mesh_driver_transmit_from_node_sta(void* h, void* buffer, size_t len)
{
    mesh_data_t data;
    ESP_LOGD(TAG, "Sending to root, dest addr: " MACSTR ", size: %d", MAC2STR((uint8_t*)buffer), len);
    data.data = buffer;
    data.size = len;
    data.proto = MESH_PROTO_AP; // Node's station transmits data to root's AP
    data.tos = MESH_TOS_P2P;
    esp_err_t err = esp_mesh_send(NULL, &data, MESH_DATA_TODS, NULL, 0);
    if (err != ESP_OK) {
        LOGE(TAG, "STA: Send err 0x%X %s", err, esp_err_to_name(err));
    }
    return err;
}
```

Questions: 
- why is the esp_mesh_send waiting on a queue? i would expect it to add to a queue instead
- Could i add MESH_DATA_NONBLOCK  to the call? do you have advices on how to handle non-blocking queues? 
 
i will check if this small change will solve the stuck issue i'm seeing meanwhile

KonssnoK, 2023-04-25:
adding `MESH_DATA_NONBLOCK  `did not help.
i'm now also forcing a 2 seconds timeout with

`ESP_ERROR_CHECK(esp_mesh_send_block_time(2000));`

KonssnoK, 2023-04-25:
@zhangyanjiaoesp could you also please explain the difference between 
`ESP_ERROR_CHECK(esp_mesh_send_block_time(2000));`
and using
`    esp_err_t err = esp_mesh_send(NULL, &data, MESH_DATA_TODS, NULL, 0);`
or 
`    esp_err_t err = esp_mesh_send(NULL, &data, MESH_DATA_TODS | MESH_DATA_NONBLOCK , NULL, 0);`

Is MESH_DATA_NONBLOCK necessary for esp_mesh_send to be non-blocking if i set esp_mesh_send_block_time?
Explained in another way: if i do esp_mesh_send_block_time and then i call esp_mesh_send WITHOUT MESH_DATA_NONBLOCK , will it block forever OR will it block for 2 seconds??

Thanks!

KonssnoK, 2023-04-25:
@zhangyanjiaoesp , also sometimes we have wifi heap corruption.
What is the best procedure for this? we open a new issue? sadly it's quite difficult to reproduce but quite diffused amongst devices (multiple crashes everyday on multiple devices out of the 150 we have in the pilot).

[image]

zhangyanjiaoesp, 2023-04-26:
> @zhangyanjiaoesp , also sometimes we have wifi heap corruption. What is the best procedure for this? we open a new issue? sadly it's quite difficult to reproduce but quite diffused amongst devices (multiple crashes everyday on multiple devices out of the 150 we have in the pilot).
> 
> [image]
> 
> ```
> _rootless:1
> W (11:16:04.404) monitor: 3W3P:: rule deactivated       
> I (1298619) wifi:I (11:16:07.733) mesh_main: <MESH_EVENT_FIND_NETWORK>new channel:1, router BSSID:00:00:00:00:00:00
> mode : sta (7c:df:a1:e2:a3:04) + softAP (7c:df:a1:e2:a3:05)
> W (1298625) wifi:<MESH AP>adjust channel:1, secondary channel offset:1(40U)
> I (1298637) wifi:Total power save buffer number: 8      
> W (11:16:10.404) monitor: 3W3P:: rule activated
> W (11:16:11.404) monitor: 3W3P:: rule deactivated       
> I (1304711) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<1,0>, prof:1
> I (1305695) wifi:state: init -> auth (b0)
> CORRUPT HEAP: Bad head at 0x3dea961c. Expected 0xabba1234 got 0x3de00014
> 
> assert failed: multi_heap_free multi_heap_poisoning.c:259 (head != NULL)
> Setting breakpoint at 0x403761c5 and returning...       
> xtensa-esp32s3-elf-addr2line -pfiaC -e c:\src\v3\firmware\build\firmware.elf 0x403761c5: [WinError 2] The system cannot find the file specified
> ```

@KonssnoK Suggest to create a new issue to tracing the heap corruption issue.

zhangyanjiaoesp, 2023-04-26:
@KonssnoK 
When you set the `data.tos = MESH_TOS_P2P`, the `esp_mesh_send()` function will be blocked, it will send the mesh data to wifi task, and then waiting on a queue to recv the tx state. 
That is to say, if you set `data.tos = MESH_TOS_P2P`, then set the flag to include `MESH_DATA_NONBLOCK` will not work. At this time, you need to call `esp_mesh_send_block_time()` to specify a blocking time.

KonssnoK, 2023-06-20:
as requested,
i opened https://github.com/espressif/esp-idf/issues/11713
to track WIFI related crashes.
@zhangyanjiaoesp @mhdong 
Meanwhile we continue to investigate slowness of transfer with our LTE SIM provider

KonssnoK, 2023-12-19:
We are still investigating with the network providers and the modem manufacturer. For now this does not seem related to the mesh per se. I will close the issue for now and reopen in case we find that both modem and network are ok.

closed 2023-12-19 by KonssnoK

kieennt13, 2024-11-05:
Hello @KonssnoK ,
I have seen the ip_internal_network example in your forked esp-idf repository (branch lc/fixed_root_53) and have tried running it.

When Wi-Fi is available, everything works as expected.

However, when I disable Wi-Fi, although both the root and node switch to fixed root mode, their behavior confuses me. Below is the log of the root after switching to fixed root mode:

'''
I (156372) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201 ERROR
W (156372) ping: Ignore counters while transitioning. error 201
W (157052) mesh_hand: Triggering FIXED ROOT handover
I (157052) mesh: <MESH_NWK_PARENT_DISCONNECTED>already disconnected, ignore it
I (157052) mesh: [IO]disable self-organizing<reconnect>
I (157072) mesh: <nvs>write layer:0
I (157082) mesh: <nvs>write assoc:0
I (157112) mesh: [CONFIG]connect to router:XXLL, 00:00:00:00:00:00
W (157122) ping: From 8.8.8.8 icmp_seq=52 timeout
E (157122) ping_sock: send error=0
I (157162) mesh: <mesh_connect_to_router,540>parent is set<stop reconnect>g_is_wifi_connecting:1, g_is_wifi_disconnecting:0, g_mesh_stop_reconnection:0
I (157162) mesh: [wifi]disconnected reason:106(scan fail), continuous:6/max:12, root, vote(,stopped)<><>I (157162) wifi:station: b0:a7:32:17:26:40 leave, AID = 1, bss_flags is 134243, bss:0x3ffba9ec
I (157182) wifi:new:<13,0>, old:<13,2>, ap:<13,2>, sta:<13,0>, prof:13
I (157192) wifi:mode : sta (cc:7b:5c:27:09:70)
I (157162) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:106 ERROR
W (157222) mesh_hand: Stop mesh auto-reconnect
I (157222) mesh: [IO]disable self-organizing<stop reconnect>
I (157232) mesh: [scan]new scanning time:300ms, beacon interval:100ms
I (157232) mesh_main: <MESH_EVENT_CHILD_DISCONNECTED>aid:1, b0:a7:32:17:26:40
I (157242) mesh: [IO]disable self-organizing<stop reconnect>
I (159432) mesh: [wifi]disconnected reason:201(), continuous:0/max:12, root, vote(,stopped)<><>
I (159432) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:201 ERROR
I (159432) mesh: <mesh_nwk_task_main,4587>parent is set<stop reconnect>g_is_wifi_connecting:0, g_is_wifi_disconnecting:0, g_mesh_stop_reconnection:1
I (159452) mesh_main: <MESH_EVENT_STOP_RECONNECTION>
'''

I am not sure if it is actually participating in the mesh network or acting as the root. Could this be because my root is currently not connected to the SIM module?

And here is the log of the node:

'''
I (156489) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason:2 ERROR
W (156489) ping: Ignore counters while transitioning. error 2
I (156529) mesh_main: <MESH_EVENT_NETWORK_STATE>is_rootless:1
I (156529) mesh: 5206[healing]looking for a new parent, [L:2]try layer:1[revote][scan]
W (157139) wifi:scan number is 0
I (157139) mesh: [SCAN][ch:13]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][ba:08:8c:94:20:17]<>
I (157139) mesh: [FAIL][1]root:0, fail:1, normal:0, <pre>backoff:0

W (157559) ping: From 8.8.8.8 icmp_seq=12 timeout
E (157559) ping_sock: send error=0
W (157749) wifi:scan number is 0
I (157749) mesh: [SCAN][ch:13]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][ba:08:8c:94:20:17]<>
I (157749) mesh: [FAIL][2]root:0, fail:2, normal:0, <pre>backoff:0

W (158359) wifi:scan number is 0
I (158359) mesh: [SCAN][ch:13]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][ba:08:8c:94:20:17]<>
I (158359) mesh: [FAIL][3]root:0, fail:3, normal:0, <pre>backoff:0

W (158969) wifi:scan number is 0
I (158969) mesh: [SCAN][ch:13]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][ba:08:8c:94:20:17]<>
I (158979) mesh: [FAIL][4]root:0, fail:4, normal:0, <pre>backoff:0

W (159579) wifi:scan number is 0
I (159579) mesh: [SCAN][ch:13]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][ba:08:8c:94:20:17]<>
I (159589) mesh: [FAIL][5]root:0, fail:5, normal:0, <pre>backoff:0

W (160189) wifi:scan number is 0
I (160189) mesh: [SCAN][ch:13]AP:0, other(ID:0, RD:0), MAP:0, idle:0, candidate:0, root:0, topMAP:0[c:2,i:1][ba:08:8c:94:20:17]<>
I (160199) mesh: [FAIL][6]root:0, fail:6, normal:0, <pre>backoff:0

W (160559) ping: From 8.8.8.8 icmp_seq=13 timeout
[... 195 lines cut ...]
W (223579) ping: From 8.8.8.8 icmp_seq=34 timeout
E (223579) ping_sock: send error=0
W (226579) ping: From 8.8.8.8 icmp_seq=35 timeout
E (226579) ping_sock: send error=0
W (229579) ping: From 8.8.8.8 icmp_seq=36 timeout
E (229579) ping_sock: send error=0
W (232579) ping: From 8.8.8.8 icmp_seq=37 timeout
E (232579) ping_sock: send error=0
W (235579) ping: From 8.8.8.8 icmp_seq=38 timeout
E (235579) ping_sock: send error=0
W (238579) ping: From 8.8.8.8 icmp_seq=39 timeout
E (238579) ping_sock: send error=0
W (241579) ping: From 8.8.8.8 icmp_seq=40 timeout
E (241579) ping_sock: send error=0
I (241619) mesh_hand: Checking wifi connectivity (count 0)[LOCAL]
I (244459) mesh: 4016<scan>self-organized is disabled, users shall read out scan results by themselves.
I (244459) mesh_main: <MESH_EVENT_SCAN_DONE>number:14
W (244459) wifi:Haven't to connect to a suitable AP now!
W (244469) mesh_hand: scan_done: found 0, fixed_root 1, has_parent 0, forced 1
W (244469) mesh_hand: All nodes have bad connection, allow any RSSI.
'''

I tried checking the root by calling esp_mesh_get_type() in mesh_handover_disconnect_callback(), and it still returns a value of 1.

I would greatly appreciate it if you could help explain this behavior.

KonssnoK, 2024-11-05:
hi @kieennt13, 
this branch is an export of part of our internal production code.
since it is moved on the espressif example we had to strip all the functionalities related to the modem itself.
your modem should create a ppp interface with priority higher than the WIFI one.
This way, once the modem will receive the IP, the interface will start routing traffic through the modem.

All "FAKE MODEM" related functions were made to try to simulate a modem in the example connecting to a different WIFI, but they should be ignored and replaced with real modem-related functions. (I always have a real modem-equipped device in the network)

The node should start scanning every 30 seconds , they won't connect unless they find a root device or another leaf device already connected to the fixed root.

KonssnoK, 2024-11-05:
> We are still investigating with the network providers and the modem manufacturer. For now this does not seem related to the mesh per se. I will close the issue for now and reopen in case we find that both modem and network are ok.

and BTW we are still having huge issues with our production Modem, which came out to be the responsible for all the bad performances.

kieennt13, 2024-11-06:
Hi, @KonssnoK ,
Thanks for your reply!
I am working on a project where the LTE connection is established using a library rewritten from TinyGSM to be compatible with ESP-IDF. LTE is connected before initializing the mesh, and it has its own functions for handling data transmission and reception.

The mesh network I created is based on a combination of the manual_networking and internal_communication examples from ESP-IDF, so it lacks the mesh_netif library for customization. 
In the Wi-Fi-based mesh scenario, I don’t even need to set up netif because my program has already connected to Wi-Fi beforehand.

My question is, in a scenario where the mesh switches from Wi-Fi to LTE, how can I ensure that the PPP interface created by the modem has a higher priority than that of Wi-Fi? 
And, since I do not have mesh_netifs_stop(), how can I disable the Wi-Fi netif?

Also, I would greatly appreciate it if you could provide logs for the root and nodes in the scenario when switching to fixed-root mode.
Thanks!

KonssnoK, 2024-11-06:
Sorry we do not have capacity to provide support on our pieces of code :(

kieennt13, 2024-11-06:
So sad to hear that :(
Anyway thanks for support!

kieennt13, 2024-11-06:
@zhangyanjiaoesp could you please help me figured out this problem?

KonssnoK, 2024-11-06:
@kieennt13 here is an example log of an handover

```
W (15:42:21.436) mesh_hand: Triggering FIXED ROOT handover
I (183999) mesh: <MESH_NWK_PARENT_DISCONNECTED>already disconnected, ignore it
I (184000) mesh: [IO]disable self-organizing<reconnect>
I (184052) mesh: [CONFIG]connect to router:KI, 00:00:00:00:00:00
I (184060) mesh: <mesh_connect_to_router,541>parent is set<stop reconnect>g_is_wifi_connecting:1, g_is_wifi_disconnecting:0, g_mesh_stop_reconnection:0
I (184073) mesh: [wifi]disconnected reason:106(scan fail), continuous:5/max:12, root, vote(,stopped)<><>
I (184131) mesh: [IO]disable self-organizing<stop reconnect>
W (15:42:21.575) mesh_main: <MESH_EVENT_PARENT_DISCONNECTED>reason: 106 MESH_REASON_SCAN_FAIL
W (15:42:21.577) modem_comm: Signal: RSSI = -80.0/-75.0
W (15:42:21.583) mesh_hand: Stop mesh auto-reconnect
I (184156) mesh: [IO]disable self-organizing<stop reconnect>
W (15:42:21.600) modem_comm: LTE signal: RSRP -105.0/-100.0, DPL 105.0/110.0, SiNR 5.0/10.0
W (15:42:22.129) modem_comm: AT+CREG?: 2,5,"2F08","0071DF21",7
W (15:42:22.133) modem_comm: AT+CEREG?: 5,5,"2F08","0071DF21",7
W (15:42:22.137) modem_comm: AT+CGREG?: 2,0
W (15:42:22.144) modem_comm: LTE signal: RSRP -105.0/-100.0, DPL 105.0/110.0, SiNR 5.0/10.0
W (15:42:22.166) modem_comm: Selected operator: 0,0,"vodafone IT",7
W (15:42:22.171) modem_comm: Active LTE band(s): 0,0000000000000000000004
I (187013) mesh: [wifi]disconnected reason:201(), continuous:6/max:12, root, vote(,stopped)<><>
I (187027) mesh: <mesh_nwk_task_main,4597>parent is set<stop reconnect>g_is_wifi_connecting:0, g_is_wifi_disconnecting:0, g_mesh_stop_reconnection:1
I (15:42:24.478) mesh_main: <MESH_EVENT_STOP_RECONNECTION>
I (15:42:25.116) tmpr_propagation: Propagated local temperature
I (15:42:28.214) esp-netif_lwip-ppp: Connected
I (15:42:28.215) modem_comm: Modem connected to PPP server
W (15:42:28.216) network: GOT IP from ppp_sta
I (15:42:28.221) network: Network connected
I (15:42:28.225) network: Root node can access external IP network
W (190788) wifi:Haven't to connect to a suitable AP now!
I (190799) mesh: <MESH_NWK_ROOT_TODS_STATE>toDS:1
I (15:42:28.238) mesh_netif: Interface AP
I (15:42:28.246) mqtt_app: MQTT_EVENT_BEFORE_CONNECT
I (15:42:28.252) mqtt_app: Try [1] connection to mqtts://mqtt-stg.tiko.energy:8883
W (15:42:28.260) mesh_netif: DNS server [0]: 192.168.10.110
W (15:42:28.267) mesh_netif: DNS server [1]: 194.51.3.56
I (15:42:28.274) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 10.0.0.1
W (190844) wifi:Haven't to connect to a suitable AP now!
I (15:42:28.286) mesh_netif: Clearing interface <sta>
I (15:42:28.293) mesh_netif: It was a wifi station, removing handlers
I (15:42:28.301) mesh_netif: Interface PPP
W (15:42:28.304) mesh_main: <MESH_EVENT_TODS_REACHABLE>reachable:1
I (15:42:31.575) stats: iteration_time_us=999987 Name=last% IDLE0=47 IDLE1=98 mqtt_task=51 
I (15:42:31.806) mqtt_app: MQTT_EVENT_CONNECTED 
```

kieennt13, 2024-11-06:
Thanks a lots for those logs! @KonssnoK 
Does your modem_comm based on the Pppos_client example in ESP-IDF or something else?
I'd love to dive into it and see how to create a PPP interface for mesh.

KonssnoK, 2024-11-07:
@kieennt13 we use esp-protocols as addon for supporting AT commands

kieennt13, 2024-11-07:
@KonssnoK sorry, I don't have much knowlegde about the PPP interface and others related, how do you create a PPP interface with just AT Commands, because based on the Pppos_client example, I see they using modem_dce:
```
 /* Configure the PPP netif */
    esp_modem_dce_config_t dce_config = ESP_MODEM_DCE_DEFAULT_CONFIG(CONFIG_EXAMPLE_MODEM_PPP_APN);
    esp_netif_config_t netif_ppp_config = ESP_NETIF_DEFAULT_PPP();
    esp_netif_t *esp_netif = esp_netif_new(&netif_ppp_config);
    assert(esp_netif);
```
 to configured it.

KonssnoK, 2024-11-07:
@kieennt13 esp has a default for a PPP netif interface.
We also do the same as in that example probably, this part was not written by me, so i can't help.

kieennt13, 2024-11-07:
@KonssnoK 
Is that so... :(
Btw, is there any log indicating that the new device has successfully become the root and that the node has connected to the root after the root starts the modem? 
I know I am asking too much, but I would be very grateful if you could provide me with the full logs for both the node and the root.
Thanks!