espressif/esp-idf #11145: espcoredump.py is not recognized as internal or external command.

mayankbytebeam, 2023-04-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I tried to run espcoredump.py script and got the following error.

[image]

mayankbytebeam, 2023-04-11:
@espressif-bot

erhankur, 2025-03-11:
I think, in the ESP-IDF environment, `python -m esp_coredump` should work. Let us know if you still have a problem while running the script.

closed 2025-03-11 by erhankur