espressif/esp-idf #11146: AVRCP: TG: Not able to get ESP_AVRC_RN_TRACK_CHANGE notification from CT

espradio, 2023-04-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

ESP-IDF version:   v4.4.4-340-g4c7d97e2bd

I am trying to configure ESP32 as AVRCP TG mode to receive ESP_AVRC_RN_TRACK_CHANGE event from a Bluetooth speaker. i.e Waiting for notification on ESP32 while pressing "NEXT" button on Bluetooth speaker.

Using following APIs, able to get ESP_AVRC_TG_CONNECTION_STATE_EVT(with conn_stat.connected as true) and 
ESP_AVRC_TG_REMOTE_FEATURES_EVT(with AVRC remote features: 24b, CT features: 1) event
```
esp_avrc_tg_init();
esp_avrc_tg_register_callback(bt_app_rc_tg_cb));

```
Now based on examples, one should set the local TG capability for ESP_AVRC_RN_TRACK_CHANGE
```
esp_avrc_rn_evt_cap_mask_t evt_set = {0};
esp_avrc_rn_evt_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_SET, &evt_set, ESP_AVRC_RN_TRACK_CHANGE);
esp_avrc_tg_set_rn_evt_cap(&evt_set));
```
But esp_avrc_tg_set_rn_evt_cap() returns ESP_ERR_NOT_SUPPORTED due to the fact that `cs_rn_allowed_evt` configure 0x2000. Unlike, CT mode where it has `esp_avrc_ct_send_register_notification_cmd() `to register a specific event, TG mode does not have specific API to register the notification. So I assuming` esp_avrc_tg_set_rn_evt_cap() `needs to be called to advertise the capability.

Questions:

1.  Is above programming model correct?
2.  Is this case where IDF does not support ESP_AVRC_RN_TRACK_CHANGE notification from CT to TG? i.e feature is not implemented? if so, any plan for this feature support?
3. If the above programming model not correct, could you share the example API sequence to enable this use-case?

Thanks in advance.

xiongweichao, 2023-04-07:
Hi @espradio ,

I'm not sure if the peer device supports this feature. You can provide the following log or check whether bit 3 in the `bitmask` is 1 by yourself.
```
RC_CT: remote rn_cap: count 5, bitmask 0xc26
```

Thanks

espradio, 2023-04-07:
Hi @xiongweichao 

Please note, I am trying configure esp32 as TG and other side a bluetooth speaker as CT.

Remote CT supports this as ESP_AVRC_TG_REMOTE_FEATURES_EVT returns as  "AVRC remote features: 24b,"

Looks like esp32 TG implementation supports only volume change.
file: components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c

See
```
const static uint16_t cs_rn_allowed_evt = \
        0x2000; /* bit mask: 0=rsvd, 1=play_status, 2=track_chg, 3=reached_end,
                             4=reached_start, 5=pos_chg, 6=batt_status, 7=sys_status_chg,
                             8=player_app_setting, 9=now_playing_content, 10=avail_players, 11=addressed_player,
                             12=UID, 13=volume_chg */
```
```

static void btc_avrc_tg_send_rn_rsp(esp_avrc_rn_event_ids_t event_id, esp_avrc_rn_rsp_t rsp, const esp_avrc_rn_param_t *param)
{
    tAVRC_RESPONSE avrc_rsp;
    if (! btc_avrc_tg_connected_p()) {
        BTC_TRACE_WARNING("%s, RC unconnected, operation fail, event_id 0x%x", __FUNCTION__, event_id);
        return;
    }

    if (btc_rc_cb.rc_ntf[event_id - 1].registered == FALSE) {
        BTC_TRACE_ERROR("Event id not registered: event_id = %x", event_id);
        return;
    }
    memset(&(avrc_rsp.reg_notif), 0, sizeof(tAVRC_REG_NOTIF_RSP));
    avrc_rsp.reg_notif.event_id = event_id;

    switch (event_id) {
    case ESP_AVRC_RN_VOLUME_CHANGE:
        avrc_rsp.reg_notif.param.volume = param->volume;
        break;
    // todo: implement other event notifications
    default:
        BTC_TRACE_WARNING("%s : Unhandled event ID : 0x%x", __FUNCTION__, event_id);
        return;
    }
```

xiongweichao, 2023-04-07:
Hi @espradio ,

Sorry. Remote TG does not support this function for now.  Please try the following changes to see if the problem can be solved.
```
const static uint16_t cs_rn_allowed_evt = \
        0x2004; /* bit mask: 0=rsvd, 1=play_status, 2=track_chg, 3=reached_end,
                             4=reached_start, 5=pos_chg, 6=batt_status, 7=sys_status_chg,
                             8=player_app_setting, 9=now_playing_content, 10=avail_players, 11=addressed_player,
                             12=UID, 13=volume_chg */
```
```

static void btc_avrc_tg_send_rn_rsp(esp_avrc_rn_event_ids_t event_id, esp_avrc_rn_rsp_t rsp, const esp_avrc_rn_param_t *param)
{
    tAVRC_RESPONSE avrc_rsp;
    if (! btc_avrc_tg_connected_p()) {
        BTC_TRACE_WARNING("%s, RC unconnected, operation fail, event_id 0x%x", __FUNCTION__, event_id);
        return;
    }

    if (btc_rc_cb.rc_ntf[event_id - 1].registered == FALSE) {
        BTC_TRACE_ERROR("Event id not registered: event_id = %x", event_id);
        return;
    }
    memset(&(avrc_rsp.reg_notif), 0, sizeof(tAVRC_REG_NOTIF_RSP));
    avrc_rsp.reg_notif.event_id = event_id;

    switch (event_id) {
    case ESP_AVRC_RN_VOLUME_CHANGE:
        avrc_rsp.reg_notif.param.volume = param->volume;
        break;
    case ESP_AVRC_RN_TRACK_CHANGE:
        memcpy(avrc_rsp.reg_notif.param.track, param->elm_id, sizeof(tAVRC_UID));
        break;
    // todo: implement other event notifications
    default:
        BTC_TRACE_WARNING("%s : Unhandled event ID : 0x%x", __FUNCTION__, event_id);
        return;
    }
```
add example
```
esp_avrc_rn_evt_cap_mask_t evt_set = {0};
esp_avrc_rn_evt_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_SET, &evt_set, ESP_AVRC_RN_TRACK_CHANGE);
esp_avrc_tg_set_rn_evt_cap(&evt_set));
```

Thanks

espradio, 2023-04-08:
With above, registration response went though. Looks like CT is looking for response to that, so got the following

```
W (708918) BT_APPL: bta_av_proc_meta_cmd unhandled RC vendor PDU: 0x20
E (714038) BT_APPL: bta_dm_pm_btm_status hci_status=17
```
Added following to skip that error, I dont  think, it is correct. Just tried.
```

diff --git a/components/bt/host/bluedroid/bta/av/bta_av_act.c b/components/bt/host/bluedroid/bta/av/bta_av_act.c
index f7f3dc3fc0..fbf8b51def 100644
--- a/components/bt/host/bluedroid/bta/av/bta_av_act.c
+++ b/components/bt/host/bluedroid/bta/av/bta_av_act.c
@@ -781,6 +781,9 @@ tBTA_AV_EVT bta_av_proc_meta_cmd(tAVRC_RESPONSE  *p_rc_rsp, tBTA_AV_RC_MSG *p_ms
             /* Setting of a value by CT does not implicitly mean that the setting will take effect on TG. */
             /* The setting shall take effect after a play command from CT. */
             break;
+       case AVRC_PDU_GET_ELEMENT_ATTR:
+            p_rc_rsp->rsp.status = BTA_AV_STS_NO_RSP;
+            break;
```
Based on AVRCP_v1.6.2.pdf, section 29.1 MSC for track change event. CT post the events as PASSTHROUGH command. Looks like that is not implemented either.
Looks like it is complicated feature, Can espressif enable this feature?

xiongweichao, 2023-04-10:
Hi @espradio ,

Sorry, this feature is more complicated, I will provide you with a patch as soon as possible.

Thanks

espradio, 2023-04-10:
Thanks. looking forward for the new feature.

espradio, 2023-05-05:
@xiongweichao Any update on this?

xiongweichao, 2023-05-06:
Hi @espradio ，

Sorry, this function is complicated and not yet completed.
[test_patch.txt](https://github.com/espressif/esp-idf/files/11411025/test_patch.txt)

Thanks

espradio, 2023-05-06:
Hi @xiongweichao , I understand the  patch is not yet completed. But I did some testing, here are the observations

1) I am testing as ESP as SOURCE <-> BT speaker as SINK. Where ESP SOURCE is TG and BT speaker is controller.
2) In my case sequence of events in TG callback are
ESP_AVRC_TG_CONNECT_STATE_EVT // connection established
ESP_AVRC_TG_CONNECT_FEATURES_EVT with AVRC remote features: 24b, CT features: 1
ESP_AVRC_TG_REGISTER_NOTIFICATION_EVT // Here it is crashing due to p_info NULL

Some of my understanding:
I changed the code to return following on ESP_AVRC_TG_REGISTER_NOTIFICATION_EVT
```
        if (rc->reg_ntf.event_id == ESP_AVRC_RN_TRACK_CHANGE)
        {
            printf("=====ESP_AVRC_RN_TRACK_CHANGE=====\n");
             esp_avrc_rn_param_t rn_param;
            for (int i = 0; i < 8; i++) {
                    rn_param.elm_id[i] = 0xFF; // Don't we need 0xFF here for the INTERM case?
            }
            esp_avrc_tg_send_rn_rsp(ESP_AVRC_RN_TRACK_CHANGE, ESP_AVRC_RN_RSP_INTERIM, &rn_param);
```

That was failing in driver layer then I added 
```
@@ -1363,6 +1389,9 @@ static void btc_avrc_tg_send_rn_rsp(esp_avrc_rn_event_ids_t event_id, esp_avrc_r
     case ESP_AVRC_RN_VOLUME_CHANGE:
         avrc_rsp.reg_notif.param.volume = param->volume;
         break;
+    case ESP_AVRC_RN_TRACK_CHANGE:
+        memcpy(avrc_rsp.reg_notif.param.track, param->elm_id, sizeof(tAVRC_UID));
+        break;
```
Then the ESP_AVRC_TG_GET_ELEM_ATTRS_EVT came returned with esp_avrc_tg_send_infor_rsp()

I was expecting  PassThrough command (FWD,PRESSED) event on FWD button presson BT speak, That did not came yet.

xiongweichao, 2023-05-09:
Hi @espradio ,

Did you call the [esp_avrc_tg_set_psth_cmd_filter](https://github.com/espressif/esp-idf/blob/17451f1fb3d6485af5113d96b6836a7ce60efee8/components/bt/host/bluedroid/api/include/api/esp_avrc_api.h#L638) API?

Thanks

espradio, 2023-05-14:
HI @xiongweichao 

Addtion of following code returns the error with ESP_ERR_NOT_SUPPORTED

```
+        esp_avrc_psth_bit_mask_t bit_mask;
+        esp_avrc_psth_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_SET, &bit_mask, ESP_AVRC_PT_CMD_CHAN_UP);
+        esp_avrc_psth_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_SET, &bit_mask, ESP_AVRC_PT_CMD_CHAN_DOWN);
+        ESP_ERROR_CHECK(esp_avrc_tg_set_psth_cmd_filter(ESP_AVRC_PSTH_FILTER_SUPPORTED_CMD, &bit_mask));
```

I am attaching test patch on top of https://github.com/espressif/esp-idf/files/11411025/test_patch.txt.

Could you please implement and check with "ESP as SOURCE <-> BT speaker as SINK" and able to get callback on Track change button press on BT speaker(SINK))
[test_patch_on_top_xiongweichao_change.txt](https://github.com/espressif/esp-idf/files/11472181/test_patch_on_top_xiongweichao_change.txt)

xiongweichao, 2023-05-15:
Hi @espradio ,

`Addtion of following code returns the error with ESP_ERR_NOT_SUPPORTED`
We do not support this [feature](https://github.com/espressif/esp-idf/blob/17451f1fb3d6485af5113d96b6836a7ce60efee8/components/bt/host/bluedroid/btc/profile/std/avrc/btc_avrc.c#LL63C42-L63C42).

We may not need to reply to him immediately when the `ESP_AVRC_TG_GET_ELEM_ATTRS_EVT` event comes up. How do you feel about replying when sourceTrack changes?

Thanks

espradio, 2023-05-15:
Thanks @xiongweichao . Setting the following is working. I can live with just PASSTHOUGH command now. We can close this ticket then.

```
esp_avrc_psth_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_SET, &bit_mask, ESP_AVRC_PT_CMD_FORWARD);
esp_avrc_psth_bit_mask_operation(ESP_AVRC_BIT_MASK_OP_SET, &bit_mask, ESP_AVRC_PT_CMD_BACKWARD);
```

closed 2023-05-15 by espradio