espressif/esp-idf #11148: report an i2c_bus.c mutex bug

HikerPan, 2023-04-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Sorry, my english is poor.

My ESP-IDF version is V4.4.4
In my project, I use two i2c port. 
I2C_NUM_0 for ES8311 I2C, It's define by ESP-ADF.
And I also have an touch chip use I2C_NUM_1.
The below code is part of my touch chip source code.
I need to init it and deinit it at some times.
And while my code deinit the I2C_NUM_1 and at the same time i use I2C_NUM_0.
The programe crashed.
`
void app_touch_i2c_init(int scl, int sda)
{
    i2c_config_t es_i2c_cfg = {
        .mode = I2C_MODE_MASTER,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 100000,
    };

    es_i2c_cfg.scl_io_num = scl;
    es_i2c_cfg.sda_io_num = sda;
    touch_i2c = i2c_bus_create(I2C_NUM_1, &es_i2c_cfg);
}

void app_touch_i2c_deinit()
{
    i2c_bus_delete(touch_i2c);
    touch_i2c = NULL;
}
`

I check the crash information.
And found the _busLock is NULL at i2c_bus.c
`
assert failed: xQueueSemaphoreTake queue.c:1545 (( pxQueue ))
`
So I think the next code may be the root cause.
When i call deinit. it will call this delete and destory _busLock.
But I2C_NUM_0 still need to use it.
So, thre crash happened.

`
esp_err_t i2c_bus_delete(i2c_bus_handle_t bus)
{
    I2C_BUS_CHECK(bus != NULL, "Handle error", ESP_FAIL);
    i2c_bus_t *p_bus = (i2c_bus_t *) bus;
    if (--p_bus->ref_count == 0) {
        i2c_driver_delete(p_bus->i2c_port);
        i2c_bus[p_bus->i2c_port] = NULL;
        audio_free(p_bus);
        mutex_destroy(_busLock);
        _busLock = NULL;
    }
    return ESP_OK;
}
`
And when i change the code to below， it works well.
`
esp_err_t i2c_bus_delete(i2c_bus_handle_t bus)
{
    I2C_BUS_CHECK(bus != NULL, "Handle error", ESP_FAIL);
    i2c_bus_t *p_bus = (i2c_bus_t *) bus;
    if (--p_bus->ref_count == 0) {
        i2c_driver_delete(p_bus->i2c_port);
        i2c_bus[p_bus->i2c_port] = NULL;
        audio_free(p_bus);
        // mutex_destroy(_busLock);
        // _busLock = NULL;
    }
    return ESP_OK;
}
`

mythbuster5, 2023-04-07:
https://github.com/espressif/esp-adf/blob/master/components/esp_peripherals/driver/i2c_bus/i2c_bus.c  Please report this issue in esp-adf

HikerPan, 2023-04-07:
sorry， i will move to esp-adf

closed 2023-04-07 by HikerPan