espressif/esp-idf #11149: Issue when BLE provisioning on Android 12 devices

ricbrue, 2023-04-07:
Hi, I am testing the BLE provisioning demo with the BLE provisioning app found on Google Play.
I've noticed that on Android 10 it works as expected meanwhile on Android 12 (maybe also Android 11) when the device is scanned and clicked from the list of discovered devices, it asks for pairing, when paired it goes forward with the provisioning procedure.
But here it comes the issue, restarting the device, unpairing it from the Bluetooth paired devices on Android and restarting the whole procedure (discovering the devices and then clicking on the BLE device) it gives connection error (saying that the device is not compatible), and I cannot came out from this situation. 
What could be the issue here? I cannot understand if it is a issue with Android 12 or with the ESP.
I've used the ESP 4.4.3 version.

rahult-github, 2023-04-07:
Hi @ricbrue ,

Please check if disabling WIFI_PROV_BLE_FORCE_ENCRYPTION ( under Component Config --> Wi Fi Provisioning Manager ) and then doing a fresh flashing helps. 

if no, then i request you to please share debug enabled console logs on esp device and btsnoop log on android phone. 

For enabling debug logs , please enable:
   Component Config  --> Log Output --> Default Log Verbosity --> Debug
  Component Config --> Bluetooth --> NimBLE Options --> NimBLE Host log verbosity --> Debug Logs

ricbrue, 2023-04-07:
Update #1
I've downgraded the ESP version to 4.3, it works as expected on Andorid 12 and it doesn't ask for pairing (good).

rahult-github, 2023-04-25:
Hi @ricbrue ,  

Does [this](https://github.com/espressif/esp-idf/issues/11149#issuecomment-1500019127) help ?

rahult-github, 2023-08-07:
Closing issue. Please feel free to reopen if the provided method doesn't help and share updated logs.

closed 2023-08-07 by rahult-github