espressif/esp-idf #11150: Cannot monitor device from docker image

listout, 2023-04-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

As mentioned in the documentation [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-docker-image.html#using-the-image-interactively), one can flash and monitor the device if the `privileged` flag is passed while running the docker container.

But unfortunately, I can't monitor the device, however, I can flash the device fine. The error message I'm getting: `termios.error: (25, 'Inappropriate ioctl for device')`

I'm invoking the docker container as: `gitlab-runner --debug exec docker 'testing' --docker-privileged`

Basically, I want to run my Unit Tests on the device connected to CI server (kind of like HIL testing)

Full log: http://ix.io/4sRN

fhrbata, 2023-04-08:
Hello @listout ,

idf monitor expects stdin to be attached to tty. The  `Using the image interactively` docs states that the docker should be started with tty allocated `Start the container with -i -t flags:`. I could be missing something, but it doesn't seem like this is possible with the gitlab-runner. In this case I think you can use e.g. `cat`, `socat` or other tools instead of idf monitor to get the output only. I quickly tried the following `.gitlab-ci.yml` with the hello_world example and `gitlab-runner exec docker test  --docker-privileged`

```
stages:
  - test

test:
  stage: test
  image: espressif/idf
  script:
    - idf.py set-target esp32s3
    - idf.py build
    - idf.py flash
    - stty -F /dev/ttyUSB0 115200 -cooked
    - cat /dev/ttyUSB0
```

HTH

listout, 2023-04-08:
I understand, I'll try this out.

I want to run my unit tests on the ESP32 connected to the system hosting GitLab-runner. Can you suggest a good solution, i.e. running the unit tests and showing the report, or just getting a log?

igrr, 2023-04-08:
One thing you can try for CI tests is https://github.com/espressif/pytest-embedded. It is used quite a lot in ESP-IDF and related projects, including in Gitlab pipelines.

There is a template repository https://github.com/espressif/gh-esp-test-template for GitHub actions, please take a look at it as well.

dobairoland, 2023-04-21:
The error message was improved (see linked commit) and will be available in `esp-idf-monitor==1.0.4` bundled with ESP-IDF v5.1 and later versions.  Thanks @fhrbata!