espressif/esp-idf #11152: Unable to set target using idf.py with Idf version v5.1

aggaddam, 2023-04-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32C3-DEVKIT

### Power Supply used.

USB

### What is the expected behavior?

1. Download the source code as a Download zip from the branch 'release/v5.1' (https://github.com/espressif/esp-idf/tree/release/v5.1)
2. Extract the files from the downloaded zip file
3. Installing the tools for target esp32c3 by running "./install.sh esp32c3"
4. Run the . ./export.sh
5. Go to hello world example project (/examples/get-started/hello_world)
6. Run idf.py set-target esp32c3
7. Observed the below error 

```
$ idf.py set-target esp32c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c3, new sdkconfig will be created.
Running cmake in directory /home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/agaddam/.espressif/python_env/idf5.1_py3.10_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=0 /home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world"...
-- Found Git: /usr/bin/git (found version "2.34.1") 
fatal: not a git repository (or any of the parent directories): .git
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32c3
CMake Error at /home/agaddam/Documents/espidfrelease_v51/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 47

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /home/agaddam/Documents/espidfrelease_v51/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at
  /home/agaddam/Documents/espidfrelease_v51/components/mqtt/CMakeLists.txt:1
  (include):

    include could not find requested file:

  

      /home/agaddam/Documents/espidfrelease_v51/components/mqtt/esp-mqtt/CMakeLists.txt

  Call Stack (most recent call first):

    /home/agaddam/Documents/espidfrelease_v51/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /home/agaddam/Documents/espidfrelease_v51/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /home/agaddam/Documents/espidfrelease_v51/tools/cmake/build.cmake:573 (__component_get_requirements)
  /home/agaddam/Documents/espidfrelease_v51/tools/cmake/project.cmake:464 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build/log/idf_py_stderr_output_5200 and /home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build/log/idf_py_stdout_output_5200
```

### What is the actual behavior?

idf.py set-taget esp32c3 should not give any error

### Steps to reproduce.

1. Download the source code as a Download zip from the branch 'release/v5.1' (https://github.com/espressif/esp-idf/tree/release/v5.1)
2. Extract the files from the downloaded zip file
3. Installing the tools for target esp32c3 by running "./install.sh esp32c3"
4. Run the . ./export.sh
5. Go to hello world example project (/examples/get-started/hello_world)
6. Run idf.py set-target esp32c3
7. Observed the below error 

```
$ idf.py set-target esp32c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c3, new sdkconfig will be created.
Running cmake in directory /home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/agaddam/.espressif/python_env/idf5.1_py3.10_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=0 /home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world"...
-- Found Git: /usr/bin/git (found version "2.34.1") 
fatal: not a git repository (or any of the parent directories): .git
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/agaddam/.espressif/tools/riscv32-esp-elf/esp-12.2.0_20230208/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32c3
CMake Error at /home/agaddam/Documents/espidfrelease_v51/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 47

  

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /home/agaddam/Documents/espidfrelease_v51/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  

  CMake Error at
  /home/agaddam/Documents/espidfrelease_v51/components/mqtt/CMakeLists.txt:1
  (include):

    include could not find requested file:

  

      /home/agaddam/Documents/espidfrelease_v51/components/mqtt/esp-mqtt/CMakeLists.txt

  Call Stack (most recent call first):

    /home/agaddam/Documents/espidfrelease_v51/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /home/agaddam/Documents/espidfrelease_v51/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

  

  

Call Stack (most recent call first):
  /home/agaddam/Documents/espidfrelease_v51/tools/cmake/build.cmake:573 (__component_get_requirements)
  /home/agaddam/Documents/espidfrelease_v51/tools/cmake/project.cmake:464 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build/log/idf_py_stderr_output_5200 and /home/agaddam/Documents/espidfrelease_v51/examples/get-started/hello_world/build/log/idf_py_stdout_output_5200
```

ESP-Marius, 2023-04-07:
Github do not include submodules when you download source code as a zip, which is why are getting mqtt related errors (the files do not exists, they are in the submodule)

To download  `release/v5.0` with git you can follow the [Getting started Guide](https://docs.espressif.com/projects/esp-idf/en/release-v5.0/esp32/get-started/linux-macos-setup.html#step-2-get-esp-idf)

Or if you for some reason are unable/dont want to clone we also package esp-idf with submodules as a downloadable zip file on our website for all stable release, e.g. [v5.0.1](https://dl.espressif.com/github_assets/espressif/esp-idf/releases/download/v5.0.1/esp-idf-v5.0.1.zip)

aggaddam, 2023-04-11:
@ESP-Marius Thanks for the response!

I also tried the below methods

Option 1: (Success - Cloning Idf v5.1 directly from github)
1. Clone the esp idf v5.1 using the below git command (`git clone --recurse-submodules -b release/v5.1 https://github.com/espressif/esp-idf.git`)
2. After a successful clone, run the `.install.sh esp32c3` command
3. Run `. ./export.s`h command
4. Navigate to the hello world example under examples folder
5. Successfully able to set target (`idf.py set-target esp32c3`) to esp32c3

Option 2: (Failure - Cloning Idf v5.1 directly from GitHub and updating the IDF v5.1 in our bitbucket or any cloud-hosted repo)
1. Clone the esp idf v5.1 using the below git command (`git clone --recurse-submodules -b release/v5.1 https://github.com/espressif/esp-idf.git`)
2. Delete .git folder in the esp-idf folder  (We don't need tracking history)
3. Commit all the changes of esp-idf folder to the bitbucket repo
4. Now, clone freshly esp-idf from bitbucket repo
5. run the `.install.sh esp32c3` command  (Successful)
6. Navigate to the hello world example under examples folder
5. Fail to set target to esp32c3 (`idf.py set-target esp32c3`)with the below error

`CMake Error at /home/agaddam/Documents/delete/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/mbedtls/mbedtls/library/CMakeLists.txt:277 (add_library):
  Cannot find source file:

    ssl_debug_helpers_generated.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at /home/agaddam/Documents/delete/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/mbedtls/mbedtls/library/CMakeLists.txt:265 (add_library):
  Cannot find source file:

    error.c

  Tried extensions .c .C .c++ .cc .cpp .cxx .cu .mpp .m .M .mm .ixx .cppm .h
  .hh .h++ .hm .hpp .hxx .in .txx .f .F .for .f77 .f90 .f95 .f03 .hip .ispc

CMake Error at /home/agaddam/Documents/delete/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/mbedtls/mbedtls/library/CMakeLists.txt:265 (add_library):
  No SOURCES given to target: mbedcrypto

CMake Error at /home/agaddam/Documents/delete/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/components/mbedtls/mbedtls/library/CMakeLists.txt:277 (add_library):
  No SOURCES given to target: mbedtls

CMake Generate step failed.  Build files cannot be regenerated correctly.
cmake failed with exit code 1, output of the command is in the /home/agaddam/Documents/delete/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_5367 and /home/agaddam/Documents/delete/li_ac_matter_fw/targets/simplifi/AC/li_esp32matter_sdk/esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_5367`

After careful analysis, it was observed that few files from mbedtls are getting committed to bitbucket repo. Here are the files that got missed while committing changes to bitbucket

1. version_features.c
2. ssl_debug_helpers_generated.c
3. psa_crypto_driver_wrappers.c
4. error.c
 
It seems some configuration files inside esp-idf ignoring these files. Please check and provide the resolution for this

ESP-Marius, 2023-04-11:
These files seems to be ignored in `components/mbedtls/mbedtls/library/.gitignore`, maybe when moving the files to bitbucket this ignore file is used somehow.

This doesnt seem to be an ESP-IDF issue though.

aggaddam, 2023-04-11:
@ESP-Marius I was following the same approach for esp-idf v4.4.2. It worked fine without any issue. Is this issue from Mbedtls library from third party ?

ESP-Marius, 2023-04-11:
The mbedtls version used in v4.4.2 did not have these file on the gitignore. 

The "issue" is from the mbedtls external library yes, but I dont think there is any actual issue with the library, rather that your workflow need to take into account this gitignore file.

aggaddam, 2023-06-22:
@ESP-Marius I was able to fix the issue by removing the below lines in .gitignore present under esp-idf\components\mbedtls\mbedtls\library

Original File:
 
```
*.o
libmbed*
*.sln
*.vcxproj

# Automatically generated files
/error.c
/version_features.c
/ssl_debug_helpers_generated.c
/psa_crypto_driver_wrappers.c
```

Modified File (Removed .c files from ignore list so that it will be committed to cloud-hosted git repo)

```
*.o
libmbed*
*.sln
*.vcxproj

```