espressif/esp-idf #11153: E (339928) esp-tls: couldn't get hostname for :xxxxxxxxxxxxxx.cloud: getaddrinfo() returns 202, addrinfo=0x0

habar84, 2023-04-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello , i'm using esp-protocol library from this repos : https://github.com/espressif/esp-protocols and try to use data mode to download a json file from my server.

The data mode is OK done but impossible to connect to my server , i got this error , any help please ? Thanks

D (339928) HTTP_CLIENT: Begin connect to: https://xxxxxxxxxxxxxxxxxxx.cloud:443
D (339928) esp-tls: host:xxxxxxx.cloud: strlen 27
E (339928) esp-tls: couldn't get hostname for :xxxxxxxxxxxxxx.cloud: getaddrinfo() returns 202, addrinfo=0x0
E (339938) esp-tls: Failed to open new connection
E (339948) transport_base: Failed to open a new connection
E (339958) HTTP_CLIENT: Connection failed, sock < 0

PS:
$ idf.py --version
ESP-IDF v5.1-dev-4290-g51772f4fb5-dirt

habar84, 2023-04-07:
I tried to change the default dns from the operator per the google one and others but i still have the same issue 

      esp_netif_dns_info_t dns_info1;
      ip4_addr_t dns_server_ip1;
      IP4_ADDR(&dns_server_ip1, 1, 1, 1,1);
      dns_info1.ip.u_addr.ip4.addr = dns_server_ip1.addr;
      esp_netif_set_dns_info(netif, ESP_NETIF_DNS_MAIN, &dns_info1);

      esp_netif_dns_info_t dns_info2;
      ip4_addr_t dns_server_ip2;
      IP4_ADDR(&dns_server_ip2, 8, 8, 8,8);
      dns_info2.ip.u_addr.ip4.addr = dns_server_ip2.addr;
      esp_netif_set_dns_info(netif, ESP_NETIF_DNS_BACKUP, &dns_info2);

closed 2023-04-10 by habar84

DerelictRobot, 2023-04-20:
@habar84 what was the resolution on this? Server-side issue?

davnec, 2023-04-20:
I had the same issue, but the problem is related to the hardware. 
SIMCOM 76xx need 2A for about 500uS. Sometime the power drop below 3.4V. In this case the module may reset, or show the error getaddrinfo() returns 202, addrinfo=0x0.
I've solved with a 1000uF capacitor in parallel of VBATT and GND.

DerelictRobot, 2023-04-25:
@davnec im using the integrated wifi radio of an esp32 connection - but wanted to share my own observations as it was a different root cause for the same presenting error.

In my case, there was an issue with the SSL version being used server-side that was causing the message my esp32 was sending to get truncated/clipped. I updated openssl and implemented the simple webserver through nginx, and the problem resolved itself for my esp32 without any change to firmware/hardware. 

I agree with your observations here, sounds like the modem was creating a high inrush current from your radio powering up to broadcast - ironically I ran into the same issue with the same cellular radio on another project recently.

CQC2016, 2023-06-19:
@habar84 I am facing the same proplem, any updates ?

habar84, 2023-06-19:
hello, i don't remember really the root cause for this issue, but I think I fixed it per enabling ssl certificate and use secure connection instead disabling SSL connection.

elanwu, 2023-10-11:
`...
E (10456) esp-tls: couldn't get hostname for :mqtt.eclipseprojects.io: getaddrinfo() returns 202, addrinfo=0x0
...`
Same "getaddrinfo() returns 202" problem.
Later, I noteced my **MI Android 14 phone** showes: "personal hotspot" >> "**connected devices**": **0**.
After I reconfig wifi option in idf, my phone showes "**connected device**": **1**, 
and "getaddrinfo() returns 202" problem disappered.

XingYu2535, 2024-03-22:
Hi,
I fixed this question by disabling config option menuconfig > 'Component config' > 'mbedTLS' > CONFIG_MBEDTLS_DATE_TIME, you guys can try it.

TroyMitchell911, 2025-03-06:
Is your router OpenWrt? If so, have you enabled `AdGuard Home`?

callmej0, 2025-05-09:
> Hi, I fixed this question by disabling config option menuconfig > 'Component config' > 'mbedTLS' > CONFIG_MBEDTLS_DATE_TIME, you guys can try it.

worked for me as well.