espressif/esp-idf #11155: 用esp-idf4.4可以编译，升级到esp-idf 5.1后编译失败

weiqifa0, 2023-04-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.0.1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

我想把esp-box项目移植到我原来的项目上。

### What is the actual behavior?

但是实际上移植总是出现问题，模块化感觉问题很多

### Steps to reproduce.

1. Step
2. Step
3. Step
...
就是编译，步骤就是下载工程，然后编译

### Build or installation Logs.

```plain
Running cmake in directory /Users/crisqifawei/esp/ESP32S3-SP/8.new_speech_rec/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /Users/crisqifawei/esp/ESP32S3-SP/8.new_speech_rec"...
-- Found Git: /opt/homebrew/bin/git (found version "2.39.0")
CMake Error at /Users/crisqifawei/esp/esp-idf/tools/cmake/project.cmake:243 (message):
  Directory specified in EXTRA_COMPONENT_DIRS doesn't exist:
  /Users/crisqifawei/esp/esp-idf/examples/common_components/led_strip
Call Stack (most recent call first):
  /Users/crisqifawei/esp/esp-idf/tools/cmake/project.cmake:327 (__project_init)
  CMakeLists.txt:13 (project)

-- Configuring incomplete, errors occurred!
HINT: The component with path specified in the EXTRA_COMPONENT_DIRS variable has been moved to IDF component manager (or has been removed).
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details.
cmake failed with exit code 1, output of the command is in the /Users/crisqifawei/esp/ESP32S3-SP/8.new_speech_rec/build/log/idf_py_stderr_output_9800 and /Users/crisqifawei/esp/ESP32S3-SP/8.new_speech_rec/build/log/idf_py_stdout_output_9800
(base) crisqifawei@CRISQIFAWEI-MC0 8.new_speech_rec %
```

suda-morris, 2023-04-08:
Hi @weiqifa0 the led_strip (ws2812) used to be an example level component but since 5.0, we recommend the user use the **component registry** to add such 3rd party components. For example, if your application wants to use the [led_strip](https://components.espressif.com/components/espressif/led_strip) component, you may need to use the command `idf.py add-dependency "espressif/led_strip^2.3.0"`  under your project. which in return can be reflected in the [menifest file](https://github.com/espressif/esp-idf/blob/master/examples/get-started/blink/main/idf_component.yml).

BTW, you might meet other issues when you upgrade your projects from 4.4 to 5.0, I recommend you read this [migration guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/index.html) first.

Also, here is the video that may help you get started with the component manager: https://www.bilibili.com/video/BV1Ym4y1w7AA/?spm_id_from=333.788&vd_source=5facc390324ce87bd14041128ce34a77

weiqifa0, 2023-04-08:
这个问题已经解决，谢谢你们的及时回复。

closed 2023-04-10 by Alvin1Zhang

WHUWeiYanLin, 2024-08-01:
请问你是怎么解决的呢，我也遇到了同样的问题：
HINT: The component with path specified in the EXTRA_COMPONENT_DIRS variable has been moved to IDF component manager (or has been removed).
Please look out for component in 'https://components.espressif.com' and add using 'idf.py add-dependency' command.
Refer to the migration guide for more details.