espressif/esp-idf #11157: install.sh fails to download packages

Oopas, 2023-04-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v2.x

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Download successful.

### What is the actual behavior?

It seems that each package has downloaded a large portion, but it is not complete.

### Steps to reproduce.

./esp-idf/install.sh

### Build or installation Logs.

```plain
Checking "python" ...
Python 3.10.6
"python" has been detected
Installing ESP-IDF tools
Current system platform: linux-amd64
Selected targets are: esp32c3, esp32, esp32s3, esp32h2, esp32s2
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp32-elf, xtensa-esp32s2-elf, xtensa-esp32s3-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32
Installing xtensa-esp-elf-gdb@11.2_20220823
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz
Destination: /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp
96%WARNING: Download failure <urlopen error retrieval incomplete: got only 25901812 out of 26947336 bytes>
WARNING: file size mismatch for /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp, expected 26947336, got 25901812
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz to /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz
Destination: /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp
96%WARNING: Download failure <urlopen error retrieval incomplete: got only 25903104 out of 26947336 bytes>
WARNING: file size mismatch for /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp, expected 26947336, got 25903104
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz to /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz
Destination: /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp
96%WARNING: Download failure <urlopen error retrieval incomplete: got only 25919488 out of 26947336 bytes>
WARNING: file size mismatch for /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp, expected 26947336, got 25919488
WARNING: Failed to download https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz to /home/none/.espressif/dist/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz.tmp
ERROR: Failed to download, and retry count has expired
Traceback (most recent call last):
  File "/home/none/esp32-arduino-lib-builder/esp-idf/tools/idf_tools.py", line 1976, in <module>
    main(sys.argv[1:])
  File "/home/none/esp32-arduino-lib-builder/esp-idf/tools/idf_tools.py", line 1972, in main
    action_func(args)
  File "/home/none/esp32-arduino-lib-builder/esp-idf/tools/idf_tools.py", line 1509, in action_install
    tool_obj.download(tool_version)
  File "/home/none/esp32-arduino-lib-builder/esp-idf/tools/idf_tools.py", line 764, in download
    raise DownloadError()
__main__.DownloadError
```

### More Information.

There is a proxy on my router (openwrt), but it does not require any settings on the operating system.

I attempted to execute install.sh on ubuntu 22.10 amd64 (Python 3.10.6), ubuntu 18.04 aarch64 (Python 3.7.6 chroot), and android 11 aarch64 (Python 3.10.6 self compiled environment), but both errors were the same

Finally, I tried using Raspberry Pi 3b+(the latest raspberry os armhf) (Python 3.9.2) to execute this script, but it ran normally and without any errors. 🤯

They are all connected to my router at the same time, I really don't know why. Although I can compile with Raspberry Pi 3b+, the efficiency is really too low.

I have found similar issues, but they seem to be caused by Python 2 and are not applicable to me.

CC-MASTER05, 2023-04-08:
please remove the dependent folder .espressif and reinstall all dependent.before use below command make shoure you have add IDF_PATH in your environment 
```
python -m pip install --user -r $IDF_PATH/requirements.txt
```

Oopas, 2023-04-08:
> 

Sadly, this bug has not been solved....

Oopas, 2023-04-08:
> please remove the dependent folder .espressif and reinstall all dependent.before use below command make shoure you have add IDF_PATH in your environment
> 
> ```
> python -m pip install --user -r $IDF_PATH/requirements.txt
> ```

Sadly, this bug has not been solved....

CC-MASTER05, 2023-04-08:
as you are using older version of idf so there are chance those resource not available right now upgrade your ide version5 and verify once

Oopas, 2023-04-13:
> 

I'm very sorry, it seems that I am using the master branch. When I manually download the corresponding file using wget, the script can run normally. This has solved my urgent need.

Oopas, 2023-04-13:
> Hi @Oopas. Do you have a proper Internet connection on your Windows machine?
> 
> `/home/none/.espressif/` - I don't understand these paths. You said that you are using Windows. This doesn't seem to me a Windows path.
> 
> `v2.x` - this doesn't look like the ESP-IDF version. Can you check the proper version please?
> 
> > https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v11.2_20220823/xtensa-esp-elf-gdb-11.2_20220823-x86_64-linux-gnu.tar.gz
> 
> Are you able to download the file shown in the error message with your Internet browser?
> 
> If download works with the Internet browser then you can download the files there and put them inside the `/home/none/.espressif/dist/` directory. After that you can re-run the install script and it won't need to download it anymore.

I'm very sorry, it seems that I am using the master branch. When I manually download the corresponding file using wget, the script can run normally. This has solved my urgent need.

closed 2023-04-14 by dobairoland

jjhax, 2026-10-16:
Why has this been closed? OP's issue was not solved, they had to download manually. I am having the exact same issues. I'm on latest branch. I've tried older branches, same issue.

Looks like macOS users can't install without jumping through hoops. `eim` is an issue too because of a hard coded path to macOS python in the binary.