espressif/esp-idf #11158: many to one communication using esp-ble-mesh vendor model

Ambika-Devi-shobha, 2023-04-08:
### Is your feature request related to a problem?

I'm doing a project requiring 5 servers esp32(slave) to communicate with 1 client esp32(master) to send sensor data. I have done this set with the vendor model consisting of one server and one client. ble_mesh_vendor_model. [https://github.com/espressif/esp-idf/tree/42261df71f12b0b995774d103ce68e40e8fba529/examples/bluetooth/esp_ble_mesh/ble_mesh_vendor_model](URL)
Now I need to increase the server count to 5 that's where I'm stuck!!

### Describe alternatives you've considered.

This is what I need to achieve
[image]

### Additional context.

Hey, developers I kindly request you to help me out, please!!! open to all suggestions and references.

ixtreme3, 2023-05-16:
@Ambika-Devi-shobha did you manage to solve the problem?

Ambika-Devi-shobha, 2023-05-16:
> @Ambika-Devi-shobha did you manage to solve the problem?

No I couldn't

Ambika-Devi-shobha, 2023-05-19:
Are you facing same issue?

ixtreme3, 2023-05-20:
> 

Yes, still don't know how to solve this problem. 

Here are the descriptions of what I want to achieve: 
https://esp32.com/viewtopic.php?f=13&t=33725
https://esp32.com/viewtopic.php?f=13&t=33624

ixtreme3, 2023-05-20:
> 

@Ambika-Devi-shobha  Yes, still don't know how to solve this problem.

Here are the descriptions of what I want to achieve:
https://esp32.com/viewtopic.php?f=13&t=33725
https://esp32.com/viewtopic.php?f=13&t=33624

Ambika-Devi-shobha, 2023-06-28:
@ixtreme3 Same is the problem statement of my project too. I am still searching on this, if you could get solution please let me know.
Here is the follow up discussion regarding this topic one has commented on how to achieve but wasn't successful.
[https://github.com/espressif/esp-idf/issues/9633](url)

Arpt-Singh, 2025-05-08:
Are you able to solve the problem !?! @ixtreme3 @Ambika-Devi-shobha