espressif/esp-idf #11159: TinyUSB stack not ready for c++?

Aggeloz, 2023-04-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to use a library from the ESP-IDF components registry called ArduinoJSON which needs a c++ compiler instead of a C compiler. I changed the main file and the CMake file to compile to C++ but i get this error 
```
error: designator order for field 'gpio_config_t::pull_up_en' does not match declaration order in 'const gpio_config_t'
```
for this piece of code

```C
    const gpio_config_t boot_button_config = {
        .pin_bit_mask = BIT64(APP_BUTTON),
        .mode = GPIO_MODE_INPUT,
        .intr_type = GPIO_INTR_DISABLE,
        .pull_up_en = true,
        .pull_down_en = false,
    };
```

203Null, 2023-04-08:
I use TinyUSB with C++ with no problem. Are you using the TInyUSB inside ESP IDF or an external one?

Aggeloz, 2023-04-08:
Yeah i added it to the project using the add-dependency command from the terminal. Although i have as a base the example from the USB Device project from their github.

Aggeloz, 2023-04-08:
> I use TinyUSB with C++ with no problem. Are you using the TInyUSB inside ESP IDF or an external one?

Yeah i added it to the project using the add-dependency command from the terminal. Although i have as a base the example from the USB Device project from their github.

closed 2023-04-08 by Aggeloz