espressif/esp-idf #11160: ULP ADC - Allow the use of multiple channels

Veldmonster, 2023-04-08:
### Is your feature request related to a problem?

...\components\ulp\ulp_common\ulp_adc.c, function:
ulp_adc_init(const ulp_adc_cfg_t *cfg)
only allows configuration of a single ADC channel.

Invoking it again with a different channel results in an error:
"E (712) adc_oneshot: adc_oneshot_new_unit(93): adc1 is already in use"

### Describe the solution you'd like.

Provide support to initialize more than one adc channel by making ulp_adc_cfg_t bit flags OR
Doing something similar to below (this is my current workaround):

ulp_adc_init(const ulp_adc_cfg_t *cfg, const adc_channel_t channels[], const uint8_t channels_length)
{
    esp_err_t ret = ESP_OK;
    ESP_GOTO_ON_FALSE(cfg, ESP_ERR_INVALID_ARG, err, TAG, "cfg == NULL");
    ESP_GOTO_ON_FALSE(cfg->adc_n == ADC_UNIT_1, ESP_ERR_INVALID_ARG, err, TAG, "Only ADC_UNIT_1 is supported for now");
    //-------------ADC1 Init---------------//
    adc_oneshot_unit_handle_t adc1_handle;
    adc_oneshot_unit_init_cfg_t init_config1 = {
        .unit_id = cfg->adc_n,
        .ulp_mode = cfg->ulp_mode,
    };
    if (init_config1.ulp_mode == ADC_ULP_MODE_DISABLE) {
        /* Default to RISCV for backward compatibility */
        ESP_LOGI(TAG, "No ulp mode specified in cfg struct, default to riscv");
        init_config1.ulp_mode = ADC_ULP_MODE_RISCV;
    }
    ESP_ERROR_CHECK(adc_oneshot_new_unit(&init_config1, &adc1_handle));
    //-------------ADC1 Config---------------//
    adc_oneshot_chan_cfg_t config = {
        .bitwidth = cfg->width,
        .atten = cfg->atten,
    };

    // Multi-Channel
    for (int i = 0; i < channels_length; i++) {
        ESP_ERROR_CHECK(adc_oneshot_config_channel(adc1_handle, channels[i], &config));
    }

    //Calibrate the ADC
#if SOC_ADC_CALIBRATION_V1_SUPPORTED
    adc_set_hw_calibration_code(cfg->adc_n, cfg->atten);
#endif

    esp_sleep_enable_adc_tsens_monitor(true);

err:
    return ret;
}

### Describe alternatives you've considered.

See workaround mentioned in described solution.

mike-2020, 2024-09-20:
Any update on this feature request? This make sense to me as well.

yangminglong, 2025-04-11:
Any update on this feature request? This make sense to me as well.

masterxq, 2025-06-22:
Why is this not possible??
I want to sample more than one pin with the ulp and it was no issue in IDF 4.