espressif/esp-idf #11164: Crash from btdm_controller_task

espradio, 2023-04-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-368-ga9c36f277f

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 devkit

### Power Supply used.

USB

### What is the expected behavior?

No crash

### What is the actual behavior?

Following crash:

Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x4011a99c  PS      : 0x00060930  A0      : 0x80024f7c  A1      : 0x3ffe8cc0  
0x4011a99c: hci_rd_rem_ext_feats_cmd_handler at ??:?

A2      : 0x00000200  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x3ffd25c8  A7      : 0x3ffd25cc  A8      : 0x8011a995  A9      : 0x3ffe8ca0  
A10     : 0x3ffb102c  A11     : 0x00000023  A12     : 0x00000006  A13     : 0x3ffb103a  
A14     : 0x3ffb1020  A15     : 0x00000000  SAR     : 0x00000019  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x000002f6  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x4011a999:0x3ffe8cc0 0x40024f79:0x3ffe8cf0 0x40086518:0x3ffe8d10 0x40019d11:0x3ffe8d40 0x40055b4d:0x3ffe8d60 0x40114feb:0x3ffe8d80 0x40115649:0x3ffe8da0 0x40097dfd:0x3ffe8dd0
0x4011a999: hci_rd_rem_ext_feats_cmd_handler at ??:?

0x40086518: ke_task_schedule at ke_task.c:?

0x40114feb: r_rw_schedule at ??:?

0x40115649: btdm_controller_task at ??:?

0x40097dfd: vPortTaskWrapper at /export/iradio/esp-idf/components/freertos/port/xtensa/port.c:142

### Steps to reproduce.

1)Happens some times(1 out of 13) when connecting to Bluetooth speaker.
2) App using WIFI + Bluetooth coexistence

### Debug Logs.

```plain
I (27) boot: ESP-IDF v4.4.4-368-ga9c36f277f 2nd stage bootloader
[... 135 lines cut ...]
CONFIG_BTDM_CTRL_PINNED_TO_CORE_1=y
CONFIG_BTDM_CTRL_PINNED_TO_CORE=1
CONFIG_BT_BLUEDROID_PINNED_TO_CORE_1=y
CONFIG_BT_BLUEDROID_PINNED_TO_CORE=1
CONFIG_BT_BTC_TASK_STACK_SIZE=3072
CONFIG_BT_BLUEDROID_MEM_DEBUG=n
CONFIG_A2DP_SINK_ENABLE=n
CONFIG_BT_STACK_NO_LOG=n
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=n

# WIFI + BT
CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE=y

# WIFI Rx perf improvement
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=64
CONFIG_ESP32_WIFI_AMPDU_RX_ENABLED=y
CONFIG_ESP32_WIFI_RX_BA_WIN=32

# LWIP Rx perf improvement
CONFIG_LWIP_TCP_WND_DEFAULT=65535
CONFIG_LWIP_TCP_RECVMBOX_SIZE=64
CONFIG_LWIP_UDP_RECVMBOX_SIZE=64
CONFIG_LWIP_TCPIP_RECVMBOX_SIZE=64
```

xiongweichao, 2023-04-10:
Hi @espradio ,

Please try to see if using this lib can solve the problem.
replace path: $IDF_PATH/components/bt/controller/lib_esp32
[esp32.zip](https://github.com/espressif/esp-idf/files/11188599/esp32.zip)

Thanks

espradio, 2023-04-10:
That is fixing it. looking forward for official update to v4.4.4 and latest branches.

Alvin1Zhang, 2023-04-25:
Thanks for reporting, fix is available, feel free to reopen.

closed 2023-04-25 by Alvin1Zhang