espressif/esp-idf #11165: The example rgb_panel in v5.0.1 FPS is too slow at 800X480

iincity, 2023-04-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

IDF version.
v5.0

Operating System used.
Windows

How did you build your project?
VS Code IDE

If you are using Windows, please specify command line type.
None

Development Kit.
ESP32-S3-WROOM-1

Power Supply used.
External 5V

What is the expected behavior?
Run the example code and see something on the display

The same hardware tested with Arduino example (driver: Arduino_GFX, Arduino_ESP32RGBPanel, Arduino_RPi_DPI_RGBPanel), the frame rate is exactly twice that of the rgb_pannel, the CPU usage is still low, the arduino is 6% cpu used and 66 FPS, rgb_pannel is 17% CPU used and 33FPS , when config double frame buffer, it just 7FPS, what is the wrong configuration?

run log:
rgp_panel:
Executing action: monitor
Serial port COM4
Connecting....
Detecting chip type... ESP32-S3
Running idf_monitor in directory d:\espressif\frameworks\esp-idf-v5.0.1\examples\peripherals\lcd\rgb_panel
Executing "D:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe D:/Espressif/frameworks/esp-idf-v5.0.1\tools/idf_monitor.py -p COM4 -b 115200 --toolchain-prefix xtensa-esp32s3-elf- --target esp32s3 d:\espressif\frameworks\esp-idf-v5.0.1\examples\peripherals\lcd\rgb_panel\build\rgb_panel.elf --force-color -m 'D:\Espressif\python_env\idf5.0_py3.8_env\Scripts\python.exe' 'D:/Espressif/frameworks/esp-idf-v5.0.1\tools\idf.py'"...
--- WARNING: GDB cannot open serial ports accessed as COMx
--- Using \\.\COM4 instead...
--- idf_monitor on \\.\COM4 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x164c
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2ef8
entry 0x403c9900
I (25) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (25) boot: compile time 15:40:34
I (25) boot: chip revision: v0.1
I (27) boot.esp32s3: Boot SPI Speed : 80MHz
I (32) boot.esp32s3: SPI Mode : DIO
I (37) boot.esp32s3: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label Usage Type ST Offset Length
I (58) boot: 0 nvs WiFi data 01 02 00009000 00006000
I (65) boot: 1 phy_init RF data 01 01 0000f000 00001000
I (73) boot: 2 factory factory app 00 00 00010000 00100000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=2567ch (153212) map
I (120) esp_image: segment 1: paddr=000356a4 vaddr=3fc93600 size=031d8h ( 12760) load
I (123) esp_image: segment 2: paddr=00038884 vaddr=40374000 size=07794h ( 30612) load
I (133) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=4cf7ch (315260) map
I (191) esp_image: segment 4: paddr=0008cfa4 vaddr=4037b794 size=07e28h ( 32296) load
[... 168 lines cut ...]
I (1454) example: Install RGB LCD panel driver
V (1504) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1504) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x90E
D (1514) intr_alloc: Connected src 24 to int 17 (cpu 0)
D (1514) gdma: new group (0) at 0x3c1f7944
D (1524) gdma: new pair (0,0) at 0x3c1f7980
D (1524) gdma: new tx channel (0,0) at 0x3c13c104
D (1524) gdma: tx channel (0,0), (4:64) bytes aligned, burst enabled
V (1534) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): checking args
V (1544) intr_alloc: esp_intr_alloc_intrstatus (cpu 0): Args okay. Resulting flags 0x80E
D (1554) intr_alloc: Connected src 71 to int 18 (cpu 0)
D (1554) gdma: install interrupt service for tx channel (0,0)
D (1564) lcd_panel.rgb: new rgb panel(0) @0x3fc983e0, num_fbs=2, fb_size=768000, bb0 @0x3fc9859c, bb1 @0x3fca0100, bb_size=16000
D (1574) lcd_panel.rgb: fb[0] @0x3c080900
D (1574) lcd_panel.rgb: fb[1] @0x3c13c140
I (1584) example: Register event callbacks
I (1584) example: Initialize RGB LCD panel
D (1594) lcd_hal: n=4,a=9,b=4,mo=2
D (1594) lcd_panel.rgb: rgb panel(0) start, pclk=18000000Hz
I (1604) example: Turn on LCD backlight
I (1604) example: Initialize LVGL library
I (1604) example: Allocate separate LVGL draw buffers from PSRAM
I (1614) example: Register display driver to LVGL
I (1624) example: Install LVGL tick timer
I (1624) example: Display LVGL Scatter Chart

espzav, 2023-04-25:
Hi @iincity, we are working on the document, which should help to increase the performance with LVGL and screens. It is not finished yet, but the PR is here: https://github.com/espressif/esp-bsp/pull/146 you can look inside and try to set it. The examples will be updated later.

iincity, 2023-04-26:
The Audino RGB:LCD: 4.3" example can reach 6CPU and 66fps.  Lib version is  ESP32 Arduino 2.0.8 based on ESP-IDF 4.4.4.

espzav, 2023-04-26:
Hi @iincity, If you want to try the example which is updated for better performance with RGB display and IDF, it is here: https://github.com/espressif/esp-bsp/tree/master/examples/display_lvgl_demos

iincity, 2023-04-30:
> Hi @iincity, If you want to try the example which is updated for better performance with RGB display and IDF, it is here: https://github.com/espressif/esp-bsp/tree/master/examples/display_lvgl_demos

I am use the lcd_rgb_panel, It does not work, it still 33fps, The sdkconfig.h:
/*
 * Automatically generated file. DO NOT EDIT.
 * Espressif IoT Development Framework (ESP-IDF) Configuration Header
 */
#pragma once
#define CONFIG_SOC_LEDC_SUPPORT_APB_CLOCK 1
#define CONFIG_SOC_LEDC_SUPPORT_XTAL_CLOCK 1
#define CONFIG_SOC_LEDC_CHANNEL_NUM 8
#define CONFIG_SOC_LEDC_TIMER_BIT_WIDE_NUM 14
#define CONFIG_SOC_LEDC_SUPPORT_FADE_STOP 1
#define CONFIG_SOC_MPU_MIN_REGION_SIZE 0x20000000
#define CONFIG_SOC_MPU_REGIONS_MAX_NUM 8
#define CONFIG_SOC_ADC_SUPPORTED 1
#define CONFIG_SOC_PCNT_SUPPORTED 1
#define CONFIG_SOC_WIFI_SUPPORTED 1
#define CONFIG_SOC_TWAI_SUPPORTED 1
#define CONFIG_SOC_GDMA_SUPPORTED 1
#define CONFIG_SOC_LCDCAM_SUPPORTED 1
#define CONFIG_SOC_MCPWM_SUPPORTED 1
#define CONFIG_SOC_DEDICATED_GPIO_SUPPORTED 1
#define CONFIG_SOC_CACHE_SUPPORT_WRAP 1
#define CONFIG_SOC_ULP_SUPPORTED 1
#define CONFIG_SOC_RISCV_COPROC_SUPPORTED 1
#define CONFIG_SOC_BT_SUPPORTED 1
#define CONFIG_SOC_USB_OTG_SUPPORTED 1
#define CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED 1
#define CONFIG_SOC_CCOMP_TIMER_SUPPORTED 1
#define CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED 1
#define CONFIG_SOC_SUPPORTS_SECURE_DL_MODE 1
#define CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD 1
#define CONFIG_SOC_SDMMC_HOST_SUPPORTED 1
#define CONFIG_SOC_RTC_FAST_MEM_SUPPORTED 1
#define CONFIG_SOC_RTC_SLOW_MEM_SUPPORTED 1
#define CONFIG_SOC_RTC_MEM_SUPPORTED 1
#define CONFIG_SOC_PSRAM_DMA_CAPABLE 1
#define CONFIG_SOC_XT_WDT_SUPPORTED 1
#define CONFIG_SOC_I2S_SUPPORTED 1
#define CONFIG_SOC_RMT_SUPPORTED 1
#define CONFIG_SOC_SDM_SUPPORTED 1
#define CONFIG_SOC_SYSTIMER_SUPPORTED 1
#define CONFIG_SOC_SUPPORT_COEXISTENCE 1
#define CONFIG_SOC_TEMP_SENSOR_SUPPORTED 1
#define CONFIG_SOC_AES_SUPPORTED 1
#define CONFIG_SOC_MPI_SUPPORTED 1
#define CONFIG_SOC_SHA_SUPPORTED 1
#define CONFIG_SOC_HMAC_SUPPORTED 1
#define CONFIG_SOC_DIG_SIGN_SUPPORTED 1
#define CONFIG_SOC_FLASH_ENC_SUPPORTED 1
#define CONFIG_SOC_SECURE_BOOT_SUPPORTED 1
#define CONFIG_SOC_MEMPROT_SUPPORTED 1
#define CONFIG_SOC_TOUCH_SENSOR_SUPPORTED 1
#define CONFIG_SOC_XTAL_SUPPORT_40M 1
#define CONFIG_SOC_APPCPU_HAS_CLOCK_GATING_BUG 1
#define CONFIG_SOC_ADC_RTC_CTRL_SUPPORTED 1
#define CONFIG_SOC_ADC_DIG_CTRL_SUPPORTED 1
#define CONFIG_SOC_ADC_ARBITER_SUPPORTED 1
#define CONFIG_SOC_ADC_FILTER_SUPPORTED 1
#define CONFIG_SOC_ADC_MONITOR_SUPPORTED 1
#define CONFIG_SOC_ADC_DMA_SUPPORTED 1
#define CONFIG_SOC_ADC_PERIPH_NUM 2
#define CONFIG_SOC_ADC_MAX_CHANNEL_NUM 10
#define CONFIG_SOC_ADC_ATTEN_NUM 4
#define CONFIG_SOC_ADC_DIGI_CONTROLLER_NUM 2
#define CONFIG_SOC_ADC_PATT_LEN_MAX 24
#define CONFIG_SOC_ADC_DIGI_MIN_BITWIDTH 12
#define CONFIG_SOC_ADC_DIGI_MAX_BITWIDTH 12
#define CONFIG_SOC_ADC_DIGI_RESULT_BYTES 4
[... 878 lines cut ...]
I (526) cpu_start: Project name:     display_lvgl_demos
I (532) cpu_start: App version:      squareline-latest-35-gedc558b-d
I (539) cpu_start: Compile time:     Apr 30 2023 10:02:01
I (545) cpu_start: ELF file SHA256:  27b728969ce9dcca...
I (551) cpu_start: ESP-IDF:          v5.0.1-dirty
I (556) cpu_start: Min chip rev:     v0.0
I (561) cpu_start: Max chip rev:     v0.99
I (566) cpu_start: Chip rev:         v0.1
I (570) heap_init: Initializing. RAM available for dynamic allocation:
I (578) heap_init: At 3FC9AA40 len 0004ECD0 (315 KiB): D/IRAM
I (584) heap_init: At 3FCE9710 len 00005724 (21 KiB): STACK/DRAM
I (591) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (597) heap_init: At 600FE010 len 00001FF0 (7 KiB): RTCRAM
I (603) esp_psram: Adding pool of 7680K of PSRAM memory to heap allocator
I (611) spi_flash: detected chip: gd
I (615) spi_flash: flash io: qio
W (619) spi_flash: Detected size(16384k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (632) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (648) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (657) example: Turn off LCD backlight
I (661) gpio: GPIO[2]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (671) example: Install RGB LCD panel driver
GetOverlappedResult failed (PermissionError(13, '???????????????', None, 5))
Waiting for the device to reconnect