espressif/esp-idf #11166: 程序在解析json数据的时候，内核0恐慌，可能是什么原因造成的呢？

king592, 2023-04-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF5.0.1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32S3

### Power Supply used.

USB

### What is the expected behavior?

程序在解析json数据的时候，内核0恐慌，设备复位，可能是什么原因造成的呢？堆栈溢出吗？

### What is the actual behavior?

设备在解析JSON数据的时候复位

### Steps to reproduce.

void Tencent_dis_model(char *topic,char *data)
{
	char topic_dat[50]={0};

	sprintf(topic_dat,"$thing/down/property/%s/%s",ProductID,DeviceName);
	if(strncmp(topic_dat,topic,strlen(topic_dat))!=0)
	{
		ESP_LOGI("TEN", "len=%d", strcmp(topic,topic_dat));
		return;
	}
	cJSON* cjson = cJSON_Parse(data);//将JSON字符串转换成JSON结构体

	if(cjson==NULL)
	{
		ESP_LOGI("TEN:", "no json");
		return;
	}

    cJSON *params = cJSON_GetObjectItem(cjson,"params");	//解析对象
	ESP_LOGI("HEAP1","size = %ld\n", esp_get_free_heap_size());
	cJSON* jsonItem = cJSON_GetObjectItem(cjson,"params")->child; //子对象    代码在这里死机复位了
	ESP_LOGI("HEAP2","size = %ld\n", esp_get_free_heap_size());
	while(jsonItem != NULL)
	{
		ESP_LOGI("HEAP3","size = %ld\n", esp_get_free_heap_size());
		ESP_LOGI("Ten","%s\n", jsonItem->string);
		ESP_LOGI("HEAP4","size = %ld\n", esp_get_free_heap_size());
		// write_driver(jsonItem->string,params);
        jsonItem = jsonItem->next;//指向下一个对象
	}
	cJSON_Delete(cjson);//清除结构体
}

### Debug Logs.

```plain
I (59176) HEAP1: size = 134832

Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x4202319a  PS      : 0x00060130  A0      : 0x8202385c  A1      : 0x3fcc77b0
0x4202319a: Tencent_dis_model at F:/work/Project/Sank-HUB/code/Sank-HUB_1.1/main/Tencent_iot/Tencent_iot.c:191

A2      : 0x3c0d6c58  A3      : 0x3fcc85f0  A4      : 0x3c0d6e18  A5      : 0x0000e728
A6      : 0x3c10f578  A7      : 0x0000cdcd  A8      : 0x8202319a  A9      : 0x3fcc7790
A10     : 0x00000000  A11     : 0x3c0d6c58  A12     : 0x3c0d6e18  A13     : 0x0000e728
A14     : 0x3c0d6e10  A15     : 0x00020eb0  SAR     : 0x00000004  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000008  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xfffffffe

Backtrace: 0x42023197:0x3fcc77b0 0x42023859:0x3fcc7810 0x420c9fad:0x3fcc7830 0x420c9a52:0x3fcc7860 0x4202a10e:0x3fcc78a0 0x4202a760:0x3fcc78c0 0x4202acc7:0x3fcc78f0 0x4202b1e4:0x3fcc7920 0x40381ff1:0x3fcc7950
0x42023197: Tencent_dis_model at F:/work/Project/Sank-HUB/code/Sank-HUB_1.1/main/Tencent_iot/Tencent_iot.c:191

0x42023859: mqtt_event_handler at F:/work/Project/Sank-HUB/code/Sank-HUB_1.1/main/MQTT/MQTT.c:52 (discriminator 13)

0x420c9fad: handler_execute at G:/ESP-IDF/esp-idf/components/esp_event/esp_event.c:137

0x420c9a52: esp_event_loop_run at G:/ESP-IDF/esp-idf/components/esp_event/esp_event.c:582 (discriminator 3)

0x4202a10e: esp_mqtt_dispatch_event at G:/ESP-IDF/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:953

0x4202a760: deliver_publish at G:/ESP-IDF/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1020 (discriminator 4)

0x4202acc7: mqtt_process_receive at G:/ESP-IDF/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1291 (discriminator 15)

0x4202b1e4: esp_mqtt_task at G:/ESP-IDF/esp-idf/components/mqtt/esp-mqtt/mqtt_client.c:1522

0x40381ff1: vPortTaskWrapper at G:/ESP-IDF/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:154

ELF file SHA256: 4a125f407627cb63

Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375a3d
0x40375a3d: esp_restart_noos_dig at G:/ESP-IDF/esp-idf/components/esp_system/esp_system.c:64 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x1648
load:0x403c9700,len:0xbe0
load:0x403cc700,len:0x2ef8
entry 0x403c9900
```

Silicon-wzh, 2023-04-09:
@king592 应该是 cJSON_GetObjectItem(cjson,"params")  返回空指针了

分析：在coredump的信息中，PC: 0x4202319a 表示发生异常时的 PC 位置，LoadProhibited 表示异常原因是load数据时发生错误，取数据的地址为: EXCVADDR: 0x00000008 因此判定应该是cJSON_GetObjectItem 返回空指针了。

参考：https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/fatal-errors.html#guru-meditation-errors

king592, 2023-04-10:
是的确实是这个问题，非常感谢

closed 2023-04-10 by igrr