espressif/esp-idf #11167: Linking CXX executable failed/ undefined reference to `tud_hid_n_report'

Aggeloz, 2023-04-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, ive been having an issue with breaking apart my code to different files. I use C++ and the libraries TinyUSB and ArduinoJSON. I have 3 files of code, the main one, one for my rotary encoder and one for the USB. Ever since i put the USB code in a separate file i cant compile and i get a huge error message. I get that it somehow cant source some part of the TinyUSB library but i dont understand why.

Also in the main code file i use extern "C" to be able to use my code in C++ mode. I of course changed the CMake file to include all 3 of the code files.

The undefined reference refers to this line of code : 

```C
tud_hid_report(0, "test", 4);
```

Error in the CLI:

```shell
[1/1] cd /home/edgie/esp/read_rotary/build/bootloader/esp-idf/esptool_py && /home/edgie/.espr...--offset 0x8000 bootloader 0x1000 /home/edgie/esp/read_rotary/build/bootloader/bootloader.binBootloader binary size 0x5440 bytes. 0x1bc0 bytes (25%) free.
[2/4] Linking CXX executable read_rotary.elfFAILED: read_rotary.elf 
: && /home/edgie/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/xtensa-esp32s2-elf-g++ -mlongcalls  -g -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32S2=0 -Wl,--Map=/home/edgie/esp/read_rotary/build/read_rotary.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32s2.peripherals.ld -T esp32s2.rom.ld -T esp32s2.rom.api.ld -T esp32s2.rom.libgcc.ld -T esp32s2.rom.newlib-funcs.ld -T esp32s2.rom.newlib-data.ld -T esp32s2.rom.spiflash.ld -T memory.ld -T sections.ld CMakeFiles/read_rotary.elf.dir/project_elf_src_esp32s2.c.obj -o read_rotary.elf -L/home/edgie/esp/esp-idf/components/soc/esp32s2/ld   -L/home/edgie/esp/esp-idf/components/esp_rom/esp32s2/ld   -L/home/edgie/esp/read_rotary/build/esp-idf/esp_system/ld   -L/home/edgie/esp/esp-idf/components/esp_phy/lib/esp32s2   -L/home/edgie/esp/esp-idf/components/esp_wifi/lib/esp32s2 esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/fatfs/libfatfs.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/perfmon/libperfmon.a  esp-idf/spiffs/libspiffs.a  esp-idf/touch_element/libtouch_element.a  esp-idf/ulp/libulp.a  esp-idf/usb/libusb.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/espressif__tinyusb/libespressif__tinyusb.a  esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/unity/libunity.a  esp-idf/protocomm/libprotocomm.a  esp-idf/console/libconsole.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/espressif__esp_tinyusb/libespressif__esp_tinyusb.a  esp-idf/fatfs/libfatfs.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/usb/libusb.a  esp-idf/espressif__tinyusb/libespressif__tinyusb.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/ulp/libulp.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  /home/edgie/esp/esp-idf/components/esp_wifi/lib/esp32s2/libcore.a  /home/edgie/esp/esp-idf/components/esp_wifi/lib/esp32s2/libespnow.a  /home/edgie/esp/esp-idf/components/esp_wifi/lib/esp32s2/libmesh.a  /home/edgie/esp/esp-idf/components/esp_wifi/lib/esp32s2/libnet80211.a  /home/edgie/esp/esp-idf/components/esp_wifi/lib/esp32s2/libpp.a  /home/edgie/esp/esp-idf/components/esp_wifi/lib/esp32s2/libsmartconfig.a  /home/edgie/esp/esp-idf/components/esp_wifi/lib/esp32s2/libwapi.a  esp-idf/xtensa/libxtensa.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/efuse/libefuse.a  esp-idf/driver/libdriver.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/pthread/libpthread.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/cxx/libcxx.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/ulp/libulp.a  esp-idf/mbedtls/mb [... cut ...]
/home/edgie/.espressif/tools/xtensa-esp32s2-elf/esp-12.2.0_20230208/xtensa-esp32s2-elf/bin/../lib/gcc/xtensa-esp32s2-elf/12.2.0/../../../../xtensa-esp32s2-elf/bin/ld: esp-idf/main/libmain.a(read_rotary.cpp.obj): in function `_ZL17app_send_hid_demov':
/home/edgie/esp/read_rotary/main/setUsb.cpp:107: undefined reference to `tud_hid_n_report'
collect2: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/edgie/esp/read_rotary/build/log/idf_py_stderr_output_35804 and /home/edgie/esp/read_rotary/build/log/idf_py_stdout_output_35804

```

0xjakob, 2023-04-10:
Hi @Aggeloz, the function you provided is `tud_hid_report`, but the linker is looking for `tud_hid_n_report`. Is this a typo or does it actually reflect your issue?

Aggeloz, 2023-04-10:
> Hi @Aggeloz, the function you provided is `tud_hid_report`, but the linker is looking for `tud_hid_n_report`. Is this a typo or does it actually reflect your issue?

It's not a typo `tud_hid_n_report` is called by `tud_hid_report`internally from inside the TinyUSB library.

0xjakob, 2023-04-10:
Can you confirm that the symbol `tud_hid_n_report` actually exists? I.e., check the output of `xtensa-esp32s2-elf-objdump -t <library_containing_tud_hid_n_report>` for `tud_hid_n_report`.

Aggeloz, 2023-04-10:
> Can you confirm that the symbol `tud_hid_n_report` actually exists? I.e., check the output of `xtensa-esp32s2-elf-objdump -t <library_containing_tud_hid_n_report>` for `tud_hid_n_report`.

Where is the object of the compiled library supposed to be stored?

Aggeloz, 2023-04-10:
Ill leave this issue open but i want to add that when i copy paste the example tinyUSB HID project, the project compiles and runs but when i create one it doesn't. I dont understand what the issue is.

0xjakob, 2023-04-11:
> cant run this command on the problematic project because it will not compile at all

The output looks like it compiled fine already, but only the linking fails (correct me if this is wrong). At this point, all components are built already, hence you can examine their libraries.

If you could provide a simple reproducer, it would probably be easy for us to figure out what is wrong.

Aggeloz, 2023-04-11:
> The output looks like it compiled fine already, but only the linking fails (correct me if this is wrong). At this point, all components are built already, hence you can examine their libraries.
>
> If [...]

What do you mean by reproducer? If you mean uploading the project to a repo i can do that easily.

0xjakob, 2023-04-12:
> What do you mean by reproducer?

A simple example, possibly without any unnecessary dependencies, that reproduces the issue and that we can check and run on our side, that would be great.

Aggeloz, 2023-04-12:
> > What do you mean by reproducer?
> 
> A simple example, possibly without any unnecessary dependencies, that reproduces the issue and that we can check and run on our side, that would be great.

I've made this repo, you can download it and just try to build it. I have set the target as an ESP32-S2.

https://github.com/Aggeloz/usb_problematic

Aggeloz, 2023-04-12:
> Could you try enabling HID support in TinyUSB by setting CONFIG_TINYUSB_HID_COUNT=1 in sdkconfig?

That actually fixed it, i have no idea how i skipped over this setting. Thank you very much.

closed 2023-04-12 by Aggeloz