espressif/esp-idf #11168: ESP32-C3 ADC Threshold (monitor mode)

Cuongha1995, 2023-04-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Interrupts work in monitor mode　（function : isr_ADC()）
code : 

//ADC
#define GPIO_INPUT_LOADSTS	ADC1_CHANNEL_4

#define ADC_CONV_LIMIT_EN   0
#define ADC_CONV_MODE       ADC_CONV_SINGLE_UNIT_1
#define ADC_OUTPUT_TYPE     ADC_DIGI_OUTPUT_FORMAT_TYPE2
#define ADC_DIGI_MONITOR0_H	1500
#define ADC_DIGI_MONITOR0_L	0

#define ADC_DIGI_MONITOR1_H	4096
#define ADC_DIGI_MONITOR1_L	1500

#define GET_UNIT(x)        ((x>>3) & 0x1)

static intr_handle_t adc_isr_handle = NULL;

/* -- Definition functions for internal use ----------------- */
void CALCUL_adc_init(uint16_t adc1_chan_mask, adc_channel_t *channel, uint8_t channel_num);
//void CALCUL_adc_read(void *arg);
static void IRAM_ATTR isr_ADC(void *args);

/* -- Entity of variables for internal use ------------------ */
static uint16_t adc1_chan_mask = BIT(4);
static uint16_t adc2_chan_mask = 0;
static adc_channel_t channel[1] = {GPIO_INPUT_LOADSTS};

void TKTble_init (sBLE_STS *para)
{	//adc
	CALCUL_adc_init(adc1_chan_mask, channel, sizeof(channel) / sizeof(adc_channel_t));
	adc1_config_width(ADC_WIDTH_BIT_12);
	adc_digi_start();
}

/****************************************************************************************/
/*FUNCTION:		CALCUL_adc1_init																*/
/*DESCRIPTION:																			*/
/*PARAMETERS:																			*/
/*																						*/
/*RETURN:																				*/
/****************************************************************************************/
void CALCUL_adc_init(uint16_t adc1_chan_mask, adc_channel_t *channel, uint8_t channel_num)
{
	adc_digi_init_config_t adc_dma_config = {
        .max_store_buf_size = ADC_TIMES,
        .conv_num_each_intr = ADC_TIMES,
        .adc1_chan_mask = adc1_chan_mask,
		.adc2_chan_mask = adc2_chan_mask,
    };
    ESP_ERROR_CHECK(adc_digi_initialize(&adc_dma_config));

    adc_digi_configuration_t dig_cfg = {
        .conv_limit_en = ADC_CONV_LIMIT_EN,
        .conv_limit_num = 250,
        .sample_freq_hz = 50 * 1000,//50khz (0~83333)//channel
        .conv_mode = ADC_CONV_MODE,
        .format = ADC_OUTPUT_TYPE,
    };

    adc_digi_pattern_config_t adc_pattern[1] = {0};
    dig_cfg.pattern_num = channel_num;

    for (int i = 0; i < channel_num; i++) {
    	uint8_t unit = GET_UNIT(channel[i]);
        uint8_t ch = channel[i] & 0x7;
        adc_pattern[i].atten = ADC_ATTEN_DB_11;
        adc_pattern[i].channel = ch;
        adc_pattern[i].unit = unit;
        adc_pattern[i].bit_width = SOC_ADC_DIGI_MAX_BITWIDTH;
    }
    dig_cfg.adc_pattern = adc_pattern;
    adc_digi_controller_configure(&dig_cfg);

    adc_digi_monitor_t mcfg;
	mcfg.adc_unit = ADC_UNIT_1;
	mcfg.channel = ADC_CHANNEL_4;
	mcfg.mode = ADC_DIGI_MONITOR_EN;
	mcfg.h_threshold = 2000;
	mcfg.l_threshold = 0;

	adc_digi_monitor_set_config(ADC_DIGI_MONITOR_IDX0, &mcfg);
	adc_digi_monitor_enable(ADC_DIGI_MONITOR_IDX0, true);

	esp_intr_alloc(ETS_APB_ADC_INTR_SOURCE, 0, isr_ADC, NULL, &adc_isr_handle);//ETS_APB_ADC_INTR_SOURCE
	APB_SARADC.int_ena.thres0_high = 1;
	APB_SARADC.int_ena.thres1_high = 1;
	APB_SARADC.int_ena.thres0_low = 1;
	APB_SARADC.int_ena.thres1_low = 1;
	APB_SARADC.thres_ctrl.thres_all_en = 1;
}

static void IRAM_ATTR isr_ADC(void *args)
{
//	APB_SARADC.int_ena.thres0_high = 0;
	printf("isr ADC\n");//check functions
}

where is the mistake⁇

### What is the actual behavior?

Interrupts didn't work in monitor mode

### Steps to reproduce.

Read ADC : checked (result : OK)

wanckl, 2023-04-11:
@Cuongha1995 Hello! ncie to meet you here!   
seems your code no problem   
I also have a try on this, I remember that you need set register `APB_SARADC.thres_ctrl.thres0_en` and `xxx.thres1_en` independently to enable it, `thres_all_en` have no effect, don't know why,,, :dog2: besides, `printf` used in ISR will crash, you can use `ESP_EARLY_LOGI` instead.   
while I heard that this feature is in developing by espressif official, maybe we can use it on master recently :+1:

Cuongha1995, 2023-04-11:
@wanckl Hello! Nice to meet you too
Thank you for confirming this.
I understand the above, but the results of implementing another Monitor0 or Monitor1 are the same.
Is there a problem with this function esp_intr_alloc()?
If possible, could you please share the source code that worked?
Best regards.

wanckl, 2023-04-12:
@Cuongha1995 
emmmm, I think you forgot the `esp_intr_enable()`.

wanckl, 2023-04-14:
@Cuongha1995 Hi, is your code working? I think I have no point yet

abellofiore, 2023-04-14:
Coincidentally I was starting some testing using ADC thresholding myself and came upon this thread.
I do have this working on the ESP32-C3.  Consider these changes:

1. Your thresholds might be wrong.  For instance, to catch a range between 100 and 200 counts, you would set l_threshold=200 and h_threshold=100; then, BOTH interrupts must be asserted at the same time to be within range (not sure what you were going for so you may be OK if you only wanted to capture counts >= 2000).
2. You must set APB_SARADC.thres_ctrl.thresX_en and APB_SARADC.thres_ctrl.thres_all_en=1, as mentioned.
3. You should call esp_intr_enable(adc_isr_handle), as mentioned.
4. You must call adc_digi_start() to get this to work.
5. In the ISR, you should check for and clear the interrupt sources (in my code, I only care about thresh0 high/low):

`
uint32_t thres_high_counter = 0;
uint32_t thres_low_counter = 0;
uint32_t in_between_counter = 0;

static void IRAM_ATTR isr_ADC(void *args)
{
	int x = 0;
	if( APB_SARADC.int_st.thres0_high )
	{
		++thres_high_counter;
		APB_SARADC.int_clr.thres0_high = 1;
		x++;
	}
	if( APB_SARADC.int_st.thres0_low )
	{
		++thres_low_counter;
		APB_SARADC.int_clr.thres0_low = 1;
		x++;
	}

	if( x == 2 )
		++in_between_counter;
}
`

Cuongha1995, 2023-04-15:
> @Cuongha1995 Hi, is your code working? I think I have no point yet

@wanckl　Thank you for the advice
I forgot this function esp_intr_enable()
Currently working as designed.
Thanks you so much

Cuongha1995, 2023-04-15:
@abellofiore Thank you for the advice
I forgot this function esp_intr_enable().
Thanks you so much

Alvin1Zhang, 2023-04-17:
Thanks for reporting and sharing the updates. Feel free to reopen.

closed 2023-04-17 by Alvin1Zhang