espressif/esp-idf #11169: Major issue on A2DP with ESP32D0WDR2

0x0fe, 2023-04-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

So the context is a consumer product design which uses an external DAC (ES7148) and provides audio playback via different sources, MMC, wifi and bluetooth (A2DP), initial tests were conducted on ESP32D0WDQ6 and everything worked fine, the A2DP is smooth, no stuttering or anything.

For final implementation we switched to D0WDR2 since we use both wifi and bluetooth and a relatively large application logic, additional PSRAM is required.

Sadly we found that the D0WDR2 cannot work normally for the A2DP, there is a massive stuttering, and the playback is totally abnormal. There is no playback issue when playing audio from MMC.

In order to confirm the problem we performed multiple test, at first changing compile time options such as Flash speed (40M and 80M) CPU speed (160M and 240M) PSRAM enabled/disabled, and even buffer sizes for I2S and A2DP, we found no difference at all, in all cases the stuttering was still present with the same magnitude on the D0WDR2 and no stuttering at all on D0WDQ6.

Then we decided to make comparative tests with devkit instead of the product hardware, we changed the DAC for one without master clock (MCLK) as i suspected it could be related to MCLK, so we used :
-one devkit with D0WDQ6 + MAX98357A
-one devkit with D0WDR2 + MAX98357A
Identical pinmap, and identical firmware binary flashed in both.

We tested the same way, with an iphone as source and running a simple A2DP firmware (also tried different source such as PC):
-D0WDQ6 : perfectly smooth, no stuttering at all
-D0WDR2 : massive stuttering all over, unusable

With the log in verbose mode, there is absolutely no error or warning of any kind, both output exactly the same thing. So the problem lies somewhere deep and is not a simple buffer underrun or something (well unless it doesnt gets catched).

Sadly, we can now confirm the problem comes from the D0WDR2 chip itself and not the hardware, the DAC or the firmware.
I have no idea yet of what causes this, nor what exactly changed in the die between D0WDQ6 and D0WDR2, both are rev 3 so i would have expected the D0WDR2 just be a D0WDQ6 with a PSRAM attached, but apaprently it is not the case ,I2S block? wifi stack? who knows.

So please espressif have a look at that. Attached is a quick audio recording of the problem (from an iphone).

[Dowdr2 a2dp stuttering.zip](https://github.com/espressif/esp-idf/files/11192274/Dowdr2.a2dp.stuttering.zip)

wmy-espressif, 2023-04-12:
Hi @0x0fe , thanks for your question.  I am a Bluetooth guy and I will check the Bluetooth stack for you first.

I would like to ask some questions on how to reproduce the issue, so that I can perform some local tests:
1. You mentioned you use a simple A2DP firmware, did you use the a2dp_sink example provided with esp-idf, or you an example in esp-adf? Do you make some changes on the example? I would like to know how I can get a minimum application to reproduce.
2. As far as I know, DOWDR2 has embedded PSRAM(2M), compared with D0WDQ6. You have mentioned you have performed multile tests including enable/disable PSRAM. Can you provide your sdkconfig file used to reproduce this issue, both for the cases that PSRAM is enabled and disabled?

0x0fe, 2023-04-12:
Hello, thank you for the return.
For testing we used this implementation https://github.com/pschatzmann/ESP32-A2DP which is a direct port of the IDF example for A2DP sink. Yes it has been tested with PSRAM enabled and disabled, via the menu in arduino since the firmware was generated here.
By the way, we tested on D0WD too, with external PSRAM (a wrover module) and it works fine, so the problem really only occurs on D0WDR2. IT does not occur on D0WD nor D0WDQ6.

For the firmware, i suggest you try with the same configuration we used, I can provide ready made binaries if you want, If you use a different ADC let me know the I2S mode.On our side for production we use ES7148 and for tests we used MAX98357A. In both case the pinmap is as follow.

```
#define MCK             3
#define WS              5
#define BCK             18
#define DAT             21
#define MUTE            4
```

wmy-espressif, 2023-04-13:
Hi @0x0fe  thanks for the information. I will try to perform some test locally. According to my information, there are some espressif customers that used D0WDR2 in their audio application products.

By the way, how many pieces of D0WDR2 chips have you tested, did they all have this issue and make no differences?

0x0fe, 2023-04-13:
Hi @wmy-espressif 
We have tested several units, so it is not a problem with one units, all D0WDR2 chips we tested shown the same problem, also, as mentioned, it is not structly an I2S problem because when we play audio from other sources, such as SD_MMC there is no problem, the audio is not stuttering. So it only occurs with the A2DP.

xiongweichao, 2023-04-14:
Hi @0x0fe ,

Sorry, I don't know how to run your project. Can you test it in the following way?
```
#include "BluetoothA2DPSink.h"

BluetoothA2DPSink a2dp_sink;

void setup() {

    static i2s_config_t i2s_config = {
      .mode = (i2s_mode_t) (I2S_MODE_MASTER | I2S_MODE_TX),
      .sample_rate = 44100, // updated automatically by A2DP
      .bits_per_sample = (i2s_bits_per_sample_t)32,
      .channel_format = I2S_CHANNEL_FMT_RIGHT_LEFT,
      .communication_format = (i2s_comm_format_t) (I2S_COMM_FORMAT_STAND_I2S),
      .intr_alloc_flags = 0, // default interrupt priority
      .dma_buf_count = 8,
      .dma_buf_len = 64,
      .use_apll = true,
      .tx_desc_auto_clear = true // avoiding noise in case of data unavailability
    };
    i2s_pin_config_t my_pin_config = {
    .bck_io_num = 26,
    .ws_io_num = 22,
    .data_out_num = 25,
    .data_in_num = I2S_PIN_NO_CHANGE
    };
    a2dp_sink.set_pin_config(my_pin_config);

    a2dp_sink.set_i2s_config(i2s_config);
    a2dp_sink.start("MyMusic");
}

void loop() {
  // disable watchdog
}
```

Thanks

0x0fe, 2023-04-14:
ok, let me test

0x0fe, 2023-04-14:
actually, it is even worst. 
[D0WDR2 stuttering.zip](https://github.com/espressif/esp-idf/files/11232160/D0WDR2.stuttering.zip)
I tried from iphone (13 last IOS) and from PC (win10) and i notice that on iphone it cuts completely at regular interval of varying length  (silent), on PC it doesnt cut, but in both case the stuttering is as bad as it can be.

0x0fe, 2023-04-14:
please note, we use 16bit as our DAC only supports 16 and 24 bit, not 32. Also, pin mapping is different than your code, you may want to adjust your test platform to match ours, in case this isue has anything to do with the pin mapping.

xiongweichao, 2023-04-17:
Hi @0x0fe ,

1. Do you have this problem when using other mobile phones (Android system) to test?
2. Some suggestions about the example are as follows:
 a. [write_audio](https://github.com/pschatzmann/ESP32-A2DP/blob/5f87ad290b8e349e13ab0a80739abb908193991f/src/BluetoothA2DPSinkQueued.cpp#L62) should not block. 
    ```BaseType_t rc = xRingbufferSend(s_ringbuf_i2s, (void *)data, size, (portTickType)0);```
 b. [i2s_ringbuffer_size](https://github.com/pschatzmann/ESP32-A2DP/blob/5f87ad290b8e349e13ab0a80739abb908193991f/src/BluetoothA2DPSinkQueued.h#L34) may be too small.
3. Can you enable [bluedroid log](https://github.com/espressif/esp-idf/blob/e8bdaf91986a41678adc6be13888fc037b1acb68/components/bt/host/bluedroid/Kconfig.in#L248)?

BTW, I hope you can provide a test log after the test.

Thanks

0x0fe, 2023-04-17:
Hi @xiongweichao 
1) we have tested with android too, actually on some old android even the D0WD, but on newer android and on IOS / PC it is working fine. On D0WDR2 it is failing with all platform, old android, new Android, new IOS, and PC (win10).

2 a) i am not sure what the suggestion is here, do you want me to verify wether this call is blocking inside A2DP sink queue? 

2 b) as mentioned previously, we have already tested with various size and number of buffers for I2S (and for A2DP too), last time i tested with 8 buffer of 512 byte on the I2S and it was still stuttering a lot. For the sake of it i will test again.

3) ok, i will see how we can enable this log in the arduino core, hopefully this is possible with a flag, otherwise i would have to move the testing code to IDF, which will take time.

xiongweichao, 2023-04-17:
Hi @0x0fe ,

2a) As you said, verify whether it is blocked.
2b) Maybe this test is not important and only test **2a** is needed.

Enabling the [log](https://github.com/espressif/esp-idf/blob/e8bdaf91986a41678adc6be13888fc037b1acb68/components/bt/host/bluedroid/Kconfig.in#L248) is to verify whether there is any packet loss in Bluetooth. If enabling bluedroid log is difficult, I suggest to use idf [a2dp_sink](https://github.com/espressif/esp-idf/tree/v4.4.4/examples/bluetooth/bluedroid/classic_bt/a2dp_sink) example.

Thanks

0x0fe, 2023-04-17:
so, i decided to give a try to IDF 5.0.1, and i can confirm the problem is till here, however enabling the bluetooth logs indees hows a lot of buffering issues : 

```
[0;32mI (30) boot: ESP-IDF v5.0.1 2nd stage bootloader[0m
[0;32mI (30) boot: compile time 21:29:16[0m
[0;32mI (30) boot: chip revision: v3.0[0m
[0;32mI (33) boot.esp32: SPI Speed      : 40MHz[0m
[0;32mI (38) boot.esp32: SPI Mode       : DIO[0m
[0;32mI (42) boot.esp32: SPI Flash Size : 2MB[0m
[0;32mI (47) boot: Enabling RNG early entropy source...[0m
[0;32mI (52) boot: Partition Table:[0m
[0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length[0m
[0;32mI (63) boot:  0 nvs              WiFi data        01 02 00009000 00006000[0m
[0;32mI (70) boot:  1 phy_init         RF data          01 01 0000f000 00001000[0m
[0;32mI (78) boot:  2 factory          factory app      00 00 00010000 00100000[0m
[0;32mI (85) boot: End of partition table[0m
[0;32mI (89) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2b188h (176520) map[0m
[0;32mI (162) esp_image: segment 1: paddr=0003b1b0 vaddr=3ffbdb60 size=042f8h ( 17144) load[0m
[0;32mI (169) esp_image: segment 2: paddr=0003f4b0 vaddr=40080000 size=00b68h (  2920) load[0m
[0;32mI (170) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=a8dech (691692) map[0m
[0;32mI (426) esp_image: segment 4: paddr=000e8e14 vaddr=40080b68 size=17ef0h ( 98032) load[0m
[0;32mI (479) boot: Loaded app from partition at offset 0x10000[0m
[0;32mI (479) boot: Disabling RNG early entropy source...[0m
[0;32mI (490) cpu_start: Pro cpu up.[0m
[0;32mI (491) cpu_start: Starting app cpu, entry point is 0x40081414[0m
[0;32mI (0) cpu_start: App cpu up.[0m
[0;32mI (507) cpu_start: Pro cpu start user code[0m
[0;32mI (507) cpu_start: cpu freq: 160000000 Hz[0m
[0;32mI (507) cpu_start: Application information:[0m
[0;32mI (512) cpu_start: Project name:     a2dp_sink[0m
[0;32mI (517) cpu_start: App version:      v5.0.1[0m
[0;32mI (522) cpu_start: Compile time:     Apr 17 2023 21:28:58[0m
[0;32mI (528) cpu_start: ELF file SHA256:  472feb94cb193eb6...[0m
[0;32mI (534) cpu_start: ESP-IDF:          v5.0.1[0m
[0;32mI (539) cpu_start: Min chip rev:     v0.0[0m
[0;32mI (543) cpu_start: Max chip rev:     v3.99 [0m
[0;32mI (548) cpu_start: Chip rev:         v3.0[0m
[0;32mI (553) heap_init: Initializing. RAM available for dynamic allocation:[0m
[0;32mI (560) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM[0m
[0;32mI (566) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM[0m
[0;32mI (572) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM[0m
[0;32mI (578) heap_init: At 3FFCB480 len 00014B80 (82 KiB): DRAM[0m
[0;32mI (585) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
[0;32mI (591) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
[0;32mI (597) heap_init: At 40098A58 len 000075A8 (29 KiB): IRAM[0m
[0;32mI (605) spi_flash: detected chip: generic[0m
[0;32mI (608) spi_flash: flash io: dio[0m
[0;33mW (612) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.[0m
[0;32mI (627) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (733) BTDM_INIT: BT controller compile version [60aae55][0m
[0;32mI (733) system_api: Base MAC address is not set[0m
[0;32mI (733) system_api: read default base MAC address from EFUSE[0m
[0;32mI (743) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07[0m
[0;33mW (1443) BT_BTC: A2DP Enable with AVRC[0m
[0;32mI (1453) BT_AV: A2DP PROF STATE: Init Complete[0m
[0;32mI (1453) BT_AV: event: 10[0m
[0;32mI (1463) BT_AV: event: 10[0m
[0;32mI (1463) BT_AV: event: 10[0m
[0;32mI (1463) BT_AV: Get delay report value: delay_value: 1200 * 1/10 ms[0m
[0;32mI (1463) BT_AV: Set delay report value: success, delay_value: 1250 * 1/10 ms[0m
[0;33mW (4783) BT_HCI: hcif conn complete: hdl 0x80, st 0x0[0m
[0;32mI (4783) BT_AV: ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT Connected to [a4:c6:f0:85:a9:9a], status: 0x100[0m
[0;33mW (4883) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000[0m
[0;33mW (5113) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 32000[0m
[0;33mW (5223) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000[0m
[0;31mE (5223) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0[0m
[0;32mI (5223) BT_AV: A2DP connection state: Connecting, [a4:c6:f0:85:a9:9a][0m
[0;32mI (5303) RC_TG: AVRC conn_state evt: state 1, [a4:c6:f0:85:a9:9a][0m
[0;32mI (5303) RC_TG: start volume change simulation[0m
[... 455 lines cut ...]
[0;32mI (54953) BT_APP_CORE: ringbuffer underflowed! mode changed: RINGBUFFER_MODE_PREFETCHING[0m
[0;32mI (55003) BT_APP_CORE: ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING[0m
[0;32mI (55153) BT_APP_CORE: ringbuffer underflowed! mode changed: RINGBUFFER_MODE_PREFETCHING[0m
[0;32mI (55203) BT_APP_CORE: ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING[0m
[0;32mI (55303) RC_TG: Volume is set locally to: 74%[0m
[0;32mI (55403) BT_APP_CORE: ringbuffer underflowed! mode changed: RINGBUFFER_MODE_PREFETCHING[0m
[0;33mW (55423) BT_APPL: Sequence numbers error, recv:0x28f, expect:0x289, recount:0x0[0m
[0;32mI (55453) BT_APP_CORE: ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING[0m
[... cut ...]
[0;33mW (56083) BT_APP_CORE: ringbuffer overflowed, ready to decrease data! mode changed: RINGBUFFER_MODE_DROPPING[0m
[0;33mW (56093) BT_APP_CORE: ringbuffer is full, drop this packet![0m
[0;33mW (56093) BT_APP_CORE: ringbuffer is full, drop this packet![0m
[0;33mW (56113) BT_APP_CORE: ringbuffer is full, drop this packet![0m
[0;33mW (56133) BT_APP_CORE: ringbuffer is full, drop this packet![0m
[0;32mI (56133) BT_APP_CORE: ringbuffer data decreased! mode changed: RINGBUFFER_MODE_PROCESSING[0m
[0;32mI (57233) BT_AV: Audio packet count: 600[0m
[0;32mI (59563) BT_AV: Audio packet count: 700[0m

```

0x0fe, 2023-04-17:
so, here i disabled the color formatting for better clarify, i also enabled the driver to us external PSRAM first and allowed dynamic memory alloation, just for test, and i can say the result is even more horrible because now it also completely saturates the PA and the volume is inoperant, that is, i can set it to 5% and it is show in the log, but the actual output on the DAC is still way passed overload.
Not sure how this related to external PSRAM and memory management, because that is the only thing i changed, but this overloading and lost of volume control did not occur before i changed these two settings.

```
I (459) boot: Loaded app from partition at offset 0x10000
I (459) boot: Disabling RNG early entropy source...
I (468) cpu_start: Pro cpu up.
I (468) cpu_start: Starting app cpu, entry point is 0x4008141c
I (0) cpu_start: App cpu up.
I (480) cpu_start: Pro cpu start user code
I (480) cpu_start: cpu freq: 160000000 Hz
I (480) cpu_start: Application information:
I (482) cpu_start: Project name:     a2dp_sink
I (486) cpu_start: App version:      v5.0.1-dirty
I (490) cpu_start: Compile time:     Apr 17 2023 23:09:49
I (495) cpu_start: ELF file SHA256:  a11fc9cde7c57745...
I (500) cpu_start: ESP-IDF:          v5.0.1-dirty
I (505) cpu_start: Min chip rev:     v0.0
I (509) cpu_start: Max chip rev:     v3.99 
I (513) cpu_start: Chip rev:         v3.0
I (517) heap_init: Initializing. RAM available for dynamic allocation:
I (523) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (528) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (533) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (538) heap_init: At 3FFC3DF0 len 0001C210 (112 KiB): DRAM
I (543) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (549) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (554) heap_init: At 40098B78 len 00007488 (29 KiB): IRAM
I (561) spi_flash: detected chip: generic
I (563) spi_flash: flash io: dio
W (566) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (580) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (684) BTDM_INIT: BT controller compile version [60aae55]
I (684) system_api: Base MAC address is not set
I (684) system_api: read default base MAC address from EFUSE
I (684) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (1394) BT_BTC: A2DP Enable with AVRC
I (1404) BT_AV: A2DP PROF STATE: Init Complete
I (1414) BT_AV: event: 10
I (1414) BT_AV: event: 10
I (1414) BT_AV: event: 10
I (1414) BT_AV: event: 10
I (1414) BT_AV: Get delay report value: delay_value: 1200 * 1/10 ms
I (1424) BT_AV: Set delay report value: success, delay_value: 1250 * 1/10 ms
W (15084) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
I (15084) BT_AV: ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT Connected to [a4:c6:f0:85:a9:9a], status: 0x100
W (15194) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
W (15514) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 32000
W (15794) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
E (15804) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (15814) BT_AV: A2DP connection state: Connecting, [a4:c6:f0:85:a9:9a]
I (15824) RC_TG: AVRC conn_state evt: state 1, [a4:c6:f0:85:a9:9a]
I (15824) RC_TG: start volume change simulation
I (15964) BT_AV: A2DP audio stream configuration, codec type: 0
I (15964) BT_AV: Configure audio player: 21-15-2-35
I (15964) BT_AV: Audio player configured, sample rate: 44100
I (15964) BT_AV: protocol service capabilities configured: 0x1 
I (15974) BT_AV: Peer device support delay reporting
I (16034) RC_TG: AVRC register event notification: 13, param: 0x0
I (16234) RC_CT: AVRC conn_state event: state 1, [a4:c6:f0:85:a9:9a]
I (16234) RC_CT: AVRC remote features 25b, TG features d1
I (16234) RC_TG: AVRC remote features: 25b, CT features: 2
W (16304) BT_APPL: new conn_srvc id:19, app_id:0
I (16304) BT_AV: A2DP connection state: Connected, [a4:c6:f0:85:a9:9a]
I (16304) BT_APP_CORE: ringbuffer data empty! mode changed: RINGBUFFER_MODE_PREFETCHING
I (16364) RC_CT: remote rn_cap: count 7, bitmask 0x1f06
I (16474) RC_CT: AVRC metadata rsp: attribute id 0x1, 
I (16474) RC_CT: AVRC metadata rsp: attribute id 0x2, 
I (16474) RC_CT: AVRC metadata rsp: attribute id 0x4, 
[... 836 lines cut ...]
W (106224) BT_APPL: Sequence numbers error, recv:0x590, expect:0x58f, recount:0x0
W (106234) BT_APPL: Sequence numbers error, recv:0x592, expect:0x591, recount:0x0
I (106244) BT_APP_CORE: ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING
W (106294) BT_APP_CORE: ringbuffer overflowed, ready to decrease data! mode changed: RINGBUFFER_MODE_DROPPING
W (106294) BT_APP_CORE: ringbuffer is full, drop this packet!
W (106304) BT_APP_CORE: ringbuffer is full, drop this packet!
I (106324) RC_TG: AVRC register event notification: 13, param: 0x0
W (106334) BT_APP_CORE: ringbuffer is full, drop this packet!
I (106334) BT_APP_CORE: ringbuffer data decreased! mode changed: RINGBUFFER_MODE_PROCESSING
I (107884) BT_AV: Audio packet count: 1300
I (110194) BT_AV: Audio packet count: 1400
I (112504) BT_AV: Audio packet count: 1500
I (112874) BT_AV: A2DP audio state: Suspended
I (112994) BT_APP_CORE: ringbuffer underflowed! mode changed: RINGBUFFER_MODE_PREFETCHING
I (115824) RC_TG: Volume is set locally to: 29%
I (115924) RC_TG: AVRC register event notification: 13, param: 0x0
W (116074) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(30 54)
W (116114) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 54, status 0x0
I (116114) BT_AV: ESP_BT_GAP_MODE_CHG_EVT mode: 2
I (125824) RC_TG: Volume is set locally to: 33%
I (125894) RC_TG: AVRC register event notification: 13, param: 0x0
I (135824) RC_TG: Volume is set locally to: 37%
I (135924) RC_TG: AVRC register event notification: 13, param: 0x0

```

0x0fe, 2023-04-17:
i think the problem may be related to MCLK, because when i tested at first i did not enabled the MCLK (just for test) and i did not get any underflow error in the log, of course i could not hear anything because my DAC needs MCLK but i suspect it would have been ok on a DAC without MCLK. 
However, as soon as i enabled MCLK, i got sound but i also got these underflows all over.

0x0fe, 2023-04-17:
more log, after i enabled in PSRAM, and corrected the I2S mode to philips, overall no difference, volume control still inoperant and still as many underruns.

```
I (434) esp_image: segment 4: paddr=000f904c vaddr=4008025c size=1d7bch (120764) load
I (499) boot: Loaded app from partition at offset 0x10000
I (499) boot: Disabling RNG early entropy source...
I (509) quad_psram: This chip is ESP32-D0WDR2-V3
I (511) esp_psram: Found 2MB PSRAM device
I (511) esp_psram: Speed: 40MHz
I (511) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
I (517) cpu_start: Pro cpu up.
I (520) cpu_start: Starting app cpu, entry point is 0x400816e8
I (0) cpu_start: App cpu up.
I (989) esp_psram: SPI SRAM memory test OK
I (997) cpu_start: Pro cpu start user code
I (997) cpu_start: cpu freq: 160000000 Hz
I (997) cpu_start: Application information:
I (997) cpu_start: Project name:     a2dp_sink
I (1001) cpu_start: App version:      v5.0.1-dirty
I (1006) cpu_start: Compile time:     Apr 17 2023 23:50:00
I (1011) cpu_start: ELF file SHA256:  06828950afdb6f1d...
I (1016) cpu_start: ESP-IDF:          v5.0.1-dirty
I (1021) cpu_start: Min chip rev:     v0.0
I (1024) cpu_start: Max chip rev:     v3.99 
I (1028) cpu_start: Chip rev:         v3.0
I (1033) heap_init: Initializing. RAM available for dynamic allocation:
I (1039) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1044) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1049) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1054) heap_init: At 3FFCB8F8 len 00014708 (81 KiB): DRAM
I (1060) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1065) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1071) heap_init: At 4009DA18 len 000025E8 (9 KiB): IRAM
I (1076) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (1084) spi_flash: detected chip: generic
I (1086) spi_flash: flash io: dio
W (1089) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (1103) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1110) esp_psram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1180) BTDM_INIT: BT controller compile version [60aae55]
I (1180) system_api: Base MAC address is not set
I (1180) system_api: read default base MAC address from EFUSE
I (1180) phy_init: phy_version 4670,719f9f6,Feb 18 2021,17:07:07
W (1900) BT_BTC: A2DP Enable with AVRC
I (1910) BT_AV: A2DP PROF STATE: Init Complete
I (1920) BT_AV: event: 10
I (1920) BT_AV: event: 10
I (1920) BT_AV: event: 10
I (1920) BT_AV: event: 10
I (1930) BT_AV: Get delay report value: delay_value: 1200 * 1/10 ms
I (1930) BT_AV: Set delay report value: success, delay_value: 1250 * 1/10 ms
W (15570) BT_HCI: hcif conn complete: hdl 0x80, st 0x0
I (15580) BT_AV: ESP_BT_GAP_ACL_CONN_CMPL_STAT_EVT Connected to [a4:c6:f0:85:a9:9a], status: 0x100
W (15670) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
W (15920) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 32000
W (16010) BT_HCI: hcif link supv_to changed: hdl 0x80, supv_to 8000
E (16020) BT_APPL: bta_av_rc_create ACP handle exist for shdl:0
I (16030) BT_AV: A2DP connection state: Connecting, [a4:c6:f0:85:a9:9a]
I (16060) RC_TG: AVRC conn_state evt: state 1, [a4:c6:f0:85:a9:9a]
I (16060) RC_TG: start volume change simulation
I (16230) BT_AV: A2DP audio stream configuration, codec type: 0
I (16240) BT_AV: Configure audio player: 21-15-2-35
I (16240) BT_AV: Audio player configured, sample rate: 44100
I (16240) BT_AV: protocol service capabilities configured: 0x1 
I (16240) BT_AV: Peer device support delay reporting
I (16300) RC_TG: AVRC register event notification: 13, param: 0x0
I (16500) RC_CT: AVRC conn_state event: state 1, [a4:c6:f0:85:a9:9a]
I (16500) RC_CT: AVRC remote features 25b, TG features d1
I (16500) RC_TG: AVRC remote features: 25b, CT features: 2
[... 475 lines cut ...]
W (57500) BT_APPL: Sequence numbers error, recv:0x59e, expect:0x59c, recount:0x0
I (57550) BT_APP_CORE: ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING
W (57580) BT_APP_CORE: ringbuffer overflowed, ready to decrease data! mode changed: RINGBUFFER_MODE_DROPPING
I (57770) BT_APP_CORE: ringbuffer underflowed! mode changed: RINGBUFFER_MODE_PREFETCHING
W (57790) BT_APPL: Sequence numbers error, recv:0x5ab, expect:0x5a9, recount:0x0
I (57840) BT_APP_CORE: ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING
I (58060) BT_APP_CORE: ringbuffer underflowed! mode changed: RINGBUFFER_MODE_PREFETCHING
W (58080) BT_APPL: Sequence numbers error, recv:0x5b7, expect:0x5b5, recount:0x0
I (58130) BT_APP_CORE: ringbuffer data increased! mode changed: RINGBUFFER_MODE_PROCESSING
W (58160) BT_APP_CORE: ringbuffer overflowed, ready to decrease data! mode changed: RINGBUFFER_MODE_DROPPING
W (58230) BT_APP_CORE: ringbuffer is full, drop this packet!
I (58230) BT_APP_CORE: ringbuffer data decreased! mode changed: RINGBUFFER_MODE_PROCESSING
W (58270) BT_APP_CORE: ringbuffer overflowed, ready to decrease data! mode changed: RINGBUFFER_MODE_DROPPING
W (58330) BT_APP_CORE: ringbuffer is full, drop this packet!
I (58330) BT_APP_CORE: ringbuffer data decreased! mode changed: RINGBUFFER_MODE_PROCESSING
I (59360) BT_AV: Audio packet count: 1300
I (60100) BT_AV: A2DP audio state: Suspended
I (60220) BT_APP_CORE: ringbuffer underflowed! mode changed: RINGBUFFER_MODE_PREFETCHING
W (63300) BT_HCI: hci cmd send: sniff: hdl 0x80, intv(30 54)
W (63310) BT_HCI: hcif mode change: hdl 0x80, mode 2, intv 54, status 0x0
I (63320) BT_AV: ESP_BT_GAP_MODE_CHG_EVT mode: 2
I (66060) RC_TG: Volume is set locally to: 21%
I (66130) RC_TG: AVRC register event notification: 13, param: 0x0

```

0x0fe, 2023-04-17:
So, i think i found a big clue, if i change the GPIOs assigned to I2S to those used by default (22,25,26), the problem instantly disappears. No more underruns. Of course i cannot check the audio, because i use our actual hardware and thee is not easy or convenient way to strap/route these GPIos to the DAC, but anyway i am quite sure there would be no stuttering.
So now the questions are : 

-why and how would the I2S GPIos have an impact on the buffer underruns / BT performance / PER.
-how comes this is not the case on D0WD and D0WDQ6?
-which mitigation?

[image]

```
ets Jul 29 2019 12:21:46
rst:0x1 (POWERON_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 153911750, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6056
load:0x40078000,len:15500
ho 0 tail 12 room 4
load:0x40080400,len:3844
entry 0x4008064c
I (30) boot: ESP-IDF v5.0.1-dirty 2nd stage bootloader
I (30) boot: compile time 23:50:27
I (30) boot: chip revision: v3.0
I (30) boot.esp32: SPI Speed      : 40MHz
I (34) boot.esp32: SPI Mode       : DIO
I (38) boot.esp32: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (48) boot: ## Label            Usage          Type ST Offset   Length
I (55) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (61) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (68) boot:  2 factory          factory app      00 00 00010000 00177000
I (74) boot: End of partition table
I (78) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=2b718h (177944) map
I (149) esp_image: segment 1: paddr=0003b740 vaddr=3ffbdb60 size=04674h ( 18036) load
I (157) esp_image: segment 2: paddr=0003fdbc vaddr=40080000 size=0025ch (   604) load
I (157) esp_image: segment 3: paddr=00040020 vaddr=400d0020 size=b9024h (757796) map
I (434) esp_image: segment 4: paddr=000f904c vaddr=4008025c size=1d7bch (120764) load
I (499) boot: Loaded app from partition at offset 0x10000
I (499) boot: Disabling RNG early entropy source...
I (509) quad_psram: This chip is ESP32-D0WDR2-V3
I (511) esp_psram: Found 2MB PSRAM device
I (511) esp_psram: Speed: 40MHz
I (511) esp_psram: PSRAM initialized, cache is in low/high (2-core) mode.
I (517) cpu_start: Pro cpu up.
I (520) cpu_start: Starting app cpu, entry point is 0x400816e8
I (0) cpu_start: App cpu up.
I (989) esp_psram: SPI SRAM memory test OK
I (997) cpu_start: Pro cpu start user code
I (997) cpu_start: cpu freq: 160000000 Hz
I (997) cpu_start: Application information:
I (997) cpu_start: Project name:     a2dp_sink
I (1001) cpu_start: App version:      v5.0.1-dirty
I (1006) cpu_start: Compile time:     Apr 17 2023 23:50:00
I (1011) cpu_start: ELF file SHA256:  ffa45bac1c783651...
I (1016) cpu_start: ESP-IDF:          v5.0.1-dirty
I (1021) cpu_start: Min chip rev:     v0.0
I (1024) cpu_start: Max chip rev:     v3.99 
I (1028) cpu_start: Chip rev:         v3.0
I (1033) heap_init: Initializing. RAM available for dynamic allocation:
I (1039) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (1044) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (1049) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (1054) heap_init: At 3FFCB8F8 len 00014708 (81 KiB): DRAM
I (1059) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1065) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1071) heap_init: At 4009DA18 len 000025E8 (9 KiB): IRAM
I (1076) esp_psram: Adding pool of 2048K of PSRAM memory to heap allocator
I (1084) spi_flash: detected chip: generic
I (1086) spi_flash: flash io: dio
[... 70 lines cut ...]
I (32130) BT_AV: Audio packet count: 700
I (34400) RC_TG: Volume is set locally to: 92%
I (34420) RC_TG: AVRC register event notification: 13, param: 0x0
I (34420) RC_TG: AVRC set absolute volume: 92%
I (34420) RC_TG: Volume is set by remote controller to: 92%
I (34470) BT_AV: Audio packet count: 800
I (36770) BT_AV: Audio packet count: 900
I (39110) BT_AV: Audio packet count: 1000
I (41440) BT_AV: Audio packet count: 1100
I (43740) BT_AV: Audio packet count: 1200
I (44400) RC_TG: Volume is set locally to: 96%
I (44420) RC_TG: AVRC register event notification: 13, param: 0x0
I (46100) BT_AV: Audio packet count: 1300
I (48420) BT_AV: Audio packet count: 1400
I (50740) BT_AV: Audio packet count: 1500
I (53050) BT_AV: Audio packet count: 1600
I (54400) RC_TG: Volume is set locally to: 100%
I (54490) RC_TG: AVRC register event notification: 13, param: 0x0
I (55400) BT_AV: Audio packet count: 1700
I (57720) BT_AV: Audio packet count: 1800
I (60020) BT_AV: Audio packet count: 1900
I (62340) BT_AV: Audio packet count: 2000
I (64400) RC_TG: Volume is set locally to: 3%
I (64450) RC_TG: AVRC register event notification: 13, param: 0x0
```

0x0fe, 2023-04-17:
after more test, i can be even more specific : this problem is acuse by using GPIO21 for DATA, as son as i change DATA gpio to somethign else, for example 19, the problem of underrun disappear.

0x0fe, 2023-04-17:
so, it is a little more complicated apparently, not only GPIO21 cause the issue, in fact many combination of GPIOs cause these underruns, so far only one is glitch free

[image]

0x0fe, 2023-04-18:
so, to be more precise about the proble with audio control, 
1) it does not change the audio level at the DAC
2) there is volume notification, sent from within A2DP even though we dont do anything, at regular interval of few second, and the volume is increased by 4 each time (but it doesnt afect the actual audio level since it doesnt work).
3) there is an horrible clipping going on, never got this clipping before and it is not coming from the PA since we use class A. This si really a digital clipping.
In the log below you can see the olume changed which are sent automatically by A2DP. This test has been done from a PC, earlier i tested from iphone, in both case the random volumes changes are occuring (and reported on the device). Same for the clipping.

I (1008730) BT_AV: Audio packet count: 40600
I (1009090) RC_TG: Volume is set locally to: 96%
I (1009110) RC_TG: AVRC register event notification: 13, param: 0x0
I (1010190) BT_AV: Audio packet count: 40700
I (1011140) RC_TG: AVRC set absolute volume: 99%
I (1011140) RC_TG: Volume is set by remote controller to: 99%
I (1011640) BT_AV: Audio packet count: 40800
I (1011950) RC_TG: AVRC set absolute volume: 74%
I (1011950) RC_TG: Volume is set by remote controller to: 74%
I (1012920) RC_TG: AVRC set absolute volume: 86%
I (1012920) RC_TG: Volume is set by remote controller to: 86%
I (1013090) BT_AV: Audio packet count: 40900
I (1014540) BT_AV: Audio packet count: 41000
I (1015990) BT_AV: Audio packet count: 41100
I (1017450) BT_AV: Audio packet count: 41200
I (1018890) BT_AV: Audio packet count: 41300
I (1019090) RC_TG: Volume is set locally to: 90%
I (1019100) RC_TG: AVRC register event notification: 13, param: 0x0
I (1020340) BT_AV: Audio packet count: 41400
I (1021800) BT_AV: Audio packet count: 41500
I (1023250) BT_AV: Audio packet count: 41600
I (1024700) BT_AV: Audio packet count: 41700
I (1026150) BT_AV: Audio packet count: 41800
I (1027610) BT_AV: Audio packet count: 41900
I (1029050) BT_AV: Audio packet count: 42000
I (1029090) RC_TG: Volume is set locally to: 94%
I (1029100) RC_TG: AVRC register event notification: 13, param: 0x0
I (1030500) BT_AV: Audio packet count: 42100
I (1031950) BT_AV: Audio packet count: 42200
I (1033410) BT_AV: Audio packet count: 42300
I (1034860) BT_AV: Audio packet count: 42400
I (1036310) BT_AV: Audio packet count: 42500
I (1037760) BT_AV: Audio packet count: 42600
I (1039090) RC_TG: Volume is set locally to: 98%
I (1039130) RC_TG: AVRC register event notification: 13, param: 0x0
I (1039210) BT_AV: Audio packet count: 42700
I (1040660) BT_AV: Audio packet count: 42800
I (1042110) BT_AV: Audio packet count: 42900
I (1043560) BT_AV: Audio packet count: 43000
I (1045030) BT_AV: Audio packet count: 43100
I (1046480) BT_AV: Audio packet count: 43200
I (1047920) BT_AV: Audio packet count: 43300
I (1049090) RC_TG: Volume is set locally to: 1%
I (1049110) RC_TG: AVRC register event notification: 13, param: 0x0
I (1049370) BT_AV: Audio packet count: 43400
I (1050820) BT_AV: Audio packet count: 43500
I (1052270) BT_AV: Audio packet count: 43600
I (1053720) BT_AV: Audio packet count: 43700
I (1055170) BT_AV: Audio packet count: 43800
I (1056630) BT_AV: Audio packet count: 43900
I (1058080) BT_AV: Audio packet count: 44000
I (1059090) RC_TG: Volume is set locally to: 5%
I (1059110) RC_TG: AVRC register event notification: 13, param: 0x0
I (1059530) BT_AV: Audio packet count: 44100
I (1060980) BT_AV: Audio packet count: 44200
I (1062430) BT_AV: Audio packet count: 44300
I (1063880) BT_AV: Audio packet count: 44400
I (1065330) BT_AV: Audio packet count: 44500
I (1066780) BT_AV: Audio packet count: 44600
I (1068240) BT_AV: Audio packet count: 44700
I (1069090) RC_TG: Volume is set locally to: 9%
I (1069110) RC_TG: AVRC register event notification: 13, param: 0x0
I (1069690) BT_AV: Audio packet count: 44800
I (1071140) BT_AV: Audio packet count: 44900
[... 209 lines cut ...]
I (1288820) BT_AV: Audio packet count: 59900
I (1289090) RC_TG: Volume is set locally to: 80%
I (1289120) RC_TG: AVRC register event notification: 13, param: 0x0
I (1290270) BT_AV: Audio packet count: 60000
I (1291720) BT_AV: Audio packet count: 60100
I (1293170) BT_AV: Audio packet count: 60200
I (1294620) BT_AV: Audio packet count: 60300
I (1296070) BT_AV: Audio packet count: 60400
I (1297520) BT_AV: Audio packet count: 60500
I (1298970) BT_AV: Audio packet count: 60600
I (1299090) RC_TG: Volume is set locally to: 84%
I (1299110) RC_TG: AVRC register event notification: 13, param: 0x0
I (1300430) BT_AV: Audio packet count: 60700
I (1301890) BT_AV: Audio packet count: 60800
I (1303330) BT_AV: Audio packet count: 60900
I (1304780) BT_AV: Audio packet count: 61000
I (1306230) BT_AV: Audio packet count: 61100
I (1307680) BT_AV: Audio packet count: 61200
I (1309090) RC_TG: Volume is set locally to: 88%
I (1309110) RC_TG: AVRC register event notification: 13, param: 0x0
I (1309130) BT_AV: Audio packet count: 61300
I (1310580) BT_AV: Audio packet count: 61400
I (1312040) BT_AV: Audio packet count: 61500
I (1313490) BT_AV: Audio packet count: 61600
```

0x0fe, 2023-04-18:
i juste ran the ad2p library test (so, with IDF 4.4) , 

1) there is no clipping and no stuttering, everything is smooth when using GPIO19 for DOUT. 

2)  level is lower than IDF 5.0.1 A2DP sink example (using identical source) so it seems IDF 5.0.1 is indeed overdriving the I2S datas somehow, to the point of feeding clipped datas to the DAC.

3) the volume control works fine on the A2DP library running over IDF 4.4 so it seems this problem also relates to IDF 5.0.1.

So, it confirm the root cause of the problem is related to pin muxing or "cross bar" or maybe some other issues tightly related to the pin mapping. As far as we are concerned we will jsut use GPIO19 instead of GPIO21 to go around the issue since we already wasted a lot of time on this, but it is certainly worth investigating what causes this issue, because again it oes not occur on D0WD nor D0WDQ6, and it seriously affects the usability of I2S whenever A2DP will be used, because it seems man pin conbinations will not work at all.

xiongweichao, 2023-08-15:
@0x0fe  May I ask if this problem still exists in the latest idf v5.0?