espressif/esp-idf #11173: [v5.1] [Heap Trace Standalone] hash map uses ~10x more ram than necessary

chipweinberger, 2023-04-10:
### Is your feature request related to a problem?

@SoucheSouche 

Again, thanks for your work on [this PR.](https://github.com/espressif/esp-idf/pull/10793)

It looks like the hash map entries are storing a *an entire copy* of the record needlessly.

 i.e., all the backtraces, etc, which is like 100+ bytes per hash map entry.

```
/* Define struct: linked list of records */
TAILQ_HEAD(heap_trace_record_list_struct_t, heap_trace_record_t);
typedef struct heap_trace_record_list_struct_t heap_trace_record_list_t;
```

In my PR it was just 8 bytes per hash map entry:

```
typedef struct heap_trace_hashmap_entry_t {
    void* address;                 ///< ptr returned by malloc/calloc/realloc
    heap_trace_record_t* record;   ///< associated record
} heap_trace_hashmap_entry_t;
```

You should be defining it like this:

TAILQ_HEAD(heap_trace_record_list_struct_t, heap_trace_record_t*);

chipweinberger, 2023-04-10:
Actually, I think my TAILQ knowledge is mistaken. Looks like the pointer is implied. 

```
#define TAILQ_HEAD(name, type)						\
struct name {								\
	struct type *tqh_first;	/* first element */			\
	struct type **tqh_last;	/* addr of last next element */		\
}
```

chipweinberger, 2023-04-11:
I edited this issue. Hopefully your internal issue tracker supports edits.

chipweinberger, 2023-04-11:
Note, the records themselves are doubly linked because:
- during freeing, we traverse backwards
- during printing, we traverse forwards (chronological)

But the hash map does not need this feature.

SoucheSouche, 2023-04-12:
Hi @chipweinberger, thanks for reporting this. It is true that we could use a singly linked list in the hashmap. I update the feature and get back to you with perf results.

SoucheSouche, 2023-04-12:
So it seems that the `SLIST` has a complexity of O(n) when removing elements from the list. Meanwhile, `TAILQ` has a complexity of O(1) while removing. To get the same perf with the `SLIST` has we have with the `TAILQ`, we have to increase the size of the hashmap in a way that results in taking more memory space than with `TAILQ`.

chipweinberger, 2023-04-12:
STAILQ is more appropriate than SLIST

it lets you remove from the end in O(1)

SoucheSouche, 2023-04-12:
`SLIST_REMOVE` and `STAILQ_REMOVE` are traversing the entire list. Only `TAILQ_REMOVE` allows for O91) removal from the list. So I will keep using the doubly linked list.

chipweinberger, 2023-04-12:
Ya, thats a good point. Looking at the code, you are right.

Closing this.

chipweinberger, 2023-05-26:
Just to circle back, if we wanted to use SLIST without hurting perf, we would need to replace `map_find` (which does a linear traversal),  with `map_find_and_remove`, that way we can find the record and remove it at the same time.

This would require a small code rewrite.

SoucheSouche, 2023-10-04:
Hi @chipweinberger, I am finally back from me extended leave! I will take a look at that idea as soon as I can :) I'll keep you posted!

chipweinberger, 2023-10-04:
Wow nice to hear from you! I was wondering what happened to you =P

Hope you enjoyed your time off!

SoucheSouche, 2023-10-04:
yes I did very much! I was on paternity leave :)

SoucheSouche, 2023-10-12:
@chipweinberger , I had a look at the code. I am not sure I follow you here. Are you suggesting to make only hash_map a singly linked list?

chipweinberger, 2023-10-12:
correct

```
Note, the records themselves are doubly linked because:
- during freeing, we traverse backwards
- during printing, we traverse forwards (chronological)

But the hash map does not need this feature.

Also, because the hash map should be large as possible to avoid collisions, we care more that each entry is as small as possible.
```

SoucheSouche, 2023-10-12:
I will update hash_map to a singly linked list.

SoucheSouche, 2023-10-23:
The update of the hash map was merged internally.

chipweinberger, 2023-10-23:
awesome! glad it worked out!

SoucheSouche, 2023-10-25:
the update will be available on next github sync. I will close the issue now.

closed 2023-10-25 by SoucheSouche