espressif/esp-idf #11175: monitor dont work in wsl even with all the requirements

Mathe13, 2023-04-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v3.4-77-g35375c17

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP8266_RTOS_SDK

### Power Supply used.

USB

### What is the expected behavior?

the comand make monitor to run and show the output of the code. i am using the hello_world example of ESP8266_RTOS_SDK

### What is the actual behavior?

make monitor run and trows lots of error on esp-idf python files

### Steps to reproduce.

follow the instruction on the ESP8266_RTOS_SDK tutorial using windows wsl
make monitor

### Debug Logs.

```plain
Python requirements from /home/math/esp/ESP8266_RTOS_SDK/requirements.txt are satisfied.
MONITOR
/home/math/esp/ESP8266_RTOS_SDK/tools/idf_monitor.py:53: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
  from distutils.version import StrictVersion
--- idf_monitor on /dev/ttyS5 74880 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
Exception in thread Thread-2 (_run_outer):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/math/esp/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 136, in _run_outer
    self.run()
  File "/home/math/esp/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 225, in run
    self.serial.open()
  File "/home/math/.local/lib/python3.10/site-packages/serial/serialposix.py", line 332, in open
    self._reconfigure_port(force_update=True)
  File "/home/math/.local/lib/python3.10/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (5, 'Input/output error')
```

### More Information.

i am using wsl type 1 and have acess to the com ports. i can write the code but the monitor dont work

Mathe13, 2023-04-11:
i change the baud rate of the monitor to 115200 and the output changed to 

`
sd$ܟ|l<lc|r#cgndogc8ćlrd;l8glc'|scogld`g$'{g$l xo{#g|bog$`gln;gl`xo;#g|Ǐ#og$`nd`gsۓn$`;g docǏl'oldb|oo;ldp䇄gn'ob{n$gllocǏl'oo'{ndgllncǏldn'odg'|'g'n$lllnd$g#{lbogll$l~do$l'crdcgnldlgd$ll'ldgc;cngogdl`do$lncxcgnco''gdo$l'b$g'do$|$|l|glnd$g#lbogg|o$gldoc|lc'odgodd'do$l'bpdlc'og|g|dd'do$l'bplc'ologcnnlnd$g#xldcd|'b'|cd$o?;b8|ۄl$grglndlg#8$cd|'b'|lscsÃo~ll#lgrglodlg#8$cd|nbn|#sܜo~dl#'o{n$gldocplbl|ocg|cpğc$ddcog;glnd$g#$d|'b'|clo?bǏlgnsn$gllncloglgۃlĜ|'co'$'|$gl`ۃno'n|x n$g|ldǏo'$o|#g'oo<#og$#ll`d#oxno$scno$gnl;no||bc$
Traceback (most recent call last):
  File "/home/math/esp/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 846, in <module>
    main()
  File "/home/math/esp/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 758, in main
    monitor.main_loop()
  File "/home/math/esp/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 375, in main_loop
    self.handle_serial_input(data)
  File "/home/math/esp/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 446, in handle_serial_input
    self._print(self._last_line_part)
  File "/home/math/esp/ESP8266_RTOS_SDK/tools/idf_monitor.py", line 668, in _print
    console_printer(string)
  File "/home/math/.local/lib/python3.10/site-packages/serial/tools/miniterm.py", line 66, in write_bytes
    self.byte_output.flush()
OSError: [Errno 5] Input/output error
Exception ignored in: <_io.TextIOWrapper name='<stdout>' mode='w' encoding='utf-8'>
OSError: [Errno 5] Input/output error
make: *** [/home/math/esp/ESP8266_RTOS_SDK/components/esptool_py/Makefile.projbuild:113: monitor] Error 120
`

wujiangang, 2023-04-11:
Could you check /dev/ttyS5 is accessible? By using minicom or miniterm.py?