espressif/esp-idf #11176: How do I get a unique number in ESP32?

gomez9656, 2023-04-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, there Espressif team!

I'm working with the ESP32 and I would like to get a unique number on each board that I'm using.

I'm using ESP.getEfuseMac() function but I already found two boards with the same number, and I have been reading several posts but I can't get a clear answer.

Is it possible to get a unique number for each ESP32 that I'm using?

Thanks a lot in advance

ESP-Marius, 2023-04-12:
The efuse mac should indeed be unique. 

If you somehow have two boards with the same efuse mac I would be interested to see this address. Could you email it to me at marius.vikhammer at espressif.com?

idea--list, 2023-04-12:
Several people reported this in the past. Probably it does not occur too common, however in case of a larger fleet of devices this can be an issue. The best workaround of this i can think of is storing a truly unique number in NVS or EFUSE.
Yes it means one more manual step during the production and one more little firmware just storing such a unique id before uploading the FW for delivery.

Even if using just NVS that value will not change until the partition table changes or the erase-flash command is called.

Maybe there is a way to feed the FW upload process with a list of unique numbers and after each upload it would mark already used numbers on that list...however i do not know about such.

chipweinberger, 2023-04-12:
ESP32-S3 has a unique 128bit ID burned into the efuses

https://github.com/espressif/esp-idf/issues/8819

dgduncan, 2023-04-19:
@idea--list When we are doing our production firmware flashing, we simply generate a unique key and flash that into memory. I know it may not be ideal for some use cases; however, accepting the assumption that the EFUSE mac may indeed not be unique and moving on has saved lots of headaches. It also allows us to decouple the job of unique identification from anything that Espressif is doing.

ESP-Marius, 2023-04-26:
Marking this one as answered. As others have mentioned, generating a unique ID and storing it yourself in NVS/flash during manufacturing is also an option if you want to be in total control of this process.