espressif/esp-idf #11177: ESP32 RMT clk_div error

filzek, 2023-04-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

Release V4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 wrover-e

### Power Supply used.

External 5V

### What is the expected behavior?

problem with the RMT clk_div.

To set it to 1us the clk_div =1 .

The esp32 running at 240mhz it shall use the default APB 80, but the reference is lost and when set the clk_div = 80 one RMT tick is about 80000us, so, when set the clk_div =1 the 1 tick is equal 1us.

This behavior only happen in the v4.4.4 and not in the past versions.

please clarify.

### What is the actual behavior?

esp32 worver-e with psram and sdk v4.4.4

### Steps to reproduce.

run the sdk with wifi connection, htrtp rest and rtm with 1us to show that clock division must be equal 1

suda-morris, 2023-04-12:
@filzek If the RMT is using the APB clock source, then `clk_div=80` can configure the RMT tick resolution into 80M/80=1MHz, which means 1 tick = 1us. 

See the example code:

```c
static const rmt_item32_t morse_esp[] = {
    {{{ 1000, 1, 1000, 0 }}},
    {{{ 1000, 1, 1000, 0 }}},
    {{{ 1000, 1, 1000, 0 }}},
    {{{ 1000, 1, 1000, 0 }}},
    {{{ 1000, 1, 1000, 0 }}},
    // RMT end marker
    {{{ 0, 1, 0, 0 }}}
};

static void rmt_tx_init(void)
{
    rmt_config_t config = RMT_DEFAULT_CONFIG_TX(0, 0);
    config.clk_div = 80;
    ESP_ERROR_CHECK(rmt_config(&config));
    ESP_ERROR_CHECK(rmt_driver_install(config.channel, 0, 0));
}

void app_main(void)
{
    rmt_tx_init();

    while (1) {
        ESP_ERROR_CHECK(rmt_write_items(RMT_TX_CHANNEL, morse_esp, sizeof(morse_esp) / sizeof(morse_esp[0]), true));
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```

and the waveform:

[image]

filzek, 2023-04-19:
the problem are direct on the RMT engine, we have debug all the working code and there is no memory leak or error on it.

the RMT division are somehow getting broke, so 1 tick is no us.
sometimes it is 1000us, sometimes it is 10000000000000us, complete broke.

we found a way to solve the problem in a temporary BUGFIX way, in the fucntion we call to init the RMT and before the RMT to write the items it must have some minimun amount of printf to RESET the RMT to work in 1us with:

rmt_config_t config;
    config.rmt_mode = RMT_MODE_TX;
    config.channel = RMT_RF433TX_CHANNEL;
    config.gpio_num = port_TX433;
    config.mem_block_num = 1;
    config.clk_div = 1; //RMT_CLOCK_DIVIDER;
    config.tx_config.loop_en = false;
    config.tx_config.carrier_en = false;
    config.tx_config.carrier_freq_hz = 0;
    // config.tx_config.carrier_freq_hz = 0;
    config.tx_config.carrier_level = RMT_CARRIER_LEVEL_LOW;
    config.tx_config.carrier_duty_percent = 0;
    config.tx_config.idle_level = RMT_IDLE_LEVEL_LOW;
    config.tx_config.idle_output_en = true;
    rmt_config(&config);
    rmt_driver_install(config.channel, 0, 0);

the separete code outside of a full working code is always working, the problem happen only when some interruption and other thing are working on, so, it seens that this is the way they REWORK the RMT driver in idf 5.0, maybe, maybe it will never get fixed in 4.4.4.

Alvin1Zhang, 2023-08-07:
Thanks for reporting, sorry that we failed to reproduce on our side, feel free to reopen if the issue could be reproduced.

closed 2023-08-07 by Alvin1Zhang