espressif/esp-idf #11180: idf4.4迁移到5.0以上，在4.4下编译没有报错，5.0就报重定义的错误，是CMake语法的问题吗？

LHYYSY, 2023-04-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V4.4

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

版本迁移

### What is the actual behavior?

报错

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
以下是日志：
FAILED: ESP32_C3_Framework.elf

cmd.exe /C "cd . && D:\Espressif\tools\riscv32-esp-elf\esp-2022r1-11.2.0\riscv32-esp-elf\bin\riscv32-esp-elf-g++.exe -march=rv32imc -nostartfiles -march=rv32imc --specs=nosys.specs @CMakeFiles\ESP32_C3_Framework.elf.rsp -o ESP32_C3_Framework.elf  && cd ."

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(sp_gatt_server.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:160: multiple definition of `pixels_driver_config'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:160: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(sp_gatt_server.c.obj): in function `.LANCHOR23':

C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:158: multiple definition of `dev_config'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:158: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(sp_gatt_server.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: multiple definition of `sp_upper_cmd_handles'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(handle_wrapped_multi_sources_data.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: multiple definition of `sp_upper_cmd_handles'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(sp_udp_server.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:160: multiple definition of `pixels_driver_config'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:160: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(sp_udp_server.c.obj): in function `.LANCHOR2':

C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:158: multiple definition of `dev_config'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:158: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(sp_udp_server.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: multiple definition of `sp_upper_cmd_handles'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(handle_raw_data.c.obj): in function `.LANCHOR3':

C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: multiple definition of `sp_upper_cmd_handles'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(wifiCommon.c.obj): in function `.LANCHOR15':

C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_package_fixed/wifi_provisioning/sp_wifi_provisioning/wifiCommon.c:42: multiple definition of `instance_got_ip'; esp-idf/common_package_fixed/libcommon_package_fixed.a(esp_basic_wifi_init.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_package_fixed/wifi_provisioning/esp_basic_wifi/esp_basic_wifi_init.c:15: first defined here
[... 12 lines cut ...]

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(buttons.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:158: multiple definition of `dev_config'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:158: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(buttons.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: multiple definition of `sp_upper_cmd_handles'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_components/libcommon_components.a(upper_cmd_handle_cb.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:160: multiple definition of `pixels_driver_config'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:160: first defined here

d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_components/libcommon_components.a(upper_cmd_handle_cb.c.obj): in function `.LANCHOR1':

C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:158: multiple definition of `dev_config'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/common_dev_config.h:158: first defined here

[... cut ...]
d:/espressif/tools/riscv32-esp-elf/esp-2022r1-11.2.0/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/11.2.0/../../../../riscv32-esp-elf/bin/ld.exe: esp-idf/common_package_fixed/libcommon_package_fixed.a(tcp_ota.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: multiple definition of `sp_upper_cmd_handles'; esp-idf/main/libmain.a(app_main.c.obj):C:/Users/SPERLL/Desktop/esp32c3_framework/components/common_components/configurable/upper_cmd_handle_cb.h:164: first defined here

collect2.exe: error: ld returned 1 exit status

ninja: build stopped: subcommand failed.
```

ESP-Marius, 2023-04-13:
With GCC 10, `-fno-common` is the default behavior (https://gcc.gnu.org/bugzilla/show_bug.cgi?id=85678). So if you were relying on this behavior (looks like you might be declaring global symbols on your header file?) then this will result in a build failure on newer version of IDF. See e.g. [global-variables-in-header-file](https://stackoverflow.com/questions/8108634/global-variables-in-header-file) if you need help with this.

You should also be able to get back the old behavoir (not recommended, you should rather update your code to follow the c-standard) by setting `target_compile_options(${COMPONENT_LIB} PRIVATE "-fcommon)

ESP-Marius, 2023-04-13:
Also see: https://gcc.gnu.org/gcc-10/porting_to.html, first item about fno-common

LHYYSY, 2023-04-13:
very thankful

ESP-Marius, 2023-04-13:
Marking this one as closed then, feel free to reopen if there was still an unresolved IDF issue present.

closed 2023-04-13 by ESP-Marius