espressif/esp-idf #11181: Option to use `idf.py flash` without (re)building project.

amrsoll, 2023-04-12:
### Is your feature request related to a problem?

I build the project from a different machine than the one I am flashing the device(s) from.

This message shows up:

```
Executing action: flash
Build directory '/home/flash_user/my_project/build' configured for project '/home/build_user/my_project' not '/home/flash_user/my_project'. Run 'idf.py fullclean' to start again. 
```

I could use `esptool` as the output of the build suggest, but all the options will change from one project to another and I would need to create extra logic to parse the files or create the `esptool` command on the fly after each build.

I would prefer to use `idf.py flash` as it figures out what and where to flash the binaries from the current working directory.

### Describe the solution you'd like.

`idf.py flash --skip-build` would ignore the build step and use the files in `work_dir/build/` to figure out what to flash.

`idf.py build flash --skip-build` has conflicting options. I suggest the `build` command takes precedence over `--skip-build`. This allows the `build` step to ignore the `--skip-build` flag.

### Describe alternatives you've considered.

Alternatively my problem could be resolved with:

### Flash over ssh

`idf.py flash` being able to run a remote flash job (say through ssh) using local build files

**Example**: `idf.py flash -p user@address:/dev/ttyUSB1`

### Relative paths

the project configuration would use **relative paths** instead of absolute paths. It would make any project much more portable.

amrsoll, 2023-04-12:
Thank you for your prompt answer.

> It figures out the flashing settings based on the project configuration 

That would actually also work for me. File transfer is no issue, I just don't have the CPU power to build the projects comfortably from the remote device meant for flashing.

Unless it is a CMake behaviour, I still think it is best to use relative paths for the project files; in the interest of portability

Wouldn't it be possible to recreate the esptool settings given the config files?

Example:

```sh
$ generate_esptool_flash_command -p /dev/ttyUSB1 --project_dir . > flash_target.sh
$ sh ./flash_target.sh
[ ... esptool output ... ]
$ idf.py monitor
[ ... monitor output ... ]
```

amrsoll, 2023-04-12:
Should I rename the issue to focus on having a more user-friendly esptool?

dobairoland, 2023-04-12:
I updated the title. This needs to be addressed here, not in esptool.

Thank you for the suggestion!

ESP-Marius, 2023-04-13:
What I usually do is to `cd` into the `build` folder and do `esptool.py write_flash @flash_args`, which is pretty convenient.  

Not sure if that helps in your case or not.

amrsoll, 2023-04-13:
@ESP-Marius 

> `esptool.py write_flash @flash_args`

That is actually suuuper convenient and really helps, I was totally unaware! I wish that was the default behaviour of `write_flash` when you're inside the build folder?

Current behaviour:

``` sh
.../firmware/build $ esptool.py write_flash
esptool write_flash: error: the following arguments are required: <address> <filename>
```

Perhaps this tip could be the basic showcase for [`write_flash` in the documentation](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#write-binary-data-to-flash-write-flash)? Or at least be mentioned?

Thank you again for the help 👍

igrr, 2023-04-13:
> Perhaps this tip could be the basic showcase for [write_flash in the documentation](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#write-binary-data-to-flash-write-flash) so it is more prominent?

For a long time we have tried to keep esptool code and docs as free as possible from IDF-isms, which is the reason why `write_flash @flash_args` is not mentioned in that section.

However, we did mention `@flash_args` in the [description of `merge_bin` command](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#merge-bin), so perhaps we could do the same for `write_flash`.

dobairoland, 2023-04-20:
The issue with running just `esptool.py write_flash @flash_args` is that it is not a one-to-one replacement of `idf.py flash`. There are some project specific setups missing (baudrate, chip, before & after behavior).

Here is a work-in-progress output change printed out in the end of `idf.py build`:
```
Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blink.bin
or from the "/home/dragon/esp/esp-idf/examples/get-started/blink/build" directory
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash @flash_args
```

You can see that the last command is based on the above suggestion to use `flash_args` from the build directory, and it is extended with the missing argument to behave the same as `idf.py flash` would.

closed 2023-04-27 by espressif-bot