espressif/esp-idf #11185: Gitee 库中安装总需要提效gitee帐号？

murarduino, 2023-04-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

几次通过github下载安装idf总是不成功，改用gitee库。

### What is the actual behavior?

安装过程中提示提供gitee帐号

### Steps to reproduce.

能过gitee库安装IDF。但每次都要提示提供帐号，帐号提供后又提示404文件不存在。

### Build or installation Logs.

```plain
mura@mura-virtual-machine:~/esp$ git clone --recursive https://gitee.com/EspressifSystems/esp-idf.git
Cloning into 'esp-idf'...
remote: Enumerating objects: 446099, done.
remote: Counting objects: 100% (155383/155383), done.
remote: Compressing objects: 100% (55180/55180), done.
remote: Total 446099 (delta 115351), reused 130111 (delta 93564), pack-reused 290716
Receiving objects: 100% (446099/446099), 206.43 MiB | 5.66 MiB/s, done.
Resolving deltas: 100% (334682/334682), done.
Submodule 'components/bootloader/subproject/components/micro-ecc/micro-ecc' (https://gitee.com/kmackay/micro-ecc.git) registered for path 'components/bootloader/subproject/components/micro-ecc/micro-ecc'
Submodule 'components/bt/controller/lib_esp32' (https://gitee.com/espressif/esp32-bt-lib.git) registered for path 'components/bt/controller/lib_esp32'
Submodule 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib' (https://gitee.com/espressif/esp32c2-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c2/esp32c2-bt-lib'
Submodule 'components/bt/controller/lib_esp32c3_family' (https://gitee.com/espressif/esp32c3-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c3_family'
Submodule 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib' (https://gitee.com/espressif/esp32c6-bt-lib.git) registered for path 'components/bt/controller/lib_esp32c6/esp32c6-bt-lib'
Submodule 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib' (https://gitee.com/espressif/esp32h2-bt-lib.git) registered for path 'components/bt/controller/lib_esp32h2/esp32h2-bt-lib'
Submodule 'components/bt/host/nimble/nimble' (https://gitee.com/espressif/esp-nimble.git) registered for path 'components/bt/host/nimble/nimble'
Submodule 'components/cmock/CMock' (https://gitee.com/ThrowTheSwitch/CMock.git) registered for path 'components/cmock/CMock'
Submodule 'components/esp_coex/lib' (https://gitee.com/espressif/esp-coex-lib.git) registered for path 'components/esp_coex/lib'
Submodule 'components/esp_phy/lib' (https://gitee.com/espressif/esp-phy-lib.git) registered for path 'components/esp_phy/lib'
Submodule 'components/esp_wifi/lib' (https://gitee.com/espressif/esp32-wifi-lib.git) registered for path 'components/esp_wifi/lib'
Submodule 'components/heap/tlsf' (https://gitee.com/espressif/tlsf.git) registered for path 'components/heap/tlsf'
Submodule 'components/ieee802154/lib' (https://gitee.com/espressif/esp-ieee802154-lib.git) registered for path 'components/ieee802154/lib'
Submodule 'components/json/cJSON' (https://gitee.com/DaveGamble/cJSON.git) registered for path 'components/json/cJSON'
Submodule 'components/lwip/lwip' (https://gitee.com/espressif/esp-lwip.git) registered for path 'components/lwip/lwip'
Submodule 'components/mbedtls/mbedtls' (https://gitee.com/espressif/mbedtls.git) registered for path 'components/mbedtls/mbedtls'
Submodule 'components/mqtt/esp-mqtt' (https://gitee.com/espressif/esp-mqtt.git) registered for path 'components/mqtt/esp-mqtt'
Submodule 'components/openthread/lib' (https://gitee.com/espressif/esp-thread-lib.git) registered for path 'components/openthread/lib'
Submodule 'components/openthread/openthread' (https://gitee.com/espressif/openthread.git) registered for path 'components/openthread/openthread'
Submodule 'components/protobuf-c/protobuf-c' (https://gitee.com/protobuf-c/protobuf-c.git) registered for path 'components/protobuf-c/protobuf-c'
Submodule 'components/spiffs/spiffs' (https://gitee.com/pellepl/spiffs.git) registered for path 'components/spiffs/spiffs'
Submodule 'components/unity/unity' (https://gitee.com/ThrowTheSwitch/Unity.git) registered for path 'components/unity/unity'
Cloning into '/home/mura/esp/esp-idf/components/bootloader/subproject/components/micro-ecc/micro-ecc'...
remote: Enumerating objects: 1086, done.        
remote: Total 1086 (delta 0), reused 0 (delta 0), pack-reused 1086        
Receiving objects: 100% (1086/1086), 647.85 KiB | 3.70 MiB/s, done.
Resolving deltas: 100% (637/637), done.
Cloning into '/home/mura/esp/esp-idf/components/bt/controller/lib_esp32'...
Username for 'https://gitee.com': 
Password for 'https://XXXXXXX@gitee.com': 
remote: [session-264ff5a2] 404 not found!
fatal: repository 'https://gitee.com/espressif/esp32-bt-lib.git/' not found
fatal: clone of 'https://gitee.com/espressif/esp32-bt-lib.git' into submodule path '/home/mura/esp/esp-idf/components/bt/controller/lib_esp32' failed
Failed to clone 'components/bt/controller/lib_esp32'. Retry scheduled
Cloning into '/home/mura/esp/esp-idf/components/bt/controller/lib_esp32c2/esp32c2-bt-lib'...
```

wujiangang, 2023-04-13:
gitee 仓库的话，请使用 https://gitee.com/EspressifSystems/esp-gitee-tools 里的工具拉取。参看 https://gitee.com/EspressifSystems/esp-gitee-tools/blob/master/docs/README-submodule-update.md。

或者直接用 jihu 的镜像，参看 https://gitee.com/EspressifSystems/esp-gitee-tools/blob/master/docs/README-jihu-mirror.md

murarduino, 2023-04-13:
谢谢，依据提示已经安装好IDF。

closed 2023-04-13 by murarduino