espressif/esp-idf #11186: Component has moved in 4.4.4 minor release. (Freemodbus)

madeinaus, 2023-04-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4-374-g8e559452d9  and v4.4.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected the component file path to be unchanged for all 4.4.x sources.

### What is the actual behavior?

The freemodbus component moved to a different location.

opt/esp/idf/components/freemodbus/modbus
(as in prior releases) moved to
opt/esp/idf/components/freemodbus/freemodbus/modbus
in the x.x.4 releases   (v4.4.4 and v4.4.4-374)

in particular
opt/esp/idf/components/freemodbus/modbus/include/mbconfig.h 
moved to
opt/esp/idf/components/freemodbus/freemodbus/modbus/include/mbconfig.h

Since we configure modbus, this broke our build.

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### More Information.

Since we configure modbus, moving the config file broke our build

file location was unchanged in
v4.4.3
v4.4.2
v4.3.3
v4.1.3
v4.1.1

We wouldn't have been surprised if 4.4 was different to 4.3, but seeing unannounced build changes in a bug release was unexpected.

igrr, 2023-04-13:
@madeinaus I'm afraid that we don't consider the location of the files within the component as part of the public API. We often move files around, but make sure that the build system exposes the header files of the component in the same way.

The reason why we have unfortunately broken your build is that you are accessing the internal implementation of the component (mbconfig.h) — and modifying IDF source files is not something we intentionally support.

My suggestion for this situation is as follows:
- In your project you can use IDF_VERSION_MAJOR, IDF_VERSION_MINOR, IDF_VERSION_PATCH CMake variables to detect the IDF version, and then use different paths for v4.4.3 and v4.4.4
- Open a separate feature request to make the things you need from mbconfig.h configurable by other means — at run time or via Kconfig. This will help avoid modifications of component source files in the future.

alisitsyn, 2023-12-11:
The issue is closed.

closed 2023-12-11 by alisitsyn