espressif/esp-idf #11196: SPI SLAVE DEMO recv data error

jiafeng-1, 2023-04-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

您好，我使用esp32 idf4.2版本，spi slaver receiver 例子，接收另一块板子的(SPI master)发送的数据，其中选用了

- 4线spi，
- 采用1MHz频率，
- SPI_POL0_PHA0，
- 板间通过杜邦线连接，
- 还进行了共地处理，
发现只能只接收一个字节，不管是否启用dma都不行。我也使用示波器进行了分析，发现SPI master的数据发送是正常的，但是无论我如何更改都无法正常接收到数据。并且接收的1字节还不对，我只能选用SPI_POL0_PHA1,才能正确匹配1字节。请问可能是什么原因呢？

```

//Called after a transaction is queued and ready for pickup by master. We use this to set the handshake line high.
// 在事务排队并准备好由主节点提取后调用。我们使用它来设置高握手线。
void my_post_setup_cb(spi_slave_transaction_t *trans) {
    WRITE_PERI_REG(GPIO_OUT_W1TS_REG, (1<<GPIO_HANDSHAKE));
}

//Called after transaction is sent/received. We use this to set the handshake line low.
void my_post_trans_cb(spi_slave_transaction_t *trans) {
    WRITE_PERI_REG(GPIO_OUT_W1TC_REG, (1<<GPIO_HANDSHAKE));
}

//Main application
void app_main(void)
{
    int n=0;
    esp_err_t ret;

    //Configuration for the SPI bus
    spi_bus_config_t buscfg={
        // .mosi_io_num=-1,
        .mosi_io_num=GPIO_MOSI,
        // .miso_io_num=-1,
        .miso_io_num=GPIO_MISO,
        .sclk_io_num=GPIO_SCLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
    };

    //Configuration for the SPI slave interface
    spi_slave_interface_config_t slvcfg={
        .mode=1,
        // .mode=0,
        .spics_io_num=GPIO_CS,
        .queue_size=1,
        .flags=0,
        // .post_setup_cb=my_post_setup_cb,
        // .post_trans_cb=my_post_trans_cb
        .post_setup_cb=NULL,
        .post_trans_cb=NULL
    };

    //Configuration for the handshake line
    gpio_config_t io_conf={
        .intr_type=GPIO_INTR_DISABLE,
        .mode=GPIO_MODE_OUTPUT,
        .pin_bit_mask=(1<<GPIO_HANDSHAKE)
    };

    //Configure handshake line as output
    gpio_config(&io_conf);
    //Enable pull-ups on SPI lines so we don't detect rogue pulses when no master is connected.
    // 在SPI线路上启用上拉，以便在没有连接主站时不会检测到恶意脉冲。
    gpio_set_pull_mode(GPIO_MOSI, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_SCLK, GPIO_PULLUP_ONLY);
    gpio_set_pull_mode(GPIO_CS, GPIO_PULLUP_ONLY);

    //Initialize SPI slave interface
    ret=spi_slave_initialize(RCV_HOST, &buscfg, &slvcfg, 0);
    // ret=spi_slave_initialize(RCV_HOST, &buscfg, &slvcfg, SPI_DMA_CH_AUTO);
    assert(ret==ESP_OK);

    WORD_ALIGNED_ATTR char sendbuf[129]="";
    WORD_ALIGNED_ATTR char recvbuf[129]="";
    memset(recvbuf, 0, 129);
    spi_slave_transaction_t t;
    memset(&t, 0, sizeof(t));
   
    while(1) {
        
        printf("Received\n");

        //Clear receive buffer, set send buffer to something sane
        memset(recvbuf, 0, 129);
        // memset(recvbuf, 0xA5, 129);
        sprintf(sendbuf, "This is the receiver, sending data for transmission number %04d.", n);

        //Set up a transaction of 128 bytes to send/receive
        t.length=32*8;
        // t.length=128*8;
        // t.tx_buffer=NULL;
        t.tx_buffer=sendbuf;
        t.rx_buffer=recvbuf;

        // spi_slave_transaction_t *ret_trans;
        // spi_slave_queue_trans(RCV_HOST, &t, portMAX_DELAY);
        // spi_slave_get_trans_result(RCV_HOST, &ret_trans, portMAX_DELAY);
        /* This call enables the SPI slave interface to send/receive to the sendbuf and recvbuf. The transaction is
        initialized by the SPI master, however, so it will not actually happen until the master starts a hardware transaction
        by pulling CS low and pulsing the clock etc. In this specific example, we use the handshake line, pulled up by the
        .post_setup_cb callback that is called as soon as a transaction is ready, to let the master know it is free to transfer
        data.
        */
        ret=spi_slave_transmit(RCV_HOST, &t, portMAX_DELAY);
        // ESP_LOG_BUFFER_HEXDUMP("校验出错", recvbuf, 128, ESP_LOG_INFO);
        // ESP_LOG_BUFFER_HEXDUMP("校验出错", ret_trans->rx_buffer, 128, ESP_LOG_INFO);
        ESP_LOG_BUFFER_HEX("SlaveReceived", recvbuf, t.trans_len/8);
        printf("----------------lens = %d\n", t.trans_len/8);
}
}
```
```
I (22025) SlaveReceived: 5a 
----------------lens = 1
```

但是，当我使用两个esp C3开发板，运行一个slaver和master发现能够正常接收。
```
// master init 
static void spi_init() {
    esp_err_t ret;
    spi_bus_config_t buscfg={
        .miso_io_num = GPIO_MISO,
        .mosi_io_num = GPIO_MOSI,
        .sclk_io_num = GPIO_SCLK,
        .quadwp_io_num = -1,
        .quadhd_io_num = -1,
        .max_transfer_sz = 32,
    };
    ret = spi_bus_initialize(SPI2_HOST, &buscfg, SPI_DMA_CH_AUTO);
    ESP_ERROR_CHECK(ret);
    spi_device_interface_config_t devcfg={
        .clock_speed_hz = 1000000,  // 1 MHz
        .mode = 0,                  //SPI mode 0
        .spics_io_num = GPIO_CS,     
        .queue_size = 1,
        .flags = SPI_DEVICE_HALFDUPLEX,
        .pre_cb = NULL,
        .post_cb = NULL,
    };
    ESP_ERROR_CHECK(spi_bus_add_device(SPI2_HOST, &devcfg, &spi2));
};
```

wanckl, 2023-04-19:
@jiafeng-1 Hello, 我觉得这些example都太长了，又是握手又是回调的，你可以先删减一下再测，然后把示波器的图发出来看看

Alvin1Zhang, 2024-12-18:
Thanks for reporting, will close due to inactivity, feel free to reopen with more updates, thanks.

closed 2024-12-18 by Alvin1Zhang