espressif/esp-idf #11197: Build failure of `examples/bluetooth/esp_hid_device` when configured to compile with performance optimization

finger563, 2023-04-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4557-gf404fe96b1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Build successfully with performance optimization configured via menuconfig

### What is the actual behavior?

[image]

Build failure resulting from `btc_gap_ble.c:89` when performance optimization configured via menuconfig

### Steps to reproduce.

1. Target set to ESP32
2. Set compiler optimization to `Optimize for Performance`
3. Build
4. See error

### Build or installation Logs.

```plain
In file included from /Users/bob/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/gap/btc_gap_ble.c:10:
/Users/bob/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/gap/btc_gap_ble.c: In function 'btc_gap_ble_get_dev_name_callback':
/Users/bob/esp/esp-idf/components/bt/host/bluedroid/stack/include/stack/bt_types.h:35:34: error: 'strncpy' specified bound 33 equals destination size [-Werror=stringop-truncation]
   35 | #define BCM_STRNCPY_S(x1,x2,x3)  strncpy((x1),(x2),(x3))
      |                                  ^~~~~~~~~~~~~~~~~~~~~~~
/Users/bob/esp/esp-idf/components/bt/host/bluedroid/btc/profile/std/gap/btc_gap_ble.c:89:9: note: in expansion of macro 'BCM_STRNCPY_S'
   89 |         BCM_STRNCPY_S(param.get_dev_name_cmpl.name, name, BTC_MAX_LOC_BD_NAME_LEN + 1);
      |         ^~~~~~~~~~~~~
cc1: some warnings being treated as errors
[903/1189] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/bluedroid/btc/profile/std/gatt/btc_gatts.c.objninja: build stopped: subcommand failed.
```

### More Information.

Happens on `esp-idf/main` (which is where the logs above come from) as well as `esp-idf/release/v5.1`, but doesn't seem to happen (in our cloud builds) on `release/v5.0`

finger563, 2023-04-14:
It does not happen when optimization is set to `Optimize for size` - it _only_ happens when set to `Optimize for performance`

(Of course the example cannot link when set to `Debug without optimization`, but the compilation at least doesn't fail)

xiewenxiang, 2023-04-18:
Hi @finger563 ,

Please apply this patch and test again.

[0001-component-bt-Fix-Build-failure-when-configured-to-co.zip](https://github.com/espressif/esp-idf/files/11257478/0001-component-bt-Fix-Build-failure-when-configured-to-co.zip)

finger563, 2023-04-18:
Hi @xiewenxiang yes that's almost the same patch that I'd made locally to get around the issue, thanks. Can we expect this in a 5.1 update or 5.2?

ftab, 2023-07-17:
This issue occurs in release/v5.0 now

esp-cjh, 2023-07-20:
Hi, @ftab 
We have fixed this issue, but the commit hasn't been merged on release/v5.0.

zhp0406, 2023-11-17:
Commit fixed in version 5.1: 62f5905be25e8f096032fb0f62dd768e7f0008de

closed 2023-11-17 by Alvin1Zhang