espressif/esp-idf #11199: Program ESP32 via ESP-IDF & Jetson UART port

ToastTO, 2023-04-15:
### Is your feature request related to a problem?

As many of us using Board like Jetson nano to develop ESP32 project. We need to program custom ESP32 board, which have no USB-UART bridge, via the UART port in Jetson nano. Is the ESP-IDF on Linux support non-USB port for uploading application to ESP32 ??

can we 

### Describe the solution you'd like.

Can we actually do that ?
If yes, what need to set up on the Jetson nano board, and connection to be made between jetson and esp32 chip ?

igrr, 2023-04-19:
I'm afraid we won't provide specific instructions for this particular board, but you can refer to the documentation of the flashing tool to find the required information:

Here is the documentation page which explains the connections required: https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/serial-connection.html.
(Check Jetson docs regarding the TX/RX pin mapping.)

Additionally, if you want to avoid manually placing the board into download mode each time you flash, you need to check this part of the documentation: https://docs.espressif.com/projects/esptool/en/latest/esp32/advanced-topics/boot-mode-selection.html#boot-mode

If you have trouble flashing the board, please check the troubleshooting guide here: https://docs.espressif.com/projects/esptool/en/latest/esp32/troubleshooting.html

radimkarnis, 2023-04-21:
Feel free to reopen if you have any more questions!

closed 2023-04-21 by radimkarnis