espressif/esp-idf #11201: How can I control speaker volume for MAX98357 I2S

waruqi, 2023-04-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a MAX98357 I2S module, I want to control it's volume. 

How do I read and write its gain gpio to control the volume?

waruqi, 2023-04-17:
> Do you mean controlling the gain of `MAX98357`? According to the datasheet, it seems to be controlled by pull-up/pull-down resistors.

What api should I use to control it? pwm? gpio? ledc?

waruqi, 2023-04-17:
chatgpt tell me:

```
#include "driver/gpio.h"
#include "driver/dac.h"

void set_volume(int volume) {
    gpio_set_direction(GPIO_NUM_25, GPIO_MODE_OUTPUT);
    int voltage = volume * 255 / 100;
    dac_output_voltage(DAC_CHANNEL_1, voltage);
}
```

Is it correct?

L-KAYA, 2023-04-18:
The ChatGPT didn't give a correct answer... You can search the datasheet of MAX98357 online or the official website of [Analog Devices](https://www.analog.com/en/index.html)

Here is the datasheet I found:
https://www.analog.com/media/en/technical-documentation/data-sheets/MAX98357A-MAX98357B.pdf

In the section `Gain Selection`, the table shows how to set the gain by pull-up or pull-down by specific resistor:
[image]

That means the gain should NOT controlled by a GPIO theoretically, it's supposed to be fixed according to your PCB design. If you have to control it by a GPIO, you may try to tie a GPIO to the `SD_MODE` pin of `MAX98357` and set the gpio  pulled up/ pulled down or float to select the different gain. But I can't guarantee it will work, please read the `MAX98357` datasheet carefully to see if it support to be controlled by a GPIO.

closed 2023-04-18 by L-KAYA

ekrem, 2023-11-08:
maybe problem solve whit 100k pot.
Because manual says need analog voltage