espressif/esp-idf #11202: WiFi Scan WPA2-EAP Authentication reported as WIFI_AUTH_OPEN

aquaticus, 2023-04-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1-dev-4557-gf404fe96b1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

When using WPA2-EAP authentication method is used and router enables 802.11w Management Frame Protection, WiFi scan reports network as `WIFI_AUTH_WPA2_ENTERPRISE`.

After call to `esp_wifi_scan_get_ap_records()` `wifi_auth_mode_t` member of `wifi_ap_record_t` should be set to  `WIFI_AUTH_WPA2_ENTERPRISE`.

### What is the actual behavior?

Instead WiFi scan authomode for the network is `WIFI_AUTH_OPEN`.

After call to `esp_wifi_scan_get_ap_records()` `wifi_auth_mode_t` member of `wifi_ap_record_t` is incorrectly set to  `WIFI_AUTH_OPEN`.

### Steps to reproduce.

1. On router set authentication method to **WPA2-EAP**
2. On router set **802.11w Management Frame Protection** as Required or Optional
3. Compile and flash WiFi example **scan** (https://github.com/espressif/esp-idf/tree/master/examples/wifi/scan)
4. Check the output of the sample

### Debug Logs.

```plain
I (3297) scan: SSID             test
I (3297) scan: RSSI             -43
I (3297) scan: Authmode         WIFI_AUTH_OPEN
I (3307) scan: Pairwise Cipher  WIFI_CIPHER_TYPE_CCMP
I (3307) scan: Group Cipher     WIFI_CIPHER_TYPE_CCMP
I (3317) scan: Channel          13
```

### More Information.

The issue was recreated on:
* IDF v4.4.4
* IDF v5.0.1
* IDF latest master

Chips: ESP32 and ESP32-C3

**NOTES:**
1. Using WPA2-EAP gives the same results
2. When WPA2-PSK is used Authmode is reported correctly with 802.11w frame protection
3. Other WiFi clients e.g. Linux, Windows, phones report auth mode properly for the same network
4. Test router used: OpenWrt 22.03.3 r20028-43d71ad93e; radio chipset: Qualcomm Atheros QCA9560 802.11bgn

Output from `nmcli dev wifi` on Linux for the same network (proper WPA2 802.1X)
```
test                           Infra  13    195 Mbit/s  100     ▂▄▆█  WPA2 802.1X
```

jgujarathi, 2023-04-18:
Hi @aquaticus ,
We have tried the mentioned steps with an AP configured similarly as yours and cannot seem to reproduce the same result.
Could you please provide a sniffer capture which include the beacons of the AP you are facing the issue with?

I (3420) scan: SSID 		jash_wpa2_ent
I (3420) scan: RSSI 		-25
I (3420) scan: Authmode 	WIFI_AUTH_WPA2_ENTERPRISE
I (3430) scan: Pairwise Cipher 	WIFI_CIPHER_TYPE_CCMP
I (3430) scan: Group Cipher 	WIFI_CIPHER_TYPE_CCMP
I (3440) scan: Channel 		11

aquaticus, 2023-04-19:
@jgujarathi Please find attached beacon pcap dump (unzip file).
[mng_frame_enabled_beacon.zip](https://github.com/espressif/esp-idf/files/11276463/mng_frame_enabled_beacon.zip)

AxelLin, 2023-04-21:
@jgujarathi 
Any finding in https://github.com/espressif/esp-idf/issues/11202#issuecomment-1515179412 ?
Do you need other information?

kapilkedawat, 2023-04-21:
@aquaticus thanks for the capture, this will be corrected in next software fix.

AxelLin, 2023-04-23:
Just want to check if v4.3 branch is impacted by this issue?
I'm wondering if I should wait for this fix for v4.3 branch to build my application firmware.

kapilkedawat, 2023-04-24:
Yes, v4.3 branch is also impacted by this. However its only a display issue and functionality won't change.

AxelLin, 2023-04-24:
> Yes, v4.3 branch is also impacted by this. However its only a display issue and functionality won't change.

Our configure interface does not allow user to set enterprise settings if it was recognized as WIFI_AUTH_OPEN.
So this is not just a display issue, I have to wait for your fix.

AxelLin, 2023-04-29:
@jgujarathi 
Any update? I suppose it's easy to fix this if it's just a display issue without functionality change.

AxelLin, 2023-05-07:
@jgujarathi @kapilkedawat  Any update for the fix?

AxelLin, 2023-05-15:
@jgujarathi @kapilkedawat 

All stable branches are impacted by this issue.
I cannot workaround the issue since I think the issue is in wifi-lib.
Could you fix it ASAP?

AxelLin, 2023-06-09:
> @aquaticus thanks for the capture, this will be corrected in next software fix.

There are several esp32-wifi-lib and esp-idf (including esp-idf-v4.4.5) updates recently, but still cannot find this fix in any branches.

AxelLin, 2023-06-30:
@igrr @Alvin1Zhang  Can someone help to fix this bug? The bug is in all release branches.

closed 2023-07-06 by espressif-bot