espressif/esp-idf #11203: Could `efuse_table_gen.py` be added/ moved to the espressif/esptool repo?

j4m3s, 2023-04-15:
### Is your feature request related to a problem?

The `efuse_table_gen.py` is very helpful - even essential - for reducing errors in use of EFuses.  However, it's currently located here in esp-idf rather than in esptool, meaning that it's not available to those of us using ESP-Arduino (via Platformio anyway).  I currently have to clone the ESP-IDF repo to download the tool and then run it to generate/ update the `.c` and `include/.h` files.

### Describe the solution you'd like.

Copy or move the `efuse_table_gen.py` to the esptool repository.  That way it is available automatically via platformio (via `pio pkg exec -- efuse_table_gen.py`) or by cloning the much smaller, standalone esptool repository.

The `espefuse.py` tool is already located over there so us non-IDFers can already access that one easily :)

### Describe alternatives you've considered.

Cloning the whole ESP-IDF repo to access the python script works too.

KonstantinKondrashov, 2023-04-19:
Hi @j4m3s!
I remember that you also suggested a feature for espefuse.py that can have the ability to burn efuses that are defined in a custom CSV file.
I think that both features can be implemented together and provide a seamless way to use custom efuses. It might be like this:
1. generate *.c & *.h files based on a custom CSV file:
`espefuse.py generate_c_files path/to/custom.csv`
2. burn custom efuses using the espefuse tool:
`espefuse.py --user_efuses custom.csv burn_efuse EFUSE_NAME_FROM_CSV_FILE ... `

j4m3s, 2023-04-19:
Hi @KonstantinKondrashov

Yes that sounds excellent. Having that independent of `idf.py` makes much more sense (to me) :)

KonstantinKondrashov, 2023-10-11:
Closing it for now, because the feature will be implemented in [esptool ](https://github.com/espressif/esptool) repo.

closed 2023-10-11 by KonstantinKondrashov

KonstantinKondrashov, 2023-10-11:
Yes, there is a ticket [ESPTOOL-672] in Jira but it is not visible outside. The link to changes will appear here once it is merged. Thanks.