espressif/esp-idf #11206: ESP32S3/S2 USB FIFO size inconsistency (tinyusb dcd driver)

jrahlf, 2023-04-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The overall USB FIFO size is defined as [1024 in dcd_esp32sx.c](https://github.com/espressif/tinyusb/blob/49628d8c1806fc9bf08e1402c2f8271582c9169a/src/portable/espressif/esp32sx/dcd_esp32sx.c#L50), however it says 4KB in the technical reference manual for both esp32s3 and esp32s2. Note that the register definition section is missing currently in the manual (v1.2).
Later in the code it says in a comment that the [size is either 1280 or 4096](https://github.com/espressif/tinyusb/blob/49628d8c1806fc9bf08e1402c2f8271582c9169a/src/portable/espressif/esp32sx/dcd_esp32sx.c#L101).

1. I have already requested for the missing register section to be added to the manual via the web form.
2. Can you clarify what the actual FIFO size is
3. If the size is 4096, can you adjust the driver accordingly.

This is relevant to achieve high throughputs, for example when sending lots of data as cdc device, see also #11161 / [comment](https://github.com/espressif/esp-idf/issues/11161#issuecomment-1510386748)

tore-espressif, 2023-04-19:
Hello @jrahlf , thank you for the issue. The docs and code comments are not clear, indeed.

**USB _shared_ FIFO RAM for all endpoints is 256 words -> 1024 bytes.** (The technical reference manual mentions 4KB in _memory map_, but only 1KB is implemented. This is misleading, I'll forward this to docs team).

**More details:**
* USB-OTG implementation in both S2 and S3 has 6 endpoints (including endpoint 0), out of which max 5 can be IN endpoints.
* All OUT endpoints share one FIFO. Every IN endpoint (including INOUT endpoint 0) must have their own dedicated FIFO.
* OUT FIFO size is [52 words](https://github.com/espressif/tinyusb/blob/49628d8c1806fc9bf08e1402c2f8271582c9169a/src/portable/espressif/esp32sx/dcd_esp32sx.c#L128) -> 208 bytes
* EP0 FIFO size is [16 words](https://github.com/espressif/tinyusb/blob/49628d8c1806fc9bf08e1402c2f8271582c9169a/src/portable/espressif/esp32sx/dcd_esp32sx.c#L131) -> 64 bytes
* The rest (188 words -> 752bytes) is evenly distributed among the remaining IN endpoints 188/4 = 47 words -> 188 bytes per IN FIFO

In case you don't use all 4 remaining IN endpoints, you don't utilize the FIFO RAM to its maximum (some portion of the FIFO RAM is reserved for endpoints you don't use).

tore-espressif, 2026-09-02:
Hello,

If by 'TRM updated' you mean adding descriptions of the USB registers, this will not happen. This section is protected by an NDA and cannot be made public.

Thank you for your understanding.

If you have any questions regarding NDA-protected documents, please contact us at https://www.espressif.com/en/contact-us/sales-questions

closed 2026-09-02 by tore-espressif