espressif/esp-idf #11207: tinyusb driver pushes 64 bytes to USB IN FIFO instead of the configured size

jrahlf, 2023-04-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

This relates to `dcd_esp32sx.c` [(link)](https://github.com/espressif/tinyusb/blob/master/src/portable/espressif/esp32sx/dcd_esp32sx.c)
In the function `transmit_packet`, it always pushes up to [xfer->max_size, which will be 64.](https://github.com/espressif/tinyusb/blob/49628d8c1806fc9bf08e1402c2f8271582c9169a/src/portable/espressif/esp32sx/dcd_esp32sx.c#L583). 
See inside function `dcd_edpt_open`:  `xfer->max_size = tu_edpt_packet_size(desc_edpt);`

I think it should push up to the configured FIFO size, which is determined in `dcd_edpt_open`.
The current code has 2 problems:
1. USB IN FIFO is underutilized and the full bandwidth of USB cannot be used (see issue #11161)
2. If the FIFO were smaller than 64, this would cause problems. Luckily, this is not possible with the current code.

tore-espressif, 2023-09-11:
Hell @jrahlf , I'm sorry for the late response.
You're right, the current implementation cannot utilize USB FullSpeed bandwidth.
We will add this to our USB roadmap, but at this point we can't commit to any date of delivery.
I will keep this issue updated

tore-espressif, 2025-09-15:
In latest [esp_tinyusb](https://components.espressif.com/components/espressif/esp_tinyusb) USB uses DMA transfers and configurable buffer sizes that can use full USB Full Speed bandwidth