espressif/esp-idf #11208: Support polarity for SDCard Write Protect testing

a2retro, 2023-04-16:
### Is your feature request related to a problem?

https://github.com/espressif/esp-idf/blob/668eb92d4fdedb7d1e32507a77c9bbf9f9ab5e2a/components/driver/spi/sdspi/sdspi_host.c#L136-L143

The code assumes low means a card is write protected. 
Some SDCard are the opposite to that setting for example this card
https://www.digikey.ca/en/products/detail/3m/SD-RSMT-2-MQ-WF/1616663

### Describe the solution you'd like.

Support polarity for SDCard Write Protect testing

igrr, 2023-04-18:
@a2retro Could you please add, which ESP chip are you using? I can suggest an alternative solution which will not require adding a new feature to the SDSPI driver, but this solution is a bit chip-specific.

a2retro, 2023-04-21:
Hi I am using the esp32-S3 chip. Thank You.

closed 2023-06-29 by espressif-bot