espressif/esp-idf #11209: Unable to mount SPIFFS/LittleFS partition when using flash encryption for the app

aaknitt, 2023-04-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v1.6.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C3 XIAO

### Power Supply used.

USB

### What is the expected behavior?

According to the [flash encryption documentation](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/security/flash-encryption.html), data partitions will not be encrypted unless specifically flagged in the partition table.

The following partition table is being used:
```# Name,   Type, SubType,  Offset,   Size, Flags
nvs,      data, nvs,      0x9000,   0x3000,
otadata,  data, ota,      0xe000,   0x2000,
app0,     app,  ota_0,    0x10000,  0x140000, encrypted
app1,     app,  ota_1,    0x150000, 0x140000, encrypted
spiffs,   data, spiffs,   0x290000, 0x100000,
nvs_key,  data, nvs_keys, 0x390000, 0x100,     encrypted
coredump, data, coredump, 0x3F0000, 0x10000,```

The app is encrypted and running succesfully.  Since the data partition being used for spiffs is not encrypted, the app should be able to mount and access data on it.  

### What is the actual behavior?

The app is unable to mount the spiffs partition when using either the SPIFFS or LittleFS libraries.  

When using SPIFFS:
ESP-ROM:esp32c3-api1-20210207
E (12789) SPIFFS: mount failed, -10025
SPIFFS Mount Failed

When using LittleFS:
ESP-ROM:esp32c3-api1-20210207
./components/esp_littlefs/src/littlefs/lfs.c:1229:error: Corrupted dir pair at {0x0, 0x1}
E (301) esp_littlefs: mount failed,  (-84)
E (303) esp_littlefs: Failed to initialize LittleFS
./components/esp_littlefs/src/littlefs/lfs.c:1886:debug: Bad block at 0x0
./components/esp_littlefs/src/littlefs/lfs.c:1892:warn: Superblock 0x0 has become unwritable
E (5186) esp_littlefs: Failed to format filesystem

### Steps to reproduce.

1. Enable flash encryption using partition table above.
2. Flash encrypted SPIFFS example app
3. Monitor serial output

### More Information.

There is a similar issue from several years ago that the reporter was able to self-solve by correcting the partition table:
https://github.com/espressif/esp-idf/issues/2001

I'm unsure if I similarly have an incorrect partition table or if there may be a different issue related to the ESP32-C3, which uses a slightly different flash encryption scheme that the original ESP32 devices.

aaknitt, 2023-04-22:
I was able to get this working using LittleFS after a bunch of trial and error.  

Apparently LittleFS _does_ support encrypted flash. By changing the spiffs partition to encrypted, I can read and write files using LittleFS.  My partition table now looks like this:

```
# Name,   Type, SubType,  Offset,   Size, Flags
nvs,      data, nvs,      0x9000,   0x3000,
otadata,  data, ota,      0xe000,   0x2000,
app0,     app,  ota_0,    0x10000,  0x140000, encrypted
app1,     app,  ota_1,    0x150000, 0x140000, encrypted
spiffs,   data, spiffs,   0x290000, 0x100000, encrypted
nvs_key,  data, nvs_keys, 0x390000, 0x1000,   encrypted
coredump, data, coredump, 0x3F0000, 0x10000,
```
I am using a host generated encryption key, so to package the LittleFS files for flashing I first create the LittleFS binary with something like this:
```
mklittlefs.exe -c C:\project\data -b 4096 -p 256 -s 1048576 littlefs.bin
```
Then encrypt the binary using something like this:
```
espsecure.py encrypt_flash_data --aes_xts --keyfile My_encryption_key.bin --address 0x290000 -o encrypted_littlefs.bin littlefs.bin
```
Then flash the encrypted binary as usual:
```
esptool.py --chip esp32c3 --port COM11 --baud 921600 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size 4MB 0x290000 encrypted_littlefs.bin
```

closed 2023-04-22 by aaknitt

butaikis, 2023-05-05:
@aaknitt Hello. I have a problem. Flash encryption is enabled, I use Littles. I did everything as you wrote, but still a mistake

```
components\esp_littlefs\src\littlefs\lfs.c:1225:error: Corrupted dir pair at {0x0, 0x1}
[   133][E][Little FS.cpp:95] begin(): Mounting LittleFS failed! Error: -1
```

Maybe there is some solution?

aaknitt, 2023-05-05:
> @aaknitt Hello. I have a problem. Flash encryption is enabled, I use Littles. I did everything as you wrote, but still a mistake
>
> ```
> components\esp_littlefs\src\littlefs\lfs.c:1225:error: Corrupted [...]
A few things to clarify:

- What I posted above doesn't include the full set of steps to burn the encryption key and flags into the device, it only includes the steps to get the LittleFS partition encrypted.  You'll still need to burn the encryption flags and key separately.  However, based on the error you're getting, I think you may have that working, as it looks like the application is running and attempting to mount the LittleFS
- The command to encrypt the LittleFS binary that I posted above is specific to the ESP32-C3 (and I think also the ESP32-S2/S3??), which uses a different type of encryption than the other ESP32 models.  If you're not using a C3, you may need to change the --aes_xts flag to something else.

I'm not sure if any of that is helpful or not, good luck!

butaikis, 2023-05-06:
@aaknitt Yes, I have encryption, the key is all on and loaded. It works as it should. 

The problem was with LittLeFS, I already solved it, it turned out espasyncwebserver does not understand encryption when processing storage, so my program crashed.