espressif/esp-idf #11211: md5 function not found

LHYYSY, 2023-04-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

[image]
[image]
[image]

ESP-Marius, 2023-04-17:
In IDF 5.0 we've updated mbtedtls to v3.1.0 which does not have these `_ret` crypto functions, see [description](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/protocols.html#deprecated-functions-were-removed-from-cryptography-modules).

You should use `mbedtls_md5` instead of `mbedtls_md5_ret`

LHYYSY, 2023-04-17:
Many thanks

ESP-Marius, 2023-04-18:
Marking this one as resolved then.

closed 2023-04-18 by ESP-Marius