espressif/esp-idf #11212: Flash_Encryption

AvinashSinghPES, 2023-04-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

firstly, i am using flash encryption in esp32 board. when first time we are flashing the program into the esp32 board then there some mistake in my code  the watchdog timer got triggered. but when trying to re-flash the program, we are successful but when shown a window terminal . we have got these .....

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
invalid header: 0xba8f1c66
invalid header: 0xba8f1c66
invalid header: 0xba8f1c66
invalid header: 0xba8f1c66
invalid header: 0xba8f1c66
invalid header: 0xba8f1c66
invalid header: 0xba8f1c66
ets Jul 29 2019 12:21:46

what should i do? and how to decrypt this esp32 baord?

mahavirj, 2023-04-17:
>  but when trying to re-flash the program, we are successful but when shown a window terminal . we have got these ....

What command you used to re-flash the application? Please share detailed steps, console logs when the flash enc was enabled and the device EFuse summary.

Note: `idf.py encrypted-flash` or `idf.py encrypted-app-flash` can help to flash the encrypted artifacts on the device, provided flash encryption is enabled in the "development" mode.

AvinashSinghPES, 2023-04-17:
we are using ESP-IDF IDE to re-flash the application. 

these are the summery:

C:\Espressif\frameworks\esp-idf-v4.4>espefuse.py --port COM13 summary
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP32
espefuse.py v3.2-dev
EFUSE_NAME (Block) Description  = [Meaningful Value] [Readable/Writeable] (Hex Value)
----------------------------------------------------------------------------------------
Calibration fuses:
BLK3_PART_RESERVE (BLOCK0):                        BLOCK3 partially served for ADC calibration data   = False R/W (0b0)
ADC_VREF (BLOCK0):                                 Voltage reference calibration                      = 1093 R/W (0b10001)

Config fuses:
XPD_SDIO_FORCE (BLOCK0):                           Ignore MTDI pin (GPIO12) for VDD_SDIO on reset     = False R/W (0b0)
XPD_SDIO_REG (BLOCK0):                             If XPD_SDIO_FORCE, enable VDD_SDIO reg on reset    = False R/W (0b0)
XPD_SDIO_TIEH (BLOCK0):                            If XPD_SDIO_FORCE & XPD_SDIO_REG                   = 1.8V R/W (0b0)
CLK8M_FREQ (BLOCK0):                               8MHz clock freq override                           = 57 R/W (0x39)
SPI_PAD_CONFIG_CLK (BLOCK0):                       Override SD_CLK pad (GPIO6/SPICLK)                 = 0 R/W (0b00000)
SPI_PAD_CONFIG_Q (BLOCK0):                         Override SD_DATA_0 pad (GPIO7/SPIQ)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_D (BLOCK0):                         Override SD_DATA_1 pad (GPIO8/SPID)                = 0 R/W (0b00000)
SPI_PAD_CONFIG_HD (BLOCK0):                        Override SD_DATA_2 pad (GPIO9/SPIHD)               = 0 R/W (0b00000)
SPI_PAD_CONFIG_CS0 (BLOCK0):                       Override SD_CMD pad (GPIO11/SPICS0)                = 0 R/W (0b00000)
DISABLE_SDIO_HOST (BLOCK0):                        Disable SDIO host                                  = False R/W (0b0)

Efuse fuses:
WR_DIS (BLOCK0):                                   Efuse write disable mask                           = 128 R/W (0x0080)
RD_DIS (BLOCK0):                                   Efuse read disable mask                            = 1 R/W (0x1)
CODING_SCHEME (BLOCK0):                            Efuse variable block length scheme
   = NONE (BLK1-3 len=256 bits) R/W (0b00)
KEY_STATUS (BLOCK0):                               Usage of efuse block 3 (reserved)                  = False R/W (0b0)

Identity fuses:
MAC (BLOCK0):                                      Factory MAC Address
   = 10:97:bd:d4:7e:e0 (CRC 0xb5 OK) R/W
MAC_CRC (BLOCK0):                                  CRC8 for factory MAC address                       = 181 R/W (0xb5)
CHIP_VER_REV1 (BLOCK0):                            Silicon Revision 1                                 = True R/W (0b1)
CHIP_VER_REV2 (BLOCK0):                            Silicon Revision 2                                 = True R/W (0b1)
CHIP_VERSION (BLOCK0):                             Reserved for future chip versions                  = 2 R/W (0b10)
CHIP_PACKAGE (BLOCK0):                             Chip package identifier                            = 1 R/W (0b001)
MAC_VERSION (BLOCK3):                              Version of the MAC field                           = 0 R/W (0x00)

Security fuses:
FLASH_CRYPT_CNT (BLOCK0):                          Flash encryption mode counter                      = 127 R/W (0b1111111)
UART_DOWNLOAD_DIS (BLOCK0):                        Disable UART download mode (ESP32 rev3 only)       = False R/W (0b0)
FLASH_CRYPT_CONFIG (BLOCK0):                       Flash encryption config (key tweak bits)           = 15 R/W (0xf)
CONSOLE_DEBUG_DISABLE (BLOCK0):                    Disable ROM BASIC interpreter fallback             = True R/W (0b1)
ABS_DONE_0 (BLOCK0):                               Secure boot V1 is enabled for bootloader image     = False R/W (0b0)
ABS_DONE_1 (BLOCK0):                               Secure boot V2 is enabled for bootloader image     = False R/W (0b0)
JTAG_DISABLE (BLOCK0):                             Disable JTAG                                       = True R/W (0b1)
DISABLE_DL_ENCRYPT (BLOCK0):                       Disable flash encryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_DECRYPT (BLOCK0):                       Disable flash decryption in UART bootloader        = True R/W (0b1)
DISABLE_DL_CACHE (BLOCK0):                         Disable flash cache in UART bootloader             = True R/W (0b1)
BLOCK1 (BLOCK1):                                   Flash encryption key
   = ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? ?? -/-
BLOCK2 (BLOCK2):                                   Secure boot key
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W
BLOCK3 (BLOCK3):                                   Variable Block 3
   = 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 R/W

Flash voltage (VDD_SDIO) determined by GPIO12 on reset (High for 1.8V, Low/NC for 3.3V).

mahavirj, 2023-04-18:
@AvinashSinghPES 

>  FLASH_CRYPT_CNT (BLOCK0): Flash encryption mode counter = 127 R/W (0b1111111)

This suggests that device has flash encryption enabled and all bits exhausted, hence flash enc can not be disabled on this device.

>  DISABLE_DL_ENCRYPT (BLOCK0): Disable flash encryption in UART bootloader = True R/W (0b1)

This suggests that encrypted writes are not allowed through UART DL mode and hence `idf.py encrypted-flash` command won't work here.

Most likely you have enabled the flash enc feature in the release configuration and hence its not recommended to erase or re-flash the plaintext bootloader. If you have the flash encryption key outside of the device then you may encrypt the bootloader on host side and re-flash it. If the flash encryption key was generated on the device then I am afraid that there is no way to recover this device.

mahavirj, 2023-04-19:
Unfortunately, no. Configuration has been set into the device EFuse storage which is one time programmable region, erasing the flash won't help here.

mahavirj, 2023-04-24:
Note: you can also try ESP32 security features under qemu (emulator), please see instructions available [here](https://github.com/espressif/qemu/wiki). Closing this issue, if you need any further assistance just drop a comment here.

closed 2023-04-24 by mahavirj