espressif/esp-idf #11213: Bind mqtt client to specific interface

jamesrob4, 2023-04-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I cannot find any documentation on binding an mqtt client to a single interface. My use case is I have PPPOS and Wifi interfaces and I need them to coexist. 
There will be an mqtt client on each interface, with a distinct client id. They publish to the same topic and I use a message counter to deduplicate messages on the broker. 
Most of the time both clients will be active, but I will have data saving modes where only the wifi client is active while PPPOS camps on the APN.

ESP-YJM, 2023-04-20:
@jamesrob4 You can follow the `esp_http_client` component https://github.com/espressif/esp-idf/blob/master/components/esp_http_client/include/esp_http_client.h#L159 to realize it on your side, the transport layer has supported it.

jamesrob4, 2023-06-09:
Are you able to provide any further guidance on this?

euripedesrocha, 2023-06-13:
Hi @jamesrob4 I just created an internal MR providing this feature, you can test it with the attached patch. 
[bind_interface.zip](https://github.com/espressif/esp-idf/files/11732771/bind_interface.zip)

Apply the patch to esp_mqtt to test it before we merge it.

euripedesrocha, 2023-06-19:
@jamesrob4 I think it's unrelated but, can you provide more information? It would be great if you can share the code to reproduce the error.

jamesrob4, 2023-06-21:
I reinstalled esp-idf from scratch and that problem went away. The issue I am getting now is below:

euripedesrocha, 2023-06-22:
Hi @jamesrob4,

This happens because you are not using the correct interface key. 

You can get it from `esp_netif` created by `esp_netif_create_default_wifi_sta`. This function returns the netif in use. You can use `esp_netif_get_ifkey` to retrieve the correct interface key.

closed 2023-06-22 by euripedesrocha

jamesrob4, 2023-06-25:
I did what you advised and found an issue with the size of ifr_name
And then when I worked around it I still got the same error.

```
esp_netif_t * myNetif = esp_netif_create_default_wifi_sta();
const char * wifiIfName = esp_netif_get_ifkey(myNetif);
ESP_LOGI(TAG, "wifi if_name is %s", wifiIfName);

const esp_mqtt_client_config_t mqtt_cfg = {
        .broker = {
            .address.uri = CONFIG_BROKER_URI,
            .verification.certificate = (const char *) cert,
            .address.port = 443,
        },
        .network.if_name = "WIFI_STA_DEF"
};

I (730) wifi station: wifi if_name is WIFI_STA_DEF
E (5350) esp-tls: Bind [sock=54] to interface WIFI_S fail
E (5350) esp-tls: Failed to open new connection
E (5350) transport_base: Failed to open a new connection
E (5360) mqtt_client: Error transport connect
```

In sockets.h I found that NETIF_NAMESIZE is 6

```
/* Set socket options argument */
#define IFNAMSIZ NETIF_NAMESIZE
struct ifreq {
  char ifr_name[IFNAMSIZ]; /* Interface name */
};
```

To carry on my testing I changed the defualt to WIFI_ 

```
#define ESP_NETIF_INHERENT_DEFAULT_WIFI_STA() \
    {   \
        .flags = (esp_netif_flags_t)(ESP_NETIF_IPV4_ONLY_FLAGS(ESP_NETIF_DHCP_CLIENT) | ESP_NETIF_DEFAULT_ARP_FLAGS | ESP_NETIF_DEFAULT_MLDV6_REPORT_FLAGS | ESP_NETIF_FLAG_EVENT_IP_MODIFIED), \
        ESP_COMPILER_DESIGNATED_INIT_AGGREGATE_TYPE_EMPTY(mac) \
        ESP_COMPILER_DESIGNATED_INIT_AGGREGATE_TYPE_EMPTY(ip_info) \
        .get_ip_event = IP_EVENT_STA_GOT_IP, \
        .lost_ip_event = IP_EVENT_STA_LOST_IP, \
        .if_key = "WIFI_", \
        .if_desc = "sta", \
        .route_prio = 100, \
        .bridge_info = NULL \
     }  \
```

And got the following output:

```
E (4360) esp-tls: Bind [sock=54] to interface WIFI_ fail
E (4360) esp-tls: Failed to open new connection
E (4360) transport_base: Failed to open a new connection
E (4360) mqtt_client: Error transport connect
```

euripedesrocha, 2023-06-27:
Hi @jamesrob4, after discussing it with my colleagues that work on `esp_netif` I learned that I misguided you with that function suggestion. Sorry. 
It wasn't supposed to match lwip name, that is the one you need here, and somehow it worked once I tested the solution locally. 

The correct way to get the key is like this:
```cpp
    struct ifreq ifr;
    esp_netif_get_netif_impl_name(netif, ifr.ifr_name);
```
And you pass the pointer to that struct in the mqtt config struct. 

I also suggest that you take a look on espressif/esp-protocols#286 which is an example for handling multiple netifs.