espressif/esp-idf #11216: PSRAM not used for xTaskCreate

law-ko, 2023-04-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello,

We have ESP32-PICO-MINI version which has 2MB of PSRAM. However, we realized that the internal memory are still being used even with SPIRAM enabled. Below is the sdkconfig:
```
#
# SPI RAM config
#
CONFIG_SPIRAM_TYPE_AUTO=y
# CONFIG_SPIRAM_TYPE_ESPPSRAM16 is not set
# CONFIG_SPIRAM_TYPE_ESPPSRAM32 is not set
# CONFIG_SPIRAM_TYPE_ESPPSRAM64 is not set
CONFIG_SPIRAM_SIZE=-1
CONFIG_SPIRAM_SPEED_40M=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_BOOT_INIT=y
# CONFIG_SPIRAM_IGNORE_NOTFOUND is not set
# CONFIG_SPIRAM_USE_MEMMAP is not set
# CONFIG_SPIRAM_USE_CAPS_ALLOC is not set
CONFIG_SPIRAM_USE_MALLOC=y
CONFIG_SPIRAM_MEMTEST=y
CONFIG_SPIRAM_MALLOC_ALWAYSINTERNAL=8192
CONFIG_SPIRAM_TRY_ALLOCATE_WIFI_LWIP=y
CONFIG_SPIRAM_MALLOC_RESERVE_INTERNAL=32768
# CONFIG_SPIRAM_ALLOW_BSS_SEG_EXTERNAL_MEMORY is not set
# CONFIG_SPIRAM_ALLOW_NOINIT_SEG_EXTERNAL_MEMORY is not set
CONFIG_SPIRAM_CACHE_WORKAROUND=y
```

With the current sdkconfig configuration, if the internal memory is fully allocated, will it automatically use SPIRAM? 
And how can we check internal memory space left? `printf("HEAP left: %d\n", esp_get_free_heap_size());` only shows the memory as a whole.

Thank you

ESP-Marius, 2023-04-18:
xTaskCreate will always allocate from internal memory, if you want to use external memory for tasks you need to allocate it using `xTaskCreateStatic` and pass in the buffer yourself. See [this](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/external-ram.html#restrictions) for more info.

law-ko, 2023-04-18:
@ESP-Marius ,

Thanks for the information, we have changed the task creation as below:
```
StaticTask_t xFleetProvTaskBuffer;
	StackType_t *xFleetProvStack = heap_caps_malloc(fleetprovbyclaimconfigTASK_STACK_SIZE, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT | MALLOC_CAP_32BIT);
	xTaskCreateStatic( prvFleetProvisioningByClaimTask,
					"FleetProvByClaim",
					fleetprovbyclaimconfigTASK_STACK_SIZE,
					pxNetworkContext,
					fleetprovbyclaimconfigTASK_PRIORITY,
					xFleetProvStack,
					&xFleetProvTaskBuffer );
```
We want to be flexible on using either internal or PSRAM, is setting `MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT | MALLOC_CAP_32BIT` the correct way?

In addition, we result in a `LoadProhibited` when executing `rand()` function, which does not happen when using internal RAM (by `xTaskCreate`):
```
/* Add a little randomness into the delay so the tasks don't remain
         * in lockstep. */
        vTaskDelay( pdMS_TO_TICKS( 125 ) +
                        ( rand() % 0xff ) );
```

Log:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40091982  PS      : 0x00060830  A0      : 0x800e4536  A1      : 0x3f8126d0  
0x40091982: rand at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/rand.c:80 (discriminator 5)

A2      : 0x3ffe5c30  A3      : 0x3f40c5a0  A4      : 0x3f40ccf4  A5      : 0x000027f7  
A6      : 0x3f40c5a0  A7      : 0x00000002  A8      : 0x00000001  A9      : 0x00000001  
A10     : 0x4c957f2d  A11     : 0x3f40c5a0  A12     : 0x3f40ccf4  A13     : 0x3f8126d0  
A14     : 0x3f8126b0  A15     : 0x0000000c  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000011  LBEG    : 0x40092079  LEND    : 0x40092089  LCOUNT  : 0xfffffff9  
0x40092079: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:84

0x40092089: strlen at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/strlen.S:96

Backtrace:0x4009197f:0x3f8126d00x400e4533:0x3f8126f0 0x400e4941:0x3f812760 0x40093d41:0x3f815a70 
0x4009197f: rand at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/rand.c:80 (discriminator 5)

0x400e4533: prvPublishToTopic at /home/espuser/esp-git/esp-homekit-sdk-mfi-3/examples/project_demo/build/../main/aws_iot/fleet_prov_by_claim_demo.c:793 (discriminator 15)

0x400e4941: prvFleetProvisioningByClaimTask at /home/espuser/esp-git/esp-homekit-sdk-mfi-3/examples/project_demo/build/../main/aws_iot/fleet_prov_by_claim_demo.c:1258

0x40093d41: vPortTaskWrapper at /home/espuser/esp/esp-idf/components/freertos/port/xtensa/port.c:131
```

Dazza0, 2023-04-19:
Hi @law-ko , the usage of internal RAM you are seeing is actually an intentional change.

We used to allow FreeRTOS to automatically allocate to PSRAM (as long as `CONFIG_SPIRAM_USE_MALLOC` was enabled) since any dynamic memory allocation made by FreeRTOS would just route to `malloc()`. But we've recently made a change which causes dynamic memory allocated by FreeRTOS to always use internal memory (see https://github.com/espressif/esp-idf/commit/e21ab0332b3e637d2b0ed24650011b35283f830b).

The reason for this change was that if `CONFIG_SPIRAM_USE_MALLOC`, the user no idea and limited control on what FreeRTOS objects get placed in PSRAM, as we are relying on the heap allocate to make that decision. For example,  previously if a user called `xQueueCreate()`, they won't know if that queue is in internal RAM or PSRAM. This becomes a problem when that FreeRTOS object gets accessed while cache is disabled (e.g., when writing to SPI Flash), thus leading to `Cache disabled but cached memory region accessed`.

Thus, to increase safety, we've made all FreeRTOS dynamic memory default to internal memory. If want to place a FreeRTOS object into external or custom memory, there are now two ways to do this:

1. Use one of the `CreateWithCaps()` functions added in https://github.com/espressif/esp-idf/commit/4e7cd2e7066c1b9abb7d2e80826a2fef859f8bb5. For example:

```c
#include "freertos/idf_additions.h"
...
QueueHandle_t queue_in_psram = xQueueCreateWithCaps(5, sizeof(uint32_t), MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
```
**Note: `xTaskCreateWithCaps()` and `xTimerCreateWithCaps()` are still missing. `xTaskCreateWithCaps()` is in review, while `xTimerCreateWithCaps()` due to limitations of how FreeRTOS handles static deletion.**

2. Manually allocate the memory using `heap_caps_malloc()` then create the object using the `CreateStatic()` functions of FreeRTOS. For example:

```c
    StaticTask_t * pxTaskBuffer;
    StackType_t * pxStack;
    // TCBs should always be in internal RAM
    pxTaskBuffer = heap_caps_malloc(sizeof(StaticTask_t), MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
    // Allocate the stack in PSRAM
    pxStack = heap_caps_malloc(4096, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
    // Create the task using static creation API
    TaskHandle_t static_task = xTaskCreateStatic(task_func, "task_name", 4096, task_arg, task_priority, pxStack, pxTaskBuffer);
```

I'll leave this issue open until `xTaskCreateWithCaps()` gets merged.

ESP-Marius, 2023-04-20:
> We used to allow FreeRTOS to automatically allocate to PSRAM (as long as CONFIG_SPIRAM_USE_MALLOC was enabled) since any dynamic memory allocation made by FreeRTOS would just route to malloc(). But [...]

`xTaskCreate` was always a special case wasnt it? It would never automatically allocated to PSRAM.

Dazza0, 2023-04-20:
@ESP-Marius Yes, what used to happen was that `xTaskCreate...()` functions would used to call `pvPortMallocTcbMem()` and `pvPortMallocStackMem()` to force task memory into internal memory, whereas all other objects would call the regular `pvPortMalloc()` which just routes to `malloc()` (thus can place memory into PSRAM if `CONFIG_SPIRAM_USE_MALLOC()`.

Now, all memory allocation by FreeRTOS calls `pvPortMalloc()`, which only allocates from internal memory.

law-ko, 2023-04-20:
@Dazza0 ,

Thank you for the information, with the method of using `heap_caps_malloc` as below:
```
StaticTask_t *xFleetProvTaskBuffer;
StackType_t *xFleetProvStack;

/* Create an instanace of fleetProvTask on top of PSRAM. */
ESP_LOGW(TAG, "Creating fleetProvTask on PSRAM");
xFleetProvTaskBuffer = heap_caps_malloc(sizeof(StaticTask_t), MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
xFleetProvStack = heap_caps_malloc(fleetprovbyclaimconfigTASK_STACK_SIZE, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
TaskHandle_t fleetProvStaticTask = 	xTaskCreateStatic( prvFleetProvisioningByClaimTask,
										"FleetProvByClaim",
										fleetprovbyclaimconfigTASK_STACK_SIZE,
										pxNetworkContext,
										fleetprovbyclaimconfigTASK_PRIORITY,
										xFleetProvStack,
										xFleetProvTaskBuffer );
```

However, when trying to save certificates to NVS, it causes `rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)` and reboots. This does not happen when using `xTaskCreate`
```
I (12331) fleet_prov_by_claim_demo: Saving certificate to NVS...
ets Jul 29 2019 12:21:46

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 271414342, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6616
load:0x40078000,len:14784
load:0x40080400,len:3584
0x40080400: _init at ??:?

entry 0x40080660
I (55) boot: ESP-IDF v4.4-253-g5ff09f0a06 2nd stage bootloader
I (55) boot: compile time 00:58:34
I (55) boot: chip revision: 3
I (60) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (67) boot.esp32: SPI Speed      : 40MHz
I (72) boot.esp32: SPI Mode       : DIO
I (76) boot.esp32: SPI Flash Size : 8MB
W (81) boot.esp32: PRO CPU has been reset by WDT.
W (86) boot.esp32: WDT reset info: PRO CPU PC=0x4009e7c3
0x4009e7c3: xthal_save_extra_nw at /Users/igrokhotkov/e/esp32/hal/hal/state_asm.S:72
```

law-ko, 2023-04-22:
@Dazza0 , I followed your way of setting of TaskBuffer and Stack with `heap_caps_malloc` and it would cause reset (reboot) when executing `nvs_set_str()` :
```
	/* Create an instanace of prvFleetProvisioningByClaimTask on top of PSRAM. */
	ESP_LOGW(TAG, "Creating prvFleetProvisioningByClaimTask on PSRAM");
	xFleetProvTaskBuffer = heap_caps_malloc(sizeof(StaticTask_t), MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
	xFleetProvStack = heap_caps_malloc(fleetprovbyclaimconfigTASK_STACK_SIZE, MALLOC_CAP_SPIRAM | MALLOC_CAP_8BIT);
	TaskHandle_t fleetProvStaticTask = 	xTaskCreateStatic( prvFleetProvisioningByClaimTask,
										"FleetProvByClaim",
										16384,
										pxNetworkContext,
										5,
										xFleetProvStack,
										xFleetProvTaskBuffer );
```

However, if changing `xFleetProvStack` to `MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT` it would be fine running nvs related function calls.

I am currently on `ESP-IDF v4.4-253-g5ff09f0a06` for my `idf.py --version`

closed 2023-04-25 by espressif-bot

igrr, 2023-04-25:
According to https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/kconfig.html#config-spiram-allow-stack-external-memory,

> This should only be used for tasks where the stack is never accessed while the cache is disabled.

Since reading or writing flash requires cache to be disabled, tasks which need to perform flash operations can't have their stacks in PSRAM.

Unfortunately we have missed mentioning this limitation in the ESP32 version of the docs, sorry about that!

law-ko, 2023-04-25:
@igrr Would it make sense to create another task on internal RAM specifically for NVS access when the task is on PSRAM?

So inside `xTaskCreateStatic` call `xTaskCreate` when trying to access the NVS.

Dazza0, 2023-04-26:
@law-ko 

> Would it make sense to create another task on internal RAM specifically for NVS access when the task is on PSRAM?

Generally yes, If a task will disable the cache (e.g., call an SPI flash write or NVS write operation), it would be a good idea to  place that task entirely in internal RAM. If your application has multiple places that needs to disable cache, it may be a good idea to group all of the SPI Flash/NVS writes to be done from a dedicated task that is placed in internal RAM.

The cache disabling procedure is implemented in [`cache_utils.c`](https://github.com/espressif/esp-idf/blob/v5.1-dev/components/spi_flash/cache_utils.c). Disabling/enabling the cache involves the steps described below. Assume there is a Task A on Core A, and a Task B on Core B:

- Task A calls some cache disabling function (e.g., an SPI flash write)
- The cache disabling procedure will...
    - Prevent Task A from being preempted
    - Preempt Task B with the IPC task. The IPC is task on Core B simply loops at the highest priority (prevent any other task from running on Core B)
- Task A now has carries out its SPI flash write
    - Task A on Core A must ensure that during this section of code, it does not touch any external memory (including static memory, dynamic memory, stack etc).
    - Core B is guaranteed to only access internal memory at this time since the IPC task spinning on Core B is placed in IRAM
- Task A has finished its SPI flash. The cache reenabling procedure will...
    - Allow Task A can now be preempted again
    - Block the IPC task so that core B can now go back to running task B