espressif/esp-idf #11217: sdmmc_card_init failed

JuneSheng11, 2023-04-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v4.4.4

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32-s3-korvo-2

### Power Supply used.

USB

### What is the expected behavior?

解决

### What is the actual behavior?

运行 example/sd_card/sdmmc idf.py -p COM4 monitor
打印以下错误：
I (290) sleep: Configure to isolate all GPIO pins in sleep state
I (296) sleep: Enable automatic switching of GPIO sleep configuration
I (303) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (324) example: Initializing SD card
I (324) example: Using SPI peripheral
I (324) example: Mounting filesystem
I (334) gpio: GPIO[34]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (384) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (384) vfs_fat_sdmmc: sdmmc_card_init failed (0x108).
I (384) gpio: GPIO[34]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (394) example: Failed to initialize the card (ESP_ERR_INVALID_RESPONSE). Make sure SD card lines have pull-up resistors in place.

### Steps to reproduce.

1. cd example/sd_card/sdmmc
2. idf.py set-target esp32s3
3. idf.py build
4. idf.py -p COM4 flash
5. idf.py -p COM4 monitor
...

adokitkat, 2023-04-18:
Hello. Are you sure you have set the correct SDMMC GPIO pins in `idf.py menuconfig` -> `Example Config`? The `sd_card/sdmmc` example works without modification on ESP32-S3-USB-OTG board but for other boards you have to configure the correct pins. You can find the GPIO pin allocation table for S3-Korvo [here](https://espressif-docs.readthedocs-hosted.com/projects/esp-adf/en/latest/design-guide/dev-boards/user-guide-esp32-s3-korvo-2.html#gpio-allocation-summary)  under MicroSD column. Note on S3-Korvo you can only use [1-line SD mode](CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_1) (`CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_1`) which you have to also set in `idf.py menuconfig`.

JuneSheng11, 2023-04-18:
when I run 
1.  cd esp-skainet\examples\voice_communication
2.  idf.py set-target esp32s3
3. idf.py menuconfig
can't see Example Confige and line SD mode](https://github.com/espressif/esp-idf/issues/CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_1)

adokitkat, 2023-04-21:
Is your issue solved please? I noticed you posted a comment here and then deleted it.

adokitkat, 2023-04-24:
I am sorry but this is not related to the original issue. Please open issue on [esp-skainet](https://github.com/espressif/esp-skainet/issues).

closed 2023-04-26 by adokitkat