espressif/esp-idf #11220: Offline OTA

franz-hoepfinger-at-hm-edu, 2023-04-18:
### Is your feature request related to a problem?

Is there a Way of Offline OTA ? 

the Engeneer going to the Machine has the bin File with him. 
we use the https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving Example to upload this bin to the ESP32. 

after a Reboot the ESP32 installs it like OTA, with the Difference it does not need a actual Internet Connection.

franz-ms-muc, 2023-04-18:
@igrr maybe a topic for you ?

chegewara, 2023-04-18:
There is plenty options to OTA without internet:
- http server - i believe there are examples and/or libraries to do it
- bluetooth/ble
- from pendrive (USB on esp32 S3)

igrr, 2023-04-18:
@franz-hoepfinger-at-hm-edu you could use the https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving example, but instead of writing the uploaded file to the filesystem, call esp_ota_write to save it to the OTA partition.

Other transport methods would also work: SD card, UART, BLE — whichever way you can get the stream of data, then pass it to esp_ota_write API.

hmalpani, 2023-05-08:
Hello @franz-hoepfinger-at-hm-edu 
I have created a simple example to demonstrate OTA update by uploading the firmware to the HTTP server. I hope it helps.
I am closing this issue as this will help you in implementing the requested functionality. Please feel free to reopen if you have further questions. Thanks!

Link to example: https://github.com/hmalpani/HTTP_Server_OTA

closed 2023-05-08 by hmalpani

diplfranzhoepfinger, 2023-12-28:
Hi, 
i found also in https://github.com/espressif/esp-iot-bridge/tree/master there is a OTA Button in the Web-Interface.

diplfranzhoepfinger, 2023-12-29:
Hi, 

i made a Fork: 
https://github.com/diplfranzhoepfinger/HTTP_Server_OTA

there i made a Integration: 
https://github.com/diplfranzhoepfinger/HTTP_Server_OTA/tree/feature/integrate_with_webserver

so the 
https://github.com/espressif/esp-idf/tree/master/examples/protocols/http_server/file_serving
Example works like before, only if the Name is "Application.bin" it starts the OTA Process. 

cool, right ?